micro-ROS/micro_ros_setup #315: Topic with array of predefined size type message

bjv-capra, 2021-05-12:
## Issue template

- Hardware description: STM32F7
- RTOS: FreeRTOS
- Installation type: static library
- Version or commit hash: foxy

#### Steps to reproduce the issue
I have defined a subscriber to a topic as usual on my client. the message type is

```
capra/Vector3 data[4]
```
I see the struct generated as follows
```
typedef struct capra__msg__Array
{
  capra__msg__Vector3 data[4];
} capra__msg__Array;
```
I can also see that there's a method generated on the `functions.c`, that would initialize the array of size 4 as it should.

My code compiles, I see the topic is announced, I see the correct data type, however, when the subscription is triggered, the message doesn't get filled and everything is full of 0's.

I couldn't find any example with a fixed size array.

Furthermore, one thing that got my attention is that all the array message types I found, have, besides the data member, a sequence member, which I don't have on my message. I guess this is because of the fixed size?

Anyways, I can't get my array to work. I'll try out using one of the built-in types to compare

pablogs9, 2021-05-12:
Do you have both the publisher and the subscriber in the same micro-ROS node? Or the subscriber is in ROS 2? Can you describe the setup or share a snippet to replicate this?

bjv-capra, 2021-05-12:
The publisher is ROS2 (from the terminal) the subscriber is on micro-ROS.

I'll see if @anastasiapan can provide some snippet.

anastasiapan, 2021-05-12:
The publisher is: `ros2 topic pub --once --qos-durability=transient_local --qos-reliability=best_effort /topic capra/msg/Array "{data : [{x: 1.0, y: 2.0, z: 3.0},{x: 1.0, y: 2.0, z: 3.0},{x: 1.0, y: 2.0, z: 3.0},{x: 1.0, y: 2.0, z: 3.0}]}"`

and the initialization happens just before the `rclc_executor_add_subscription`. We are using classes and the subscription part of the node looks like that:

```
capra__msg__Array__init(&msg_);

  rclc_result = rclc_executor_add_subscription(
    &executor_, &subscriber_, &msg_,
    [](const void * msgin) {
      const capra__msg__Array * msg = (const capra__msg__Array *)msgin;
    },
    ON_NEW_DATA);
```

When I publish the message it hits the breakpoint inside the callback and everything is initialized correctly to 0 and once I step over the message still remains at all zeros.

pablogs9, 2021-05-12:
Found bug, coming with solution in a while

pablogs9, 2021-05-12:
Please retest your code when https://github.com/micro-ROS/rosidl_typesupport_microxrcedds/pull/29 merged

bjv-capra, 2021-05-12:
Will do, and thanks again for the quick response.

pablogs9, 2021-05-12:
https://github.com/micro-ROS/rosidl_typesupport_microxrcedds/pull/29 merged, I'm closing this preventively bc I think that the new implementation for Namespaced Arrays in micro-ROS type support will solve the issue.

Do not hesitate to reopen if you have any problems related to this.

closed 2021-05-12 by pablogs9

bjv-capra, 2021-05-12:
It did fix the issue, we can now receive messages as expected. Thanks!