micro-ROS/micro_ros_setup #316: Service call causes all communication to stop

bjv-capra, 2021-05-14:
## Issue template

- Hardware description: STM32F7
- RTOS: FreeRTOS
- Installation type: static lib
- Version or commit hash: foxy

When running my client-agent communication I see the connection starting as expected. I have a few topics (in and out), different nodes, and a service on my client.

I see there's some constant traffic on the agent (I guess the odometry if it's coming from the client. The first call to the service works just flawlessly (from a caller's point of view), however, I can see that after I get the response, the traffic on the agent stops.

The next time I try to call the service, it sends the message (I don't see it arriving on my uc), and then, as it gets no response, the agent keeps spamming. causing that it fills all the buffers on my board and I quickly get

```
Network buffers: 49 lowest 49
Network buffers: 47 lowest 47
Network buffers: 46 lowest 46
Network buffers: 46 lowest 45
Network buffers: 44 lowest 44
Network buffers: 43 lowest 43
Network buffers: 43 lowest 42
Network buffers: 41 lowest 41
Network buffers: 41 lowest 40
Network buffers: 39 lowest 39
xProcessReceivedUDPPacket: buffer full 16 >= 16 port 13609
xProcessReceivedUDPPacket: buffer full 16 >= 16 port 13609
```

Here's the output from the agent running with -v6

```
[1620981205.811399] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4C4E3596, len: 360, data: 
0000: 81 01 3E 04 07 01 60 01 04 58 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0160: 00 00 00 00 00 00 00 00
[1620981205.811485] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 348, data: 
0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1620981205.911372] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x4C4E3596, len: 360, data: 
0000: 81 01 3F 04 07 01 60 01 04 59 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00A0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0160: 00 00 00 00 00 00 00 00
[1620981205.911458] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 348, data: 
[... 66 lines cut ...]
00C0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00E0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0140: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[1620981206.111762] debug    | Replier.cpp        | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 28, data: 
0000: 01 0F A0 A9 70 00 00 00 01 00 00 00 00 00 12 03 00 00 00 00 01 00 00 00 40 AE 01 20
[1620981206.111911] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4C4E3596, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 11 00 00 00 80
[1620981224.153834] debug    | Replier.cpp        | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 28, data: 
0000: 01 0F A0 A9 84 00 00 00 01 00 00 00 00 00 12 03 00 00 00 00 01 00 00 00 06 04 00 00
[1620981224.153892] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4C4E3596, len: 40, data: 
0000: 81 80 0E 00 09 01 20 00 00 14 00 08 01 0F A0 A9 84 00 00 00 01 00 00 00 00 00 12 03 00 00 00 00
0020: 01 00 00 00 06 04 00 00
[1620981224.354067] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4C4E3596, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0E 00 0E 00 80
[1620981224.554239] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4C4E3596, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0E 00 0E 00 80
[1620981224.754367] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4C4E3596, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0E 00 0E 00 80
[1620981224.954517] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x4C4E3596, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 0E 00 0E 00 80
```

The last message is repeated until I get the error on my board.

pablogs9, 2021-05-14:
I need the message types and some code to replicate the issue.

bjv-capra, 2021-05-14:
MyMessage.msg
```
int32 value
```

MyService.srv
```
capra_msgs_embedded/MyMsg value_in
---
# Confirm
capra_msgs_embedded/MyMsg value_out
```

I can't give you the exact code to reproduce unfortunately because it uses some internal wrappers we've made. However, the code looks something like...
```
  // Initialize support (all default)
  // Initialize Node (all default)
  // Initialize executor (all default), 1 handle

  // Create publisher
  auto rclc_result = rclc_publisher_init_best_effort(
    &publisher_, &node_, ROSIDL_GET_MSG_TYPE_SUPPORT(capra_msgs_embedded, msg, AType),
    SOME_TOPIC_OUT);

  // Initialize service
  rclc_result = rclc_service_init_default(
    &service_, &node_,
    ROSIDL_GET_SRV_TYPE_SUPPORT(
      capra_msgs_embedded, srv,
      MyService) ,
    SERVICE_TOPIC);

  capra_msgs_embedded__srv__MyService_Response res;
  capra_msgs_embedded__srv__MyService_Request req;

  rclc_executor_add_service(
    &executor_, &service_, &req, &res, [](const void * req, void * res) {
      auto req_in = *static_cast<const capra_msgs_embedded__srv__MyService_Request *>(req);
      auto res_in = *static_cast<capra_msgs_embedded__srv__MyService_Response *>(res);
    });
```

pablogs9, 2021-05-14:
Hello @bjv-capra,

I have been testing your approach and I have some questions:
- Is possible for you to isolate the service and replicate the issue with no other communication between client and agent?
- What is the rate of the other publisher(s)?

I have been testing the code attached below and everything works smooth. Can you try to use this code in your embedded platform to check if it works?

```c
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <capra/srv/my_service.h>
#include <capra/msg/my_message.h>

#include <stdio.h>
#include <unistd.h>

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

rcl_publisher_t publisher;

void service_callback(const void * req, void * res){
  capra__srv__MyService_Request * req_in = (capra__srv__MyService_Request *) req;
  capra__srv__MyService_Response * res_in = (capra__srv__MyService_Response *) res;

  printf("Service request value: %d.\n", (int) req_in->value_in.value);

  res_in->value_out.value = req_in->value_in.value + 10;
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
	(void) timer;
	(void) last_call_time;

  static capra__msg__MyMessage msg = {0};
	rcl_publish(&publisher, (const void*)&msg, NULL);
  msg.value++;
}

void main(void)
{	
    rcl_allocator_t allocator = rcl_get_default_allocator();

    // create init_options
    rclc_support_t support;
    RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

    // create node
    rcl_node_t node;
    RCCHECK(rclc_node_init_default(&node, "add_twoints_client_rclc", "", &support));

    rclc_publisher_init_best_effort(
      &publisher, 
      &node, ROSIDL_GET_MSG_TYPE_SUPPORT(capra, msg, MyMessage), 
      "capra_pub"
    );

    // create service
    rcl_service_t service;
    RCCHECK(rclc_service_init_default(&service, &node, ROSIDL_GET_SRV_TYPE_SUPPORT(capra, srv, MyService), "capra_srv"));

    // Create a 3 seconds ping timer timer,
    rcl_timer_t timer;
    RCCHECK(rclc_timer_init_default(&timer, &support, RCL_MS_TO_NS(500), timer_callback));

    // create executor
    rclc_executor_t executor;
    RCCHECK(rclc_executor_init(&executor, &support.context, 2, &allocator));

    capra__srv__MyService_Response res;
    capra__srv__MyService_Request req;
    RCCHECK(rclc_executor_add_service(&executor, &service, &req, &res, service_callback));
  	RCCHECK(rclc_executor_add_timer(&executor, &timer));

    while (1)
    {
      rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
      usleep(100000);
    }
    

    RCCHECK(rcl_service_fini(&service, &node));
    RCCHECK(rcl_node_fini(&node));
}
```

Here the test:

[image]

bjv-capra, 2021-05-14:
I'll check it out, however, take into account that my service message is nested, and I publish another type on the publisher.

I'll isolate it and make sure only the service works. Be back soon with answers.

The rate of the other publisher is 10 hz

pablogs9, 2021-05-14:
My message is also nested:

[image]

How big is the topic type in the publisher?

bjv-capra, 2021-05-14:
The topic type is quite small, it's basically an odometry message with floats instead of doubles.

I've tested it out by isolating the service, even removing the publisher. The moment I call the service, I receive the correct message, and then I've also added the step of responding to the service as you did.

Interesting, the moment I started to reply over the response, I trigger hard faults, like if the code suddenly jumps to wrong memory. If I remove the response, it'd seem I can somewhat keep sending messages.

However, I need to send the messages multiple times sometimes, before the agent actually hands it over. (ctrl+c and go again)

I've been tracing my stack usage and it all seems to be fine (all within correct ranges). I'm kinda dazzled

bjv-capra, 2021-05-14:
Btw, I'm also going to bring to attention [this](https://github.com/micro-ROS/micro_ros_setup/issues/310) issue, we had 1 or two weeks ago, and now it doesn't happen anymore.

pablogs9, 2021-05-14:
One more question, in which memory space does exists `capra_msgs_embedded__srv__MyService_Response res;` and `capra_msgs_embedded__srv__MyService_Request req;` ?

They are a couple of messages that are being provided to the executor in order to let the executor have some memory for serializing and deserializing data. If the exists in the stack of a function that ends at some point, when the executor handles the request or the reply the memory will have garbage...

Can you try to set this couple of variables `static`, or put them in the global scope?

bjv-capra, 2021-05-14:
Uhm...interesting point, let me try

bjv-capra, 2021-05-14:
So, clearly, that was a problem. I made them members of the class and problem-solved (regarding the return).

However, I'm back to square 0 where after the first call to the service, all my ROS entities die. However, if it's just the service and the publisher, it seems to be fine. On the other hand, I have to repeatedly call the service before I get to send the message, and after some retries, it just doesn't go through anymore. And I don't see any log change happening on the agent, except the publisher data coming in.

bjv-capra, 2021-05-16:
I've continued my tests, it'd seem that with a single node, I can have a publisher and the service with no problem.
I then added another node, running on another thread, first with a subscriber, immediately after I send the first call to the service, I would not be able to send anything anymore from any of the ROS threads(two nodes). However, if instead of sending a request to the service, I'd do it to the subscription, it'd work with no problems.
I then was wondering if it was due to the executors for some reason, so I changed the second node, to have a publisher instead of a subscription. However, the moment the second node comes alive, and they try to send messages, it'd send one time, then stop working, exactly as what I noticed when I send the service request.
What I did then was to try to follow deep into the code, to understand exactly WHEN would it stop running. And, with the two publishers, it seems it'd be [here](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/foxy/src/c/core/session/write_access.c#L118), I get nothing more after that point, and it's when I can't publish anymore.

I have two nodes

A) Thread 1, 1 Node, publishing odometry and with a service (with a message as described)
B) Thread 2, 1 Node, publishing a complex message type, yet smaller than the odometry.

type.msg
```
boolean something
boolean another_thing
Twist twist
```

The rclc support is initialized just once.

B is initialized first, so it starts sending messages a tiny bit before the second thread starts sending messages.

After B is set to run, A is configured (no problem)
A sends 1 message, then none of the nodes does anything more, however, the rest of the system still works. This could indicate that it's simply this lock never returning.

pablogs9, 2021-05-17:
How are you creating two nodes in two different threads? Are you calling `rclc_support_init` twice?

I think that the best approach by now for micro-ROS is to create all the entities (nodes, pubs, subs...) in an initial thread, and then spawn other threads for publishing or running executors (always taking into account that you need to enable the multithreaded support because it is disabled by default).

bjv-capra, 2021-05-17:
`rcl_support_init` is called once. I have a helper static class that takes care of handling support initialization, node creation, and executor initialization.
I don't spawn threads on the fly, everything has been already statically allocated and I have a lifecycle handler that takes care of configuring them.
Anyways, the part that loops on the "active" state of the thread is the executor or publisher. Everything else is handled in single steps.
I have the multithreaded support enabled. At least I have set the flags accordingly when building micro-ros.

I believe both service and topic publisher suffers from the same problem on the same point. As I get the reply of the service the first time I call it, not the second. So, the write blocks on the line I commented on before.

pablogs9, 2021-05-17:
I have added [these changes](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/240) in order to unlock the middleware streams in the case that your message cannot be buffered (because of lack of space in the buffers)... Can you try when https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/240 is merged? Just update the repository and rebuild your library.

BTW, can you share your `colcon.meta`?

bjv-capra, 2021-05-17:
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF",
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON",
                "-DconfigUSE_POSIX_ERRNO=1",
                "-DCMAKE_VERBOSE_MAKEFILE:BOOL=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_MULTITHREAD=ON",
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=4",
                "-DRMW_UXRCE_MAX_PUBLISHERS=2",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=10",
                "-DRMW_UXRCE_MAX_SERVICES=2",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=8",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=10.0.0.2",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=8888",
            ]
        }
    }
}
```

pablogs9, 2021-05-17:
https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/240 merged, let us know if the issue still happens

bjv-capra, 2021-05-17:
I'll do it now

bjv-capra, 2021-05-17:
Still the same issue, but I quite can't decipher where or why. I can tell you that immediately after I receive the response from the service, everything stops. As well as when there are two publishers. And when I mean stop, those two threads don't get their handle back basically, however, the rest of the system is still running. I've tried increasing the stack size (even though the high watermarks indicate there's no need and I don't get any stack overflow nor hard fault) without any success.

We could set a quick call if you'd like where I can walk you through the case

pablogs9, 2021-05-17:
I have fixed another inconsistent behavior related to services in RMW here: https://github.com/micro-ROS/rmw-microxrcedds/pull/132

Please try this new RMW commit, it this does not solves anything I'm going to try to replicate you use case on our side.

bjv-capra, 2021-05-17:
Still the same, after the first call, it doesn't work anymore. If I don't send the call, then it'll just work.

pablogs9, 2021-05-17:
Found a bug in the multithread implementation, I'm working on a solution.

pablogs9, 2021-05-17:
Ok, @bjv-capra fix here: https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/243

We are going to merge it ASAP. I'll close this when merged, but please feel free to reopen if you still have problems.

Thanks a lot for testing and reporting those kinds of issues, they are super valuable for us to keep improving micro-ROS!

bjv-capra, 2021-05-17:
Great! it's not a problem, we appreciate the timely response from you. I'll test it as soon as it's merged.

pablogs9, 2021-05-18:
@bjv-capra https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/ is merged, I'm closing, please test and reopen if neccesary.

bjv-capra, 2021-05-18:
I can't reopen the issue, however, the problem is still there. It happens, whenever I have a publisher and any other thread tries to also publish. So, if I have two publishers, when one starts, no problem, the moment the second one starts, I see one message on the agent received, and that's it. The same when I call a service, the call has no problem, the problem is when it replies back, then it crashes. Situations I've tried
1 Thread, with one service and one publisher, publish first, then call executor spin some, no problem.

2 Threads
1. 1 Service 1 publisher, publish first, then call executor spin some
2. 1 Publisher at 1 Hz (I've tried different frequencies just for the sake of trying).

1. 1 publisher at 1 Hz
2. 1 publisher at 1Hz

And different combinations of the above 2 threads. If I have multiple subscriptions there are no problems. It's only when I have multiple writers in different threads that I see the problem.

As you may have noticed from my colcon.meta, the nodes don't communicate between them, thus, there shouldn't be a need for shared memory (it's what you fixed).

pablogs9, 2021-05-18:
All those 3 1-Hz pubs and the service are each in a different thread?

bjv-capra, 2021-05-18:
Yikes, I should've seen the preview of how it came

They were 2 different scenarios, 1,2, and 1,2, each being on its own thread. And actually, they go faster than 1 Hz, I just pushed it down to 1 Hz for visualization

pablogs9, 2021-05-18:
I'm going to make some tests, back in a while.

bjv-capra, 2021-05-18:
What are the STREAM HISTORY and XML_BUFFER_LENGTH needed for?  I don't have them on my colcon.meta
I'll see to try it out later today or early tomorrow.

bjv-capra, 2021-05-18:
Another quick thing it's a little bit different is that you have only one node, while I have several (one node per thread). I don't know if this makes a significant difference

pablogs9, 2021-05-18:
`DRMW_UXRCE_STREAM_HISTORY` is related to the number of slots (of size `DUCLIENT_PROFILE_*_MTU` default 512 B) used in reliable XRCE communication. In my case 32 slots of 512 B each for sending messages and confirmation. I guess that this is not related to this problem. But in any case, try to increase it to see if we have another behavior.

`DRMW_UXRCE_XML_BUFFER_LENGTH` is currently deprecated, so don't mind.

I'm going to test with one node in each thread.

pablogs9, 2021-05-18:
It also works with: 

<details><summary>Test code</summary>
<p>

```c
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <capra/srv/my_service.h>
#include <capra/msg/my_message.h>

#include <stdio.h>
#include <unistd.h>
#include <pthread.h>

#define N_THREADS 5

rclc_support_t support;

typedef struct thread_entities_t {
  rcl_node_t node;
  rcl_publisher_t publisher;
  capra__srv__MyService_Response res;
  capra__srv__MyService_Request req;
  rcl_service_t service;
  rclc_executor_t executor;
} thread_entities_t;

thread_entities_t thread_entities[N_THREADS];

void service_callback(const void * req, void * res){
  capra__srv__MyService_Request * req_in = (capra__srv__MyService_Request *) req;
  capra__srv__MyService_Response * res_in = (capra__srv__MyService_Response *) res;

  printf("Service request value: %d on thread %ld.\n", (int) req_in->value_in.value, pthread_self());

  res_in->value_out.value = req_in->value_in.value + 10;
}

void * thread_function(void* arg){
    thread_entities_t * entities = (thread_entities_t *) arg;
    capra__msg__MyMessage msg = {0};

    while (1)
    {
      rcl_publish(
        &entities->publisher,
        (const void*)&msg,
        NULL
      );

      msg.value++;
      rclc_executor_spin_some(&entities->executor, RCL_MS_TO_NS(10));
      usleep(10000);
    }
}

void main(void)
{
    rcl_allocator_t allocator = rcl_get_default_allocator();

    // create init_options
    rclc_support_init(&support, 0, NULL, &allocator);

    // Creating entities
    for (size_t i = 0; i < N_THREADS; i++)
    {
      char buf[100];

      // create node
      sprintf(buf, "capra_node_%ld", i);
      rclc_node_init_default(&thread_entities[i].node, buf, "", &support);

      sprintf(buf, "capra_pub_%ld", i);
      rclc_publisher_init_best_effort(
        &thread_entities[i].publisher,
        &thread_entities[i].node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(capra, msg, MyMessage),
        buf
      );

      sprintf(buf, "capra_srv_%ld", i);
      rclc_service_init_default(
        &thread_entities[i].service,
        &thread_entities[i].node,
        ROSIDL_GET_SRV_TYPE_SUPPORT(capra, srv, MyService),
        buf
      );

      // create executor
      rclc_executor_init(
        &thread_entities[i].executor,
        &support.context,
        2,
        &allocator
      );

      rclc_executor_add_service(
        &thread_entities[i].executor,
        &thread_entities[i].service,
        &thread_entities[i].req,
        &thread_entities[i].res,
        service_callback
      );
    }

    // Running threads
    pthread_t ids[N_THREADS];
    for (size_t i = 0; i < N_THREADS; i++)
    {
      pthread_create(&ids[i], NULL, thread_function, &thread_entities[i]);
    }

    for (size_t i = 0; i < N_THREADS; i++)
    {
      pthread_join(ids[i], NULL);
    }
}
```

</p>
</details>

bjv-capra, 2021-05-19:
So, a quick update

First, my logic is a little bit different than yours in the sense that, my threads are already running by the time I want to configure the nodes. So, what you do on your main method, happens within a running thread, while you first configure and initialize everything before creating/running the threads.
We've implemented our own lifecycle and within it, we configure the nodes and active them, thus, we also have our kind of own executor (time-based)

Anyways,
I changed my setup so that all the nodes are first configured before trying to start doing anything. With this change, alone, things didn't change.
Then I noticed you cast the msg to `const void *` when calling publish, I did that, and at least both publishers started to work without any problem. However, the nodes NEED to be created/configured before starting to publish. That is, creating the entities in a multithreaded/async manner doesn't seem plausible.

So, thus far, great improvement, however, I still run into the issue that, when calling the service, it will simply stop publishing and won't reply back to the service.

bjv-capra, 2021-05-19:
I also noticed you have 2 handles on your executor, any reason in particular?
I haven't yet tried rebuilding micro-ros with `DRMW_UXRCE_STREAM_HISTORY` not to default

pablogs9, 2021-05-19:
Two handles in the executor are an error, just one is required for the service server.

And regarding the order, yes, let's try to make it work with a:
1. single thread entity creation stage
2. launching threads and publishing and handling services

Probably later we implement the feature of creating entities on the fly.

Regarding the stop when you call the service, can you make sure that your code is not passing thought [this line](https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/44c57929fd23e54379654a5b434b1c08e085bbfb/src/c/profile/shared_memory/shared_memory.c#L179)?

bjv-capra, 2021-05-19:
Unfortunately, I can't have a single thread creating all the entities (it would take too much re-writing), but I can make sure they're run sequentially when being created.

On the other hand! I must have now rolled back to the previous version of the library I had built because I noticed I didn't have that line of code. I now updated it and AFAIK (with the little test I have done), it does indeed work. So, that bug was definitely part of the issue.

closed 2021-05-19 by bjv-capra

bjv-capra, 2021-05-19:
So, thanks a lot for the fix and attention :). Looking forward to finding more bugs :+1:

pablogs9, 2021-05-19:
Thanks to you @bjv-capra! Keep us updated about your progress with micro-ROS.