micro-ROS/micro_ros_setup #317: QoS policy durability options

anastasiapan, 2021-05-17:
## Qos options

- Hardware description: STM32F7
- RTOS: FreeRTOS
- Installation type: static lib
- Version or commit hash: foxy

I have a subscriber of a custom type message which works fine. My problem is that I want to change the quality of service durability from transient local to volatile. I am trying to change it like this:
```
rcl_subscription_options_t qos_options = rcl_subscription_get_default_options();
qos_options.qos.durability = RMW_QOS_POLICY_DURABILITY_VOLATILE;
 rcl_ret_t rc = rcl_subscription_init(
    &subscriber_, &node_, ROSIDL_GET_MSG_TYPE_SUPPORT(my_pkg, msg, my_msg),
    "topic_name", &qos_options);
```
But in practice after I program my board the durability is still transient local. When I restart the board it just subscribes to the last published message. This is the result from `ros2 topic info --verbose`:

```
Node name: RTPSParticipant
Node namespace: /
Topic type: my_pkg/msg/my_msg
Endpoint type: SUBSCRIPTION
GID: 01.0f.98.08.2d.00.00.00.0a.00.00.00.00.00.01.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RMW_QOS_POLICY_RELIABILITY_RELIABLE
  Durability: RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL
  Lifespan: 2147483651294967295 nanoseconds
  Deadline: 2147483651294967295 nanoseconds
  Liveliness: RMW_QOS_POLICY_LIVELINESS_AUTOMATIC
  Liveliness lease duration: 2147483651294967295 nanoseconds
```
I have seen this https://micro.ros.org/docs/overview/ROS_2_feature_comparison/  but it is not clear to me in the QoS settings section which durability policies are supported.

pablogs9, 2021-05-17:
Let me check if `durability` is implemented in micro-ROS. If not we will try to implement it ASAP.

pablogs9, 2021-05-17:
Hello @anastasiapan,

We have implemented this feature:
 - https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/241
 - https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/248
 - https://github.com/micro-ROS/rmw-microxrcedds/pull/133

You need to pull the XRCE DDS Client and `rmw_microxrcedds` in order to obtain the new changes. Also, it is required to rebuild the micro-ROS Agent, if you are using it from docker, just pull the newest docker image with:

```
docker pull microros/micro-ros-agent:foxy
```

I'm closing since this is solved, but do not hesitate to reopen if you try it and find any problem.

Thanks!

closed 2021-05-17 by pablogs9

anastasiapan, 2021-05-17:
@pablogs9 I can verify that it works. Thank you!