micro-ROS/micro_ros_setup #322: ROS2 foxy can't receive microros app string topic

Needrom, 2021-05-25:
## Issue template

- Hardware description: stm32f407
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
make a string message and send it to ros2

#### Expected behavior
ros2 node can receive the message

#### Actual behavior
ros2 print nothing but the ```ros2 topic echo``` can.

#### Additional information
[image]

[image]

pablogs9, 2021-05-25:
We need the code of both ROS 2 and micro-ROS app to check what is happening...

Needrom, 2021-05-25:
microros app is here [https://github.com/Needrom/Microros-on-DIscovery-board/tree/main/freertos_apps/apps/motor_subscriber](url)

sending with /test_string topic

and ros2 app is here
```
#include<memory>

#include"rclcpp/rclcpp.hpp"
#include"std_msgs/msg/string.hpp"

using std::placeholders::_1;

class MinimalSubscriber : public rclcpp::Node{
public:
	MinimalSubscriber()
	: Node("minimal_subscriber")
	{
		subscription_ = this->create_subscription<std_msgs::msg::String>(
		"test_string", 120, std::bind(&MinimalSubscriber::topic_callback, this, 		_1));
	}

private:
	void topic_callback(const std_msgs::msg::String::SharedPtr msg) const
	{
//        printf("%s line:%d \r\n", __func__, __LINE__);
	  RCLCPP_INFO(this->get_logger(), "I heard: '%s'", msg->data.c_str());
	}
	rclcpp::Subscription<std_msgs::msg::String>::SharedPtr subscription_;
};

int main(int argc, char *argv[]){
    rclcpp::init(argc, argv);
    rclcpp::spin(std::make_shared<MinimalSubscriber>());
    rclcpp::shutdown();
    return 0;
}
```

pablogs9, 2021-05-25:
In micro-ROS you are creating a best-effort publisher and in ROS 2 you are creating a reliable subscriber, they won't match.

You have an example on how to create a best-effort subscriber in RCLCPP [here](https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/topics/listener_best_effort.cpp) and [this issue] is related(https://github.com/micro-ROS/micro_ros_setup/issues/236).

Needrom, 2021-05-26:
yes, it's.

closed 2021-05-26 by Needrom