micro-ROS/micro_ros_setup #329: could't build with configure serial type udp

Needrom, 2021-05-28:
## Issue template

- Hardware description: https://github.com/Needrom/Microros-on-DIscovery-board
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
ros2 run micro_ros_setup configure_firmware.sh motor_subscriber -t udp -i 192.168.100.233 -p 8888

ros2 run micro_ros_setup build_firmware.sh 

#### Expected behavior
could build the firmware.

#### Actual behavior

```
--- stderr: microxrcedds_client                                          
In file included from /home/mego/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/transport.h:32,
                 from /home/mego/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/util/ping.h:24,
                 from /home/mego/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/util/ping.c:1:
/home/mego/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/ip/udp/udp_transport.h:33:27: error: field 'platform' has incomplete type
     struct uxrUDPPlatform platform;
                           ^~~~~~~~
make[3]: *** [CMakeFiles/microxrcedds_client.dir/build.make:261: CMakeFiles/microxrcedds_client.dir/src/c/util/ping.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:79: CMakeFiles/microxrcedds_client.dir/all] Error 2
make[1]: *** [Makefile:133: all] Error 2
---
Failed   <<< microxrcedds_client [1.76s, exited with code 2]
Aborted  <<< tinydir_vendor [1min 45s]                                         
                                       
Summary: 5 packages finished [1min 47s]
  1 package failed: microxrcedds_client
  1 package aborted: tinydir_vendor
  2 packages had stderr output: microxrcedds_client tinydir_vendor
  53 packages not processed
make: *** [Makefile:352: colcon_compile] Error 2

```

pablogs9, 2021-05-28:
This board does not have support for UDP since it is a custom board. Please if you have troubles porting UDP transports to your board, open an issue asking your doubts.

closed 2021-05-28 by pablogs9

Needrom, 2021-05-28:
I can use network on my board but I don't know how to configure it with dds