micro-ROS/micro_ros_setup #337: Support for ST-Link v2 with flash script

Richard-Haes-Ellis, 2021-06-14:
## Issue 

Are there any plans to add support for the ST-Link V2 programmer on the flash utility?
How can I flash the Olimex board without using the command `    ros2 run micro_ros_setup flash_firmware.sh` from the tutorials?  

- Hardware description: Olimex stm32-e407
- RTOS: NuttX
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
while st-link v2 is plugged in
ros2 run micro_ros_setup flash_firmware.sh 

#### Expected behavior
Successfull upload with ST-link v2

#### Actual behavior
Flashing firmware for nuttx platform olimex-stm32-e407
Flashing firmware for nuttx platform olimex-stm32-e407
Error. Unsuported OpenOCD USB programmer

#### Additional information

pablogs9, 2021-06-14:
Hello, there are no plans in our roadmap for implementing this feature. But I think that looking into the other flashing script should be easy to implement this.

Could you consider contributing with a PR?

closed 2021-06-14 by pablogs9

Richard-Haes-Ellis, 2021-06-15:
Sure will work on it this weekend. 👍

pablogs9, 2021-06-15:
Thanks a lot!