micro-ROS/micro_ros_setup #341: Examples fail while using UDP transport with FreeRTOS and Olimex

Richard-Haes-Ellis, 2021-06-23:
## Issue 
- Hardware description: Olimex-stm32-e407
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue

ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407
ros2 run micro_ros_setup configure_firmware.sh int32_subscriber --transport udp --ip 192.168.1.3 --port 8888

I've also tried ip: 192.168.1.31 

ros2 run micro_ros_setup build_firmware.sh
ros2 run micro_ros_setup flash_firmware.sh

#### Expected behavior

No error on UART3 from olimex

#### Actual behavior

[image]

#### Additional information

I've tried ping_pong and int32_subsriber examples and both fail at the same line:

`// create init_options`
`RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));`

pablogs9, 2021-06-24:
I see that your board is using `192.168.1.3` and you are configuring the micro-ROS client to use the same IP as micro-ROS Agent IP. When you execute the `configure_firmware` step you should write the IP where the agent is running:

```bash
ros2 run micro_ros_setup configure_firmware.sh int32_subscriber --transport udp --ip [MICRO-ROS AGENT IP] --port 8888
```

How are you running the agent?

Richard-Haes-Ellis, 2021-06-24:
Thats right, I misread the docs and thought it was to assign the board an IP, thanks!!

closed 2021-06-24 by Richard-Haes-Ellis