micro-ROS/micro_ros_setup #342: MicroROS agent building failing

Vishnu-Vijay95, 2021-06-24:
## Issue template

- Hardware description: STM32E407
- RTOS: FreeRTOS
- Installation type:micro_ros_setup
- Version of micro-ROS used: foxy

#### Steps to reproduce the issue
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup build_agent.sh

#### Expected behavior
Expected to build the micro-ROS agent
#### Actual behavior
[image]

#### Additional information

pablogs9, 2021-06-24:
I have replicated the procedure and it works as expected in a clean ROS 2 Foxy docker: https://asciinema.org/a/QVsUwlBRDHHyIGtuBPxQ0N6NC

Can you retry in a clean environment?

Vishnu-Vijay95, 2021-06-24:
I tried several times, no change. Actually i had added custom message in uros folder, but the custom message is building but rmw_microxrcedds is always making build failing. Normally this issue get resolved by rebuilding.

Vishnu-Vijay95, 2021-06-24:
otherwise @pablogs9 can you tell me how to include navigation message (nav_msgs) type in micro ROS, i searched a lot for the message datatype and header file.

pablogs9, 2021-06-24:
Regarding the agent, I have prepared [this PR](https://github.com/micro-ROS/micro_ros_setup/pull/343), when merged please download again the `micro_ros_setup` repo and try to build the agent.

Regarding the `nav_msg` you have the headers in: `/ws/firmware/mcu_ws/install/nav_msgs/include/nav_msgs`. So for example you can include:

```c
#include <nav_msgs/msg/path.h>
```

pablogs9, 2021-06-24:
Closing since https://github.com/micro-ROS/micro_ros_setup/pull/343 merged

pablogs9, 2021-06-24:
Please reopen if you still have the issue

closed 2021-06-24 by pablogs9

Vishnu-Vijay95, 2021-06-29:
Thanks @pablogs9, sorry for late reply, had some health issues. I have tested your kobuki micro ROS demo on ESP32. There you are publishing robot odometry data. Can you explain how you are assessing the right and left encoder values from the kobuki base.