micro-ROS/micro_ros_setup #347: Fail to call `rcl_subscription_fini`

bjv-capra, 2021-06-29:
## Issue template

- Hardware description: STM32F7
- RTOS: FreeRTOS
- Installation type: Static lib
- Version or commit hash: foxy
- Multithread support

Hi, I have 4 nodes that have been initialized and work perfectly fine across different threads. When I notice the agent is down, I then try to proceed to shutdown all the nodes to release all the memory from entities. The `support` is the last thing to be released.

On the first node, my call to `rcl_subscription_fini` returns a non 0 code.
Tracing it, the point where it fails is https://github.com/micro-ROS/rmw-microxrcedds/blob/foxy/rmw_microxrcedds_c/src/rmw_subscription.c#L343 that returns 1 when called.

My "cleanup" looks like this.

```
  auto result = rclc_executor_fini(&executor_);
  result |= rcl_subscription_fini(&subscriber_1_, &node_);
  result |= rcl_subscription_fini(&subscriber_2_, &node_);
  result |= rcl_node_fini(&node_);
```

`rclc_executor_fini` returns correctly

pablogs9, 2021-06-29:
Ok, that's normal because you are trying to remove a ROS 2 entity from an agent that is no longer available. 

Maybe we should return an `RCL_RET_TIMEOUT` instead of `RMW_RET_ERROR` in the case that the ROS 2 entity is removed from client memory but not safely removed from the agent. What do you think?

bjv-capra, 2021-06-29:
Well, the question is, did it release the memory for the entities thus far?

AFAIK, when the agent is down, I need to stop everything, including the support, that's why I do it. Is this correct? If so, I want to release all the memory as soon as the agent is down, however, this is not possible because of needing the agent to release it? Or am I confusing something there?

pablogs9, 2021-06-29:
The point is:
- the client creates entities in the agent. For doing that the client "takes" a memory slot in the RMW and the agent creates the ROS2-DDS entity using dynamic memory.
- when you call `_fini` functions you telling the client to destroy the ROS2-DDS entities **in the agent**. Also to free the memory used in the client RMW.

But RCL/RCLC does not have the concept of "delete entities in the agent", so our only option is to return:
- `RCL_RET_OK`: if the entity has been destroyed in the agent and in the client
- `RCL_RET_TIMEOUT`: if the entity has been destroyed in the agent and in the client
- `RCL_RET_ERROR`: in other case

bjv-capra, 2021-06-29:
Btw, you typed the same for `RCL_RET_OK` and `RCL_RET_TIMEOUT`
I see, well, maybe my use case is wrong?

What I want is, given the two following cases:

1. Agent crashes (for some reason) - the link is up
2. Link goes down - the agent is still up

Ensure I can reconnect to the agent safely and all communication will resume once the connection is re-established.

As per my understanding, when the connection between agent and client has issues (1 or 2 above), I have to pretty much "restart" all my ROS entities. If this is so, what should be the procedure from the client's point of view. As it will be blind upon what happened to the agent.

One thing I do is to periodically ping the agent, now, when it reports an error, what should I do? Do I need to do anything?

bjv-capra, 2021-06-29:
Just to sum, I think it makes sense to go with what you proposed. That is:

> But RCL/RCLC does not have the concept of "delete entities in the agent", so our only option is to return:
> 
> * `RCL_RET_OK`: if the entity has been destroyed in the agent and in the client
> * `RCL_RET_TIMEOUT`: if the entity has been destroyed in the client but not in the agent
> * `RCL_RET_ERROR`: in other case

That is, in the case of the ping failing, I'd expect to receive `RCL_RET_TIMEOUT` when calling the `_fini` methods.

I think I'll always fall onto considering everything has crashed and I need to re-establish everything, as it's the safest approach, I have enough resources on the agent side to destroy and create the entities anyways.

pablogs9, 2021-06-30:
Hi, @bjv-capra check this [PR](https://github.com/micro-ROS/rmw-microxrcedds/pull/142), if you can cherrypick the patch to your branch and test the functionality in your use case will be appreciated.

Regarding the agent entity check, I have been thinking that maybe instead of using the XRCE `REPLACE` we can use `REUSE`. You can check the modes in [page 42 table 5 XRCE-DDS OMG standard](https://www.omg.org/spec/DDS-XRCE/1.0/PDF). For example, we use it [here](https://github.com/micro-ROS/rmw-microxrcedds/blob/afb51c7f01ca8a185f30e55ab219b98a5bc9d18a/rmw_microxrcedds_c/src/rmw_publisher.c#L223).

Now, if you create the same entity in the agent the older will be destroyed and a new one is created, this consumes resources in the agent machine. By using the `REUSE` flag the agent will just use the same DDS entity (maintaining the historic and all the status of the previous entity). 

What do you think about this?

PD: Remember that for creating a session and replacing/reusing the DDS entities in the agent you should [fix the XRCE client key](https://github.com/micro-ROS/micro-ROS-demos/blob/5ce76f941f830ba23b9b556cc3d7bcb10a958cfc/rclc/configuration_example/configured_publisher/main.c#L44) because if not, every time you init an XRCE the session key will be random and the agent will consider your client a completely new client.

PD2: It also can be interesting to check section 8.4.12 Getting Information on a Resource on the [OMG XRCE standard](https://www.omg.org/spec/DDS-XRCE/1.0/PDF). It is not implemented but can be interesting to add to the roadmap in order to have better control of this kind of situation. CC: @Acuadros95 @mamerlan

bjv-capra, 2021-06-30:
Hi @pablogs9. I'll see to cherry-pick it soon to test it. Thanks!

WRT to the `REUSE`  functionality, I believe that would be fine. Would you then set it to TRUE and TRUE? So, it will try to reuse, but if not, it will replace.

WRT to the agent key,
I actually don't have the key set, is there a way to set it up statically through the colcon.meta? Or only dynamically?

WRT to the 8.4.12 section. I don't see the specific use case right now, as it needs to query the agent.

pablogs9, 2021-06-30:
> Hi @pablogs9. I'll see to cherry-pick it soon to test it. Thanks!

Let us know when you test it and I'll merge

> WRT to the REUSE functionality, I believe that would be fine. Would you then set it to TRUE and TRUE? So, it will try to reuse, but if not, it will replace.

That's correct, I need to make a couple of checks and I'll create the PR.

pablogs9, 2021-06-30:
PR for `REUSE` and `REPLACE`: https://github.com/micro-ROS/rmw-microxrcedds/pull/143

> I actually don't have the key set, is there a way to set it up statically through the colcon.meta? Or only dynamically?

The client key should be set dynamically in the API because a micro-ROS client can handle multiple XRCE-DDS sessions (for example connecting to multiple agents).

bjv-capra, 2021-06-30:
@pablogs9 I checked out to that branch but when I build, I get this error on the `rmw`

```
microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_get_endpoint_network_flow.c:18:10: fatal error: rmw/get_network_flow_endpoints.h: No such file or directory
   18 | #include <rmw/get_network_flow_endpoints.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:196: CMakeFiles/rmw_microxrcedds.dir/src/rmw_get_endpoint_network_flow.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_qos_profile_check_compatible.c:24:3: error: unknown type name 'rmw_qos_compatibility_type_t'
   24 |   rmw_qos_compatibility_type_t * compatibility,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rmw_microxrcedds.dir/build.make:209: CMakeFiles/rmw_microxrcedds.dir/src/rmw_qos_profile_check_compatible.c.obj] Error 1
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c: In function 'rmw_init':
/microros_ws/firmware/mcu_ws/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_init.c:169:10: error: 'rmw_context_t' {aka 'struct rmw_context_t'} has no member named 'actual_domain_id'
  169 |   context->actual_domain_id = options->domain_id;
```

If I build with foxy, there's no problem

pablogs9, 2021-06-30:
Well as you can see both PR are on top of `main`:

[image]

Try to cherry-pick commits on top of `foxy` (just adding the branch-specific commits on top of foxy branch), because you will not be able to build the whole RMW `main` branch on a `foxy` environment.

Let me know if you find any problem if you find it easier I can rebase the branches on top of `foxy` and the backport them to `main` and `galactic`

Sorry for that, but our procedure is always to develop on top of `main`(develop or rolling branch) and the backport to stable branches.

bjv-capra, 2021-06-30:
Ah, my bad, for some reason, I completely missed the "cherry picking" part. Be back :)

pablogs9, 2021-06-30:
BTW please provide feedback on both PR if possible.

bjv-capra, 2021-06-30:
I have tested the first PR with the fix on the RMW to provide a timeout, it seems to do what was intended, however, the rcl won't forward the timeout error, thus, I'll still see a  regular ERROR as the output of rcl_subscription_fini

bjv-capra, 2021-06-30:
For the latter PR, that needs to be done on the agent, correct?

pablogs9, 2021-06-30:
> I have tested the first PR with the fix on the RMW to provide a timeout, it seems to do what was intended, however, the rcl won't forward the timeout error, thus, I'll still see a regular ERROR as the output of rcl_subscription_fini

Can you open an issue in https://github.com/ros2/rcl and link to this one? Maybe we can ask if this is intended behavior or we can propose a patch for forwarding the timeout.

> For the latter PR, that needs to be done on the agent, correct?

Just try to fini and init entities in a for loop and paste here the agent output with `-v6`

bjv-capra, 2021-06-30:
```
[1625052324.013802] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1625052324.013997] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1625052330.088881] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 16, data: 
0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
[1625052330.089441] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x00000000, len: 36, data: 
0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
0020: 00 00 00 00
[1625052330.102257] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 16, data: 
0000: 80 00 00 00 02 01 08 00 00 09 FF FD 02 00 00 00
[1625052330.102532] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x00000000, len: 36, data: 
0000: 80 00 00 00 06 01 1C 00 00 09 FF FD 00 00 01 0D 58 52 43 45 01 00 01 0F 00 01 0D 00 01 00 00 00
0020: 00 00 00 00
[1625052330.106169] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F AB CD EF 01 81 00 FC 01
[1625052330.106255] info     | Root.cpp           | create_client            | create                 | client_key: 0xABCDEF01, session_id: 0x81
[1625052330.106298] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2882400001, address: 10.46.28.2:10549
[1625052330.106382] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1625052330.107360] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xABCDEF01, len: 16, data: 
0000: 81 00 00 00 0E 01 08 00 0B 00 00 00 40 F6 D3 04
[1625052330.107490] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 32, data: 
0000: 81 00 00 00 0F 01 18 00 AA 54 DC 60 C7 69 67 06 AA 54 DC 60 59 67 67 06 0B 00 00 00 40 F6 D3 04
[1625052330.109320] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xABCDEF01, len: 36, data: 
0000: 81 80 00 00 01 07 1C 00 00 0A 00 01 01 03 00 00 0D 00 00 00 00 01 A5 A5 05 00 00 00 62 61 73 65
0020: 00 00 00 00
[1625052330.112809] debug    | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xABCDEF01, participant_id: 0x000(1)
[1625052330.112897] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1625052330.112919] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1625052330.116620] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xABCDEF01, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1625052330.118317] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xABCDEF01, len: 104, data: 
0000: 81 80 01 00 01 07 5E 00 00 0B 00 02 02 03 00 00 50 00 00 00 28 00 00 00 72 74 2F 72 6F 62 6F 74
0020: 2F 62 61 73 65 2F 69 6E 74 65 72 6E 61 6C 2F 74 69 6D 65 2F 73 79 6E 63 68 72 6F 6E 69 7A 65 00
0040: 00 01 A5 A5 1C 00 00 00 73 74 64 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A 45 6D 70
0060: 74 79 5F 00 00 01 00 00
[1625052330.118410] debug    | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x000(2), participant_id: 0x000(1)
[1625052330.118454] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 00 00
[1625052330.118473] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1625052330.119492] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xABCDEF01, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1625052330.120192] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xABCDEF01, len: 24, data: 
0000: 81 80 02 00 01 07 10 00 00 0C 00 04 04 03 00 00 02 00 00 00 00 00 00 01
[1625052330.120259] debug    | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1625052330.120298] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 14, data: 
0000: 81 80 02 00 05 01 06 00 00 0C 00 04 00 00
[1625052330.120314] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 03 00 00 00 80
[1625052330.124334] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xABCDEF01, len: 13, data: 
[... cut ...]
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
[1625052490.827340] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0xABCDEF01, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 03 00 04 00 80
[1625052490.827392] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0xABCDEF01, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 05 00 00 00 80
```

Here's the output from running a loop creating a node and a subscription, then deleting them over and over.

pablogs9, 2021-06-30:
You can see how now the agent is printing `objects matched` instead of `object created` because it is reusing the entities, I'm going to merge the patch.

bjv-capra, 2021-06-30:
to be honest, I have no clue how to see that based on the log.

Should we leave this issue open until we get some reply from rcl?

pablogs9, 2022-01-28:
Can we close this?

bjv-capra, 2022-01-28:
We still don't get a reply from RCL, but yes, we can close it

closed 2022-01-28 by bjv-capra