micro-ROS/micro_ros_setup #348: Launch file terminates node with std::logic_error

Richard-Haes-Ellis, 2021-06-29:
## Issue template

- Hardware description: olimex-stm32-e407
- RTOS: FreeRTOS
- Installation type: micro_ros_setup 
- Version or commit hash:  foxy
-
#### Steps to reproduce the issue
ros2 launch micro_ros_agent micro_ros_agent_launch.py

#### Expected behavior

[image]

#### Actual behavior
```

[INFO] [launch]: All log files can be found below /home/richard/.ros/log/2021-06-29-12-14-18-274708-richard-GL65-9SEK-67024
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [micro_ros_agent-1]: process started with pid [67026]
[micro_ros_agent-1] terminate called after throwing an instance of 'std::logic_error'
[micro_ros_agent-1]   what():  basic_string::_M_construct null not valid
[ERROR] [micro_ros_agent-1]: process has died [pid 67026, exit code -6, cmd '/home/richard/dev_ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent udp4 --port 8888 -v6 --ros-args -r __node:=micro_ros_agent'].
```

#### Additional information

launch.log file: 

```
1624961495.2396970 [INFO] [launch]: All log files can be found below /home/richard/.ros/log/2021-06-29-12-11-35-235700-richard-GL65-9SEK-66678
1624961495.2398911 [INFO] [launch]: Default logging verbosity is set to INFO
1624961495.3029954 [INFO] [micro_ros_agent-1]: process started with pid [66680]
1624961495.3071363 [micro_ros_agent-1] terminate called after throwing an instance of 'std::logic_error'
1624961495.3072314 [micro_ros_agent-1]   what():  basic_string::_M_construct null not valid
1624961495.4339836 [ERROR] [micro_ros_agent-1]: process has died [pid 66680, exit code -6, cmd '/home/richard/dev_ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent udp4 --port 8888 -v6 --ros-args -r __node:=micro_ros_agent'].
```

pablogs9, 2021-06-29:
Use `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6`

Do you have any special interest in using a launch file?

Richard-Haes-Ellis, 2021-06-29:
Yes, it would be nice to be able to launch the node in a bigger launch file with other nodes that communicate with the micro-Ros node. 

Another issue I've been having is that the olimex board dosesent seem to connect if I power the board first and then run the agent, Any way to reset the board when agent is launched?

pablogs9, 2021-07-05:
Solved here: https://github.com/micro-ROS/micro-ROS-Agent/pull/78

closed 2021-07-05 by pablogs9

Richard-Haes-Ellis, 2021-07-05:
Excellent! Thanks :D