micro-ROS/micro_ros_setup #349: Node namespace missing

RussellJoyce, 2021-07-01:
- Hardware description: ESP32 / Ubuntu 20.04
- RTOS: FreeRTOS
- Installation type: micro_ros_setup / ESP-IDF component
- Version or commit hash: foxy

#### Steps to reproduce the issue

When adding a namespace to a node in RCL, the resulting namespace is only applied to topics associated with that node, and not the node itself.

#### Expected behavior

Using the following code, the _freertos_int32_publisher_ node should be created as `/test_namespace/freertos_int32_publisher`, and the publisher should publish to the `/test_namespace/freertos_int32_publisher` topic.

```
rclc_node_init_default(&node, "freertos_int32_publisher", "test_namespace", &support);
rclc_publisher_init_default(&publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32), "freertos_int32_publisher");
```

#### Actual behavior

The _freertos_int32_publisher_ node has no namespace set on it, but the topic does have the namespace.

#### Additional information

Screenshot showing the resulting rqt_graph output from the above code:
[image]

pablogs9, 2021-07-01:
Working on it here: https://github.com/micro-ROS/rmw-microxrcedds/pull/148

pablogs9, 2021-07-01:
Is it possible for you to rebase `fix/add_namespace_node` branch onto foxy and test these changes on your setup?

Just do it inside the `rmw-microxrcedds` in your IDF component and rebuild the whole library. We are quite busy right now and I will no have time to test this week.

RussellJoyce, 2021-07-01:
Yes, I can give that a test sometime today or tomorrow. Thanks!

pablogs9, 2021-07-02:
We have tested it, I'm closing since resolved. Related PR: https://github.com/micro-ROS/rmw-microxrcedds/pull/148 and https://github.com/micro-ROS/micro-ROS-Agent/pull/72

Please provide feedback and reopen if you find any problem.

closed 2021-07-02 by pablogs9