micro-ROS/micro_ros_setup #350: Multiple firmwares in a single ROS2 workspace

dayjaby, 2021-07-13:
Hey,

at the moment I build some custom firmware with the commands:
```
ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_f446re
ros2 run micro_ros_setup configure_firmware.sh foxtech_eh2000 --transport serial
ros2 run micro_ros_setup build_firmware.sh
```

So we have the simple structure:

```
./firmware/APP
./firmware/PLATFORM
./firmware/TRANSPORT
./firmware/build/zephyr/zephyr.elf
./firmware/zephyr_apps/apps/foxtech_eh2000
...
```

However, this does not allow to create multiple firmwares, where each firmware could be built with a different OS/board/transport/app.

I would prefer a structure like this:
```
ros2 run micro_ros_setup create_firmware_ws.sh
ros2 run micro_ros_setup create_firmware.sh eh2000 --os zephyr --board nucleo_f446re
ros2 run micro_ros_setup configure_firmware.sh eh2000 --app foxtech_eh2000 --transport serial
ros2 run micro_ros_setup build_firmware.sh eh2000
ros2 run micro_ros_setup flash_firmware.sh eh2000

ros2 run micro_ros_setup create_firmware.sh second_firmware --os zephyr --board nucleo_f401rc
ros2 run micro_ros_setup configure_firmware.sh second_firmware --app ping_pong --transport serial
ros2 run micro_ros_setup build_firmware.sh second_firmware
ros2 run micro_ros_setup flash_firmware.sh second_firmware
```

->

```
./firmware/firmwares/eh2000/APP
./firmware/firmwares/eh2000/PLATFORM
./firmware/firmwares/eh2000/TRANSPORT
./firmware/firmwares/eh2000/build/zephyr/zephyr.elf
./firmware/firmwares/second_firmware/APP
./firmware/firmwares/second_firmware/PLATFORM
./firmware/firmwares/second_firmware/TRANSPORT
./firmware/firmwares/second_firmware/build/zephyr/zephyr.elf
./firmware/zephyr_apps/apps/foxtech_eh2000
./firmware/zephyr_apps/apps/ping_pong
```

Is that a sane request? Do you have other ideas?

Cheers
David

pablogs9, 2021-07-13:
Hello @dayjaby, this proposal sounds interesting. Would you mind opening a PR with the required modifications so we can discuss it?

Preferably this should be a side feature: everything should work with the same commands interface that we currently have but the functionality can be enabled using, for example, an environment variable o an extra argument.

Thanks!

Kay-Anonymous, 2022-07-23:
Hello gurus,
It looks like I've managed something similar in some previous projects, and I thought it might be possible to add a common directory, so that code that is needed on different platforms at the same time only needs to be cloned once.

```
./firmware/firmwares/${OS}/public
# such like
./firmware/firmwares/${OS}/public/common_interfaces
./firmware/firmwares/${OS}/public/rcl_interfaces
# private repo
./firmware/firmwares/${OS}/${platform or hardwave}/private
./firmware/firmwares/${OS}/${platform or hardwave}/private/APP
./firmware/firmwares/${OS}/${platform or hardwave}/private/PLATFORM
./firmware/firmwares/${OS}/${platform or hardwave}/private/TRANSPORT

...

./firmware/zephyr_apps/apps/foxtech_eh2000
./firmware/zephyr_apps/apps/ping_pong
```

and also can import some common repos configuration files.

What does this proposal look like？

Best wishes~

pablogs9, 2022-07-26:
@Kay-Anonymous please open a PR