micro-ROS/micro_ros_setup #352: ros2 node list doesn't show Micro-ROS node (but ros2 echo does echo msgs)

gavanderhoorn, 2021-07-13:
Subject describes the issue pretty well :)

Unfortunately this is a custom port on a proprietary system with a very old compiler requiring a few changes to the sources (which may not all be too *nice*).

I can't really describe steps to reproduce it (as the platform is not available), nor provide any code unfortunately. The demo application is basically [this one](https://github.com/micro-ROS/micro_ros_raspberrypi_pico_sdk/blob/6413343f37a98e5831cb9bb75ab50496331fb049/pico_micro_ros_example.c) (minus the custom transports and GPIO setup+toggling).

That doesn't really help making it easy to diagnose what's going on, I fully understand, but I'm hoping there are some general things to check when this happens.

What puzzles me is that `ros2 node list` does not show a node, but `ros2 topic echo /my_topic` does work (in this case it `echo`s the `Int32`s the Micro-ROS node is publishing).

Is there anything on the Agent side I could enable to debug this?

The DDS-XRCE and RTPS traffic (captured by Wireshark) looks "normal" as far as I can tell. I see the node name sent around as the `entityName` a few times.

I'm on `develop` for XRCE-DDS-Client and Micro-CDR, `foxy` for the rest of the repositories. Nothing has complained, although I did run into https://github.com/micro-ROS/micro_ros_setup/issues/314. With the `eprosima/micro-xrce-dds-agent:latest` things immediately worked.

gavanderhoorn, 2021-07-13:
Oh, and to not just complain: :+1: for the code in the various Micro-ROS packages. I expected many more problems and things to port, but in the end the number of patches was fairly minimal.

Being able to start thinking about the application layer already is a rather nice position to be in. So thanks again.

pablogs9, 2021-07-14:
Hello @gavanderhoorn ROS2 handles the graph using a DDS topic named `ros_discovery_info`. With this topic the ROS 2 nodes associate DDS participant to ROS 2 nodes (because every RCLCPP context is ONE participant that can have multiple ROS 2 nodes).

Said that, micro-ROS does not have access to this level of DDS configuration, so we have created the micro-ROS graph manager. This graph manager is a plugin that the **micro-ROS Agent** offers on top of the **Micro XRCE-DDS Agent**. You can check it [here](https://github.com/micro-ROS/micro-ROS-Agent/blob/galactic/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp). This graph manager is in charge of advertising the micro-ROS nodes in this `ros_discovery_info` topic.

Also, we have been fixing some bugs recently on it, so, which version of the micro-ROS Agent are you using? Are you building it from scratch? are you using the docker version? I guess that your problem will be fixed by just updating your micro-ROS Agent to the latest commit because we have seen this behavior many times (echo working and node list not working).

Let me know if updating solves the issue.

gavanderhoorn, 2021-07-14:
Thanks for the reply.

re: ROS 2: that side I know about :) Thanks for clarifying the Micro-ROS side.

as to the Agent version: as I ran into #314, I'm currently using `eprosima/micro-xrce-dds-agent:latest`. `develop` branch for the Client and CDR.

IIUC, that should all work together, correct?

> we have created the micro-ROS graph manager. This graph manager is a plugin that the **micro-ROS Agent** offers on top of the **Micro XRCE-DDS Agent**. You can check it [...]

Ok, that's good to know. Does the Agent require any special configuration for this to work? Or should the Client / RMW be configured in a certain way? `RMW_UXRCE_GRAPH` is `OFF` at the moment, but I'm not sure whether that is related.

pablogs9, 2021-07-14:
The point is that the ROS 2 handling feature is only available in the [**micro-ROS Agent**](https://github.com/micro-ROS/micro-ROS-Agent) and not in the [**Micro XRCE-DDS Agent**](https://github.com/eProsima/Micro-XRCE-DDS-Agent), they are different packages. So if you are using the Micro XRCE-DDS Agent you will not have the ROS 2 graph handling feature.

Regarding the `RMW_UXRCE_GRAPH` it enables the feature of having the ROS 2 graph in the micro-ROS nodes. By default, the way micro-ROS to ROS 2 node discovery is enabled in the micro-ROS Agent and the way ROS 2 to micro-ROS should be enabled with `RMW_UXRCE_GRAPH` because it generates more resource usage in the embedded part. Usually if you don't need to know the ROS 2 graph in your embedded application keeping it disabled is ok.

gavanderhoorn, 2021-07-14:
> The point is that the ROS 2 graph handling feature is only available in the [**micro-ROS Agent**](https://github.com/micro-ROS/micro-ROS-Agent) and not in the [**Micro XRCE-DDS Agent**](https://github.com/eProsima/Micro-XRCE-DDS-Agent), they are different packages

Ah, I'd misread that. Thanks.

> it enables the feature of having the ROS 2 graph in the micro-ROS nodes. [..] Usually if you don't need to know the ROS 2 graph in your embedded application keeping it disabled is ok.

Thanks for the clarification again. The default makes sense to me.

pablogs9, 2021-07-14:
> Edit: btw, if this was an RTFM then please say so and point me to the docs. I'd much rather avoid bothering you with these types of questions.

Don't worry, the micro-ROS Graph manager feature does not have much documentation because it should work transparently to the user. Do not hesitate to ask whatever you need.

Thanks for the feedback!

Can we close this?

gavanderhoorn, 2021-07-14:
I'll have to test it, but we can close it.

I'll re-open if I can't get it to work.

gavanderhoorn, 2021-07-14:
>  the micro-ROS Graph manager feature does not have much documentation because it should work transparently to the user

Ok.

I was more thinking about *mentioning* the Micro-ROS Agent, as opposed to the Micro-XRCE-DDS Agent and the differences between them.

The Micro-XRCE-DDS documentation doesn't mention the Micro-ROS Agent, unless I'm mistaken.

closed 2021-07-14 by gavanderhoorn

pablogs9, 2021-07-14:
Well technically Micro-XRCE-DDS has nothing to do with ROS 2 or micro-ROS, that's why XRCE documentation does not mention ROS. Is micro-ROS who uses XRCE because:
- Micro XRCE-DDS Client is the default middleware of micro-ROS Client
- micro-ROS Agent uses the Micro XRCE-DDS Agent library along with other components.

gavanderhoorn, 2021-07-14:
> Well technically [..]

all true of course :)

However when searching for information on the XRCE-DDS Client, many results link to the Micro-XRCE-DDS documentation, which only mentions the client and the "non-ROS" agent.

The Micro-ROS tutorials do mention it of course.

Perhaps a single sentence saying something similar to what you wrote in https://github.com/micro-ROS/micro_ros_setup/issues/352#issuecomment-879676194, about them being different packages, would already help.

pablogs9, 2021-07-14:
Cool, we will take it into account!

Thanks for the feedback @gavanderhoorn

gavanderhoorn, 2021-07-14:
Just tested.

The node now shows up, the topic does as well, but now I can't `ros2 topic echo` it.

Running the `microros/micro-ros-agent` image with `ros:foxy` in another container (all `--net=host`).

I'm going to assume for now this is a problem in my custom combination of Micro-XRCE-DDS-Client and other packages.

pablogs9, 2021-07-15:
The `latest` is ROS 2 Rolling distribution, it is preferable to use the same ROS 2 distro in client and agent.