micro-ROS/micro_ros_setup #358: Unable to follow tutorial for Linux

gentov, 2021-07-19:
## Issue template

- Hardware description: Tutorial for Linux
- micro_ros_setup ver: 4c21641d57a6a018f55f1bb4c1361659668721cd

When I try and "ros2 run micro_ros_setup build_agent.sh" after following the previous steps in the tutorial for: 
"First micro-ROS Application on Linux"

I get the following output:
Finished <<< microxrcedds_agent [4min 2s]
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                                     
/usr/bin/ld: /home/gabriel/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `fmt::v6::internal::basic_data<void>::signs'
/usr/bin/ld: /home/gabriel/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `spdlog::logger::log_it_(spdlog::details::log_msg const&, bool, bool)'
/usr/bin/ld: /home/gabriel/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::snprintf_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/gabriel/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > fmt::v6::internal::grouping_impl<char>(fmt::v6::internal::locale_ref)'
/usr/bin/ld: /home/gabriel/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::format_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/gabriel/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::snprintf_float<long double>(long double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
/usr/bin/ld: /home/gabriel/microros_ws/install/microxrcedds_agent/lib/libmicroxrcedds_agent.so.2.0.0: undefined reference to `int fmt::v6::internal::format_float<double>(double, int, fmt::v6::internal::float_specs, fmt::v6::internal::buffer<char>&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:173: micro_ros_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< micro_ros_agent [19.1s, exited with code 2]

Summary: 3 packages finished [4min 27s]
  1 package failed: micro_ros_agent
  2 packages had stderr output: micro_ros_agent microxrcedds_agent

I have ROS2 installed, and can run the example publisher/subscriber (talker/listener) code after installation.

Any and all help is very much appreciated! Thank you!

Acuadros95, 2021-07-19:
Hello @gentov, which distribution of ROS2 are you using?

I just tested the tutorial on clean foxy and galactic installations and it worked as expected.

gentov, 2021-07-19:
@Acuadros95 After reinstalling Foxy, and trying to follow the tutorial, I get the same results. One thing is that during this step:

ros2 run micro_ros_setup build_firmware.sh

I get the following output:
Summary: 37 packages finished [30min 6s]
  3 packages had stderr output: micro_ros_demos_rclc micro_ros_demos_rclcpp rmw_microxrcedds

It seems like they are only warnings, not errors, but still worth mentioning. 

Thanks!

pablogs9, 2021-07-20:
Have you installed spdlog library? `sudo apt install libspdlog-dev`?

gentov, 2021-07-20:
Hi @pablogs9, yep:
libspdlog-dev is already the newest version (1:1.5.0-1)

Thanks for the suggestion, though

gentov, 2021-07-20:
@Acuadros95 just following up, thank you!

Acuadros95, 2021-07-21:
@gentov How are you installing ROS?

It is recomended to install it via Debian packages: https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html, with this approach you will need to install rosdep2 (`sudo apt install python3-rosdep2`) and colcon (`sudo apt install python3-colcon-common-extensions`)

Anyway, some separated ideas:
- Comment out line 149 `find_package(spdlog ${_spdlog_version} EXACT QUIET)` in `[your_micro-ROS_workspace]/src/eProsima/Micro-XRCE-DDS-Agent/cmake/SuperBuild.cmake` to force a build of spdlog and avoid conflicts with pre installed libraries.
- Disable the logger function adding `"-DUAGENT_LOGGER_PROFILE=OFF"` to `cmake-args` of the Micro-XRCE-DDS-Agent colcon.pkg `[your_micro-ROS_workspace]/src/eProsima/Micro-XRCE-DDS-Agent/colcon.pkg`

gentov, 2021-07-22:
@Acuadros95 Sorry for the late reply. I was installing ROS using the pre-built binaries. I will try to re-install via Debian packages and get back to you with updates. 

Thanks!

gentov, 2021-07-22:
Hi @Acuadros95 I installed ROS via Debian packages, and ran the demo code. I was able to get further than before, but, now I get this error, when calling 'ros2 run micro_ros_demos_rclc ping_pong':

> micro-ROS transport: connected using UDP mode, ip: '127.0.0.1', port: '8888'

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Handle's typesupport identifier (rosidl_typesupport_c) is not supported by this library
, at /tmp/binarydeb/ros-foxy-rosidl-typesupport-c-1.0.2/src/type_support_dispatch.hpp:116'

with this new error message:

  'Undefined type support, at /home/gabriel/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_publisher.c:126'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1626981565.385116751] [rclc]: [rclc_publisher_init_best_effort] Error in rcl_publisher_init: Undefined type support, at /home/gabriel/microros_ws/src/uros/rmw_microxrcedds/rmw_microxrcedds_c/src/rmw_publisher.c:126, at /tmp/binarydeb/ros-foxy-rcl-1.1.11/src/rcl/publisher.c:180

Failed status on line 89: 1. Aborting.

any ideas?

Thanks!

Acuadros95, 2021-07-23:
Did you source the installation with `source install/local_setup.bash` before executing the demo?

gentov, 2021-07-23:
Yep, I sourced both the ROS2 installation and microROS installation before running the node

Acuadros95, 2021-07-28:
I have replicated this error sourcing the micro-ROS installation before ROS2 setup, like this:
```
source install/local_setup.bash
source /opt/ros/$ROS_DISTRO/setup.bash
```

This results in the bug you are experiencing. It doesn't matter if you source it in the correct order after, the problem persists. 
So please make sure that you are in a clean environment and follow the expected order:
```
# ROS2 installation first
source /opt/ros/$ROS_DISTRO/setup.bash
source install/local_setup.bash

# Use RMW Micro XRCE-DDS implementation
export RMW_IMPLEMENTATION=rmw_microxrcedds

# Run a micro-ROS node
ros2 run micro_ros_demos_rclc ping_pong
```

gentov, 2021-07-30:
@Acuadros95 sorry I will not get to try this this week, but I will get back to you on Monday with how this went. Thank you for your help!

Acuadros95, 2021-12-21:
Closing due to inactivity, reopen if the problem persist.

closed 2021-12-21 by Acuadros95