micro-ROS/micro_ros_setup #359: Error while echoing topic

Vishnu-Vijay95, 2021-07-21:
## Issue template

- Hardware description: STM32E407
- RTOS: freeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: version of micro-ROS used: foxy
[image]

#### Actual behavior
Normally the error gets cleared by rebuilding the firmware. Now its not getting cleared.

pablogs9, 2021-07-21:
Can you share the code used?
Can you show the output of `ros2 topic list -t`?

Vishnu-Vijay95, 2021-07-21:
in the topic now am not publishing any thing right now, but expected the default values in echo
the output of ros2 topic list -t 
[image]

pablogs9, 2021-07-21:
Please share the code

Vishnu-Vijay95, 2021-07-22:
Here is the code
```
#include <stdio.h>
#include <unistd.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <std_msgs/msg/int32.h>
#include <sensor_msgs/msg/laser_scan.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include "stm32f4xx_hal.h"

#define SCAN_BUF_SIZE (size_t) 41

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

uint8_t data[100];
uint8_t wA[] ={0xA5,0x20};
rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;
UART_HandleTypeDef huart3;
sensor_msgs__msg__LaserScan lidar_data;
int i;

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
	RCLC_UNUSED(last_call_time);
	//HAL_UART_Receive(&huart3, data, 1, 1000);
	 /*if (data[0]== 0xFA)
	 {
	  	HAL_UART_Receive(&huart3, data, SCAN_BUF_SIZE, 1000); 
		if( data[0] >= 0xA0  && data[0] <= 0xDB) // TODO: checksum
      			{
				int degree_count_num = 0;
			        int index = (data[0] - 0xA0) * 6;  

				if( index == 0)
				{
					//printf("Got Scan index %i\n" , index);

					for(uint16_t j = 4; j < 40; j = j + 6)
        				{
          					uint8_t byte0 = data[j-1];
          					uint8_t byte1 = data[j+1-1];
          					uint8_t byte2 = data[j+2-1];
          					uint8_t byte3 = data[j+3-1];

					        lidar_data.intensities = (float)((byte1 << 8) + byte0);
				 	        lidar_data.ranges     = (float)((byte3 << 8) + byte2);

						//float fRange = lidar_data.ranges/ 1000.f;
						//float Intensity = lidar_data.intensities;
						
						//if( degree_count_num == 0)
					//printf("Got Scan index %i deg %i range %f | %f\n" , index , degree_count_num , fRange , fIntensity);

						degree_count_num++;
					}
				}
			}
		}*/
		
	if (timer != NULL) 
	{
		RCSOFTCHECK(rcl_publish(&publisher, &lidar_data, NULL));
		
	}
}

void appMain(void * arg)
{
	rcl_allocator_t allocator = rcl_get_default_allocator();
	rclc_support_t support;

	// create init_options
	RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

	// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "sensor_publisher", "", &support));

	// create publisher
	RCCHECK(rclc_publisher_init_default(
		&publisher,
		&node,
		ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, LaserScan),
		"sensor"));
	HAL_UART_Transmit(&huart3, wA, 2, 10);
	// create timer,
	rcl_timer_t timer;
	const unsigned int timer_timeout = 1000;
	RCCHECK(rclc_timer_init_default(
		&timer,
		&support,
		RCL_MS_TO_NS(timer_timeout),
		timer_callback));
	HAL_UART_Receive(&huart3, data, 100, 1000);
	for(i=0;i<100;i++)
	{printf("%X ",data[i]);}
	// create executor
	rclc_executor_t executor;
	RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
	RCCHECK(rclc_executor_add_timer(&executor, &timer));

	msg.data = 0;

	while(1){
		rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
		usleep(100000);
	}

	// free resources
	RCCHECK(rcl_publisher_fini(&publisher, &node))
	RCCHECK(rcl_node_fini(&node))

  	vTaskDelete(NULL);
}
```

pablogs9, 2021-07-22:
You are not allocating memory for `lidar_data`, please check read [this tutorial](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/).

Feel free to reopen if this does not work.

closed 2021-07-22 by pablogs9

Vishnu-Vijay95, 2021-07-22:
Let me check

Vishnu-Vijay95, 2021-07-22:
How to allocate memory for struct rosidl_runtime_c__float__Sequence. what will be the c datatype for the same? For rosidl_runtime_c__int32__Sequence is uint32_t.

pablogs9, 2021-07-22:
https://github.com/ros2/rosidl/blob/d1122a2d2cb0c2d048aad6453badc6bbd17800b5/rosidl_runtime_c/src/primitives_sequence_functions.c#L66

Vishnu-Vijay95, 2021-07-22:
Great thanks

Vishnu-Vijay95, 2021-07-22:
I tried with `float `but the error is not resolved.
```
idar_data.ranges = (float *) malloc(100 * sizeof(float));
lidar_data.intensities = (float*) malloc(100 * sizeof(float)); 
```
geting an error
```
error: incompatible types when assigning to type 'rosidl_runtime_c__float__Sequence' {aka 'struct rosidl_runtime_c__float__Sequence'} from type 'float *'
  lidar_data.ranges = (float *) malloc(100 * sizeof(float));
                    ^
```

pablogs9, 2021-07-22:
Your code is wrong. Please check read [this tutorial](https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/).

Vishnu-Vijay95, 2021-07-22:
```
      lidar_data.ranges.capacity= 100;
	lidar_data.ranges.data= (float*) malloc(lidar_data.ranges.capacity*sizeof(float));
	lidar_data.ranges.size=0;
```
this looks fine right?

pablogs9, 2021-07-22:
Yes

Vishnu-Vijay95, 2021-08-12:
Hi @pablogs9, I was working with the same application and the memory handling that you pointed out was rectified and the program was working perfectly for me and but sometimes I encounter a `topic not created` problem while echoing, is that any memory-related issue. I am attaching my program and agent output here, please go through and let me know the reason
```#include <stdio.h>
#include <unistd.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <std_msgs/msg/int32.h>
#include <sensor_msgs/msg/laser_scan.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include "stm32f4xx_hal.h"
#include "RPLidar.h"
#include <math.h>

//#define SCAN_BUF_SIZE (size_t) 41
#define PAYLOAD_SIZE 100
#define DEG2RAD(x) ((x)*M_PI/180.)
uint8_t data[100];
uint8_t wA[] ={0xA5,0x20};//data packet to enable scan
rcl_publisher_t publisher;
UART_HandleTypeDef huart3;
sensor_msgs__msg__LaserScan lidar_data;
//bool inverted = false;
//bool angle_compensate = true;
float max_distance = 8.0;
int angle_compensate_multiple = 1;
//struct RPLidarMeasurement nodes[360*8];
float angle_min = DEG2RAD(0.0f);
float angle_max = DEG2RAD(359.0f);
int i;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Aborting.\n",__LINE__,(int)temp_rc);vTaskDelete(NULL);}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){printf("Failed status on line %d: %d. Continuing.\n",__LINE__,(int)temp_rc);}}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
    RCLC_UNUSED(last_call_time);
    /*lidar_data.header.stamp = start;
    lidar_data.header.frame_id = frame_id;
    scan_count++;

    
    lidar_data.angle_min =  M_PI - angle_min;
    lidar_data.angle_max =  M_PI - angle_max;
    
    
    lidar_data.angle_increment = (lidar_data.angle_max - lidar_data.angle_min) / (double)(node_count-1);

    lidar_data.scan_time = scan_time;
    lidar_data.time_increment = scan_time / (double)(node_count-1);
    lidar_data.range_min = 0.15;
    lidar_data.range_max = max_distance;//8.0;

    lidar_data.intensities.resize(node_count);
    lidar_data.ranges.resize(node_count);
    
    
    for (size_t i = 0; i < node_count; i++)
         {
            float read_value = (float) nodes[i].dist_mm_q2/4.0f/1000;
            if (read_value == 0.0)
                lidar_data.ranges[i] = std::numeric_limits<float>::infinity();
            else
            {
                lidar_data.ranges[i] = read_value;
                lidar_data.intensities[i] = (float) (nodes[i].quality >> 2);
            }
                
        }*/
    
    for (size_t i = 0; i < PAYLOAD_SIZE; i++)
     {
         lidar_data.ranges.data[i] = (float)i;
         lidar_data.ranges.size++;
     }
    if (timer != NULL) 
    {
        RCSOFTCHECK(rcl_publish(&publisher, &lidar_data, NULL));
        
    }
}

void appMain(void * arg)
{
    rcl_allocator_t allocator = rcl_get_default_allocator();
    rclc_support_t support;
    
        // create init_options
    RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

    // create node
    rcl_node_t node;
    RCCHECK(rclc_node_init_default(&node, "sensor_publisher", "", &support));

    // create publisher
    RCCHECK(rclc_publisher_init_best_effort(
        &publisher,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(sensor_msgs, msg, LaserScan),
        "sensor"));
    
    const char HEADER_ID[] = "LZR_data";
    float ranges[PAYLOAD_SIZE];
    
    sensor_msgs__msg__LaserScan__init(&lidar_data); // important!!!
    lidar_data.header.frame_id.data = HEADER_ID;
    lidar_data.header.frame_id.size = sizeof(HEADER_ID);
    lidar_data.header.frame_id.capacity = sizeof(HEADER_ID);
    lidar_data.ranges.data = (float *)calloc(PAYLOAD_SIZE, sizeof(float));
    lidar_data.ranges.capacity = PAYLOAD_SIZE;
    lidar_data.ranges.size = 0;
    lidar_data.ranges.data = &ranges;

    HAL_UART_Transmit(&huart3, wA, 2, 10);// transmiting comand packet to start scan
    
    // create timer,
    rcl_timer_t timer;
    const unsigned int timer_timeout = 1000;
    RCCHECK(rclc_timer_init_default(
        &timer,
        &support,
        RCL_MS_TO_NS(timer_timeout),
        timer_callback));
    
    HAL_UART_Receive(&huart3, data, 100, 1000);
    for(i=0;i<100;i++)
    {printf("%X ",data[i]);}
    
    // create executor
    rclc_executor_t executor;
    RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
    RCCHECK(rclc_executor_add_timer(&executor, &timer));

    
    while(1){
        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
        usleep(100000);
    }

    // free resources
    //sensor_msgs__msg__LaserScan__fini(&lidar_data);
    RCCHECK(rcl_publisher_fini(&publisher, &node))
    RCCHECK(rcl_node_fini(&node))

    vTaskDelete(NULL);
}```

And the agent output
![Screenshot from 2021-08-12 13-09-31](https://user-images.githubusercontent.com/70762767/129158758-15d08505-5b0b-467f-8288-fe054ed31708.png)

`

Vishnu-Vijay95, 2021-08-12:
agent output
[image]

pablogs9, 2021-08-12:
Please copy-paste the whole micro-ROS Agent output

Vishnu-Vijay95, 2021-08-12:
```
ignitarium@ignitarium-Vostro-15-3568:~/Desktop/freeRTOS$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
[1628753402.507527] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1628753402.507751] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1628753403.304716] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 2E 73 F5 84 81 00 FC 01
[1628753403.304901] info     | Root.cpp           | create_client            | create                 | client_key: 0x2E73F584, session_id: 0x81
[1628753403.304969] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x2E73F584, address: 192.168.0.104:12020
[1628753403.305053] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628753404.144401] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 2E 73 F5 84 81 00 FC 01
[1628753404.144508] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x2E73F584, address: 192.168.0.104:12020
[1628753404.144635] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628753404.402066] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 48, data: 
0000: 81 80 00 00 01 05 28 00 00 0A 00 01 01 03 00 00 19 00 00 00 00 01 02 08 11 00 00 00 73 65 6E 73
0020: 6F 72 5F 70 75 62 6C 69 73 68 65 72 00 00 00 00
[1628753404.416566] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x2E73F584, participant_id: 0x000(1)
[1628753404.416666] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1628753404.416694] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628753404.508919] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753404.585161] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753404.585351] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628753404.608418] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753404.608519] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628753404.709034] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753404.787244] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753404.787359] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628753404.809786] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753404.809893] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628753404.813807] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753404.813895] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628753404.817997] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753404.818127] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628753404.829163] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[... cut ...]
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753410.912386] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628753411.112508] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2E73F584, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80

```

as we can see that DataReader and topic is not getting created , but send-recv of the message is happening.

pablogs9, 2021-08-12:
It seems that your app is freezing at some point after creating the participant in the line `RCCHECK(rclc_node_init_default(&node, "sensor_publisher", "", &support));`

Can you check manually the output of this `rclc_node_init_default` call? maybe by printing somewhere the error code returned?

Have you modified stack size, heap size or colcon.meta of your application?

Vishnu-Vijay95, 2021-08-12:
I had increased the micro-ROS app size slightly. I will check that. I will the output of this statement as well. rclc_node_init_default(&node, "sensor_publisher", "", &support)); if this is the problem what may be a possible solution.
I never touched the colcon.meta file.

Vishnu-Vijay95, 2021-08-12:
Now it is working fine I had increased the micro-ROS app stack size. I had reversed to normal value now. the output of the agent is as follows.
```
ignitarium@ignitarium-Vostro-15-3568:~/Desktop/freeRTOS$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
[1628761880.663195] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1628761880.663426] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1628761882.321178] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 0A D9 43 C9 81 00 FC 01
[1628761882.321328] info     | Root.cpp           | create_client            | create                 | client_key: 0x0AD943C9, session_id: 0x81
[1628761882.321402] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x0AD943C9, address: 192.168.0.104:12020
[1628761882.321486] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0AD943C9, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1628761882.324025] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 48, data: 
0000: 81 80 00 00 01 05 28 00 00 0A 00 01 01 03 00 00 19 00 00 00 00 01 02 08 11 00 00 00 73 65 6E 73
0020: 6F 72 5F 70 75 62 6C 69 73 68 65 72 00 00 00 00
[1628761882.324109] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628761882.328048] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628761882.331050] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628761882.336285] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x0AD943C9, participant_id: 0x000(1)
[1628761882.336407] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0AD943C9, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1628761882.336435] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628761882.336447] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628761882.336460] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628761882.336474] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628761882.502154] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 00 00 00 00 80
[1628761882.502248] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628761882.502299] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 80, data: 
0000: 81 80 01 00 01 05 45 00 00 0B 00 02 02 03 00 00 37 00 00 00 0A 00 00 00 72 74 2F 73 65 6E 73 6F
0020: 72 00 00 01 23 00 00 00 73 65 6E 73 6F 72 5F 6D 73 67 73 3A 3A 6D 73 67 3A 3A 64 64 73 5F 3A 3A
0040: 4C 61 73 65 72 53 63 61 6E 5F 00 00 01 00 00 00
[1628761882.502349] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1628761882.502394] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1628761882.502396] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1628761882.502520] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1628761882.502590] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0AD943C9, topic_id: 0x000(2), participant_id: 0x000(1)
[1628761882.502626] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1628761882.502723] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x0AD943C9, len: 13, data: 
0000: 81 00 00 00 0B 01 05 00 01 00 01 00 80
[1628761882.502748] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x0AD943C9, len: 14, data: 
[... cut ...]
0120: 00 00 68 42 00 00 6C 42 00 00 70 42 00 00 74 42 00 00 78 42 00 00 7C 42 00 00 80 42 00 00 82 42
0140: 00 00 84 42 00 00 86 42 00 00 88 42 00 00 8A 42 00 00 8C 42 00 00 8E 42 00 00 90 42 00 00 92 42
0160: 00 00 94 42 00 00 96 42 00 00 98 42 00 00 9A 42 00 00 9C 42 00 00 9E 42 00 00 A0 42 00 00 A2 42
0180: 00 00 A4 42 00 00 A6 42 00 00 A8 42 00 00 AA 42 00 00 AC 42 00 00 AE 42 00 00 B0 42 00 00 B2 42
01A0: 00 00 B4 42 00 00 B6 42 00 00 B8 42 00 00 BA 42 00 00 BC 42 00 00 BE 42 00 00 C0 42 00 00 C2 42
01C0: 00 00 C4 42 00 00 C6 42 00 00 00 00
```

Vishnu-Vijay95, 2021-08-12:
As seen from the agent output the send-recv is getting freezer after a time gap.

pablogs9, 2021-08-12:
How much stack and heap are you configuring?

Vishnu-Vijay95, 2021-08-12:
I didn't touch the heap. changed only the stack size of the microROS app. Earlier I kept to 4*8000B. now I kept the default value of 4*6000 bytes.

pablogs9, 2021-08-12:
Try use a stack of at least 15000 B and if you are using FreeRTOS+TCP have a heap of at least 65000 B.

Vishnu-Vijay95, 2021-08-12:
ok i will try that