micro-ROS/micro_ros_setup #364: install/micro_ros_setup/lib/micro_ros_setup/yaml_filter.py

AIxWALL, 2021-07-26:
## Issue template

- Hardware description: 
- RTOS: 
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue

#### Expected behavior

#### Actual behavior

#### Additional information

pablogs9, 2021-07-26:
Which release of micro-ROS are you targeting? Galactic or Foxy?

AIxWALL, 2021-07-26:
foxy.I've tried many times and it's the same error

pablogs9, 2021-07-26:
Check this patch: https://github.com/micro-ROS/micro_ros_setup/pull/365

AIxWALL, 2021-07-26:
thanks.the error message disappears. it should download some packages from github,e.g. ament/ament_cmake,but it didn`t,i wonder why. 
also
when Configuring the firmware with the following command:
$ros2 run micro_ros_setup configure_firmware.sh ping_pong --transport udp --ip 127.0.0.1 --port 8888
the following error appears:
~/workspace/microros_ws_zephyr/firmware/zephyr_apps/apps: No such file or directory

pablogs9, 2021-07-26:
Which version of Python are you using?

AIxWALL, 2021-07-26:
Python 3.8.10 (default, Jun  2 2021, 10:49:15) 
[GCC 9.4.0] on linux

pablogs9, 2021-07-26:
I have had no problems replicating this: https://asciinema.org/a/v5U7tsx80e52owitp90kafZ9K

AIxWALL, 2021-07-26:
i appreciate it, and i will check

Acuadros95, 2022-07-26:
Closing, feel free to reopen on any update.

closed 2022-07-26 by Acuadros95