micro-ROS/micro_ros_setup #368: Error when run build_agent.sh

mate212, 2021-07-26:
## Issue template

Hello,

I'm currently using az ESP32 and ROS2 Galaxy, trying to complete First Steps with micro-ROS [https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/].
I could create a micro-ROS agent, but when I tried to build it, it failed with the following output:
 

```
ros2 run micro_ros_setup build_agent.sh
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Starting >>> microxrcedds_agent
--- stderr: microxrcedds_agent
make[2]: *** No rule to make target '/usr/lib64/libtinyxml2.so', needed by 'libmicroxrcedds_agent.so.2.0.0'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/microxrcedds_agent.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
---
Failed   <<< microxrcedds_agent [0.37s, exited with code 2]
Aborted  <<< micro_ros_msgs [0.54s]

Summary: 0 packages finished [0.85s]
  1 package failed: microxrcedds_agent
  1 package aborted: micro_ros_msgs
  2 packages had stderr output: micro_ros_msgs microxrcedds_agent
  1 package not processed
```

I tried to resolve this issue with #228, but it didn't help me out.

pablogs9, 2021-07-26:
Tyr to install `libtinyxml2` and rebuild the agent from scratch.

```bash
sudo apt-get install libtinyxml2-dev
```

Let me know if this solves the issue.

mate212, 2021-07-26:
It did not solved this issue.
Running the suggested command resulted this:

```
sudo apt-get install libtinyxml2-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libtinyxml2-dev is already the newest version (7.0.0+dfsg-1build1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
```
I tried to rebuild the agent anyway, but it resulted the same output.

pablogs9, 2021-07-26:
Which version of ROS2 are you using and which OS are you using?

mate212, 2021-07-26:
ROS2 Galactic and Ubuntu 20.04.2 LTS

pablogs9, 2021-07-26:
I have replicated this with your same OS and ROS 2 version and it works as expected:

https://asciinema.org/a/QtqtBIJYEBJRkOF5ooIvhw0NV

Make sure that you have a correct ROS 2 installation and that you are following the steps correctly.

mate212, 2021-07-27:
I reinstalled everything, but it resulted the same error.

pablogs9, 2021-07-28:
One idea, have you tried to install FastDDS RMW in your distribution? The agent uses some libraries of this RMW implementation, please use:

```bash
apt update
apt install ros-galactic-rmw-fastrtps-cpp
```

and rebuild from scratch the agent.

mate212, 2021-07-28:
Installed and rebuilt, but it didn't work.

pablogs9, 2021-07-28:
Could you provide a replication environment for this issue? Sorry, I'm not able to solve it if I can't replicate it.

mate212, 2021-08-05:
I reinstalled the whole system, and here is an error that may cause this issue:
 `--- stderr: micro_ros_msgs                                         
CMake Error at /home/mate/ros2_galactic/ros2-linux/share/fastrtps/cmake/fastrtps-config.cmake:50 (find_package):
  By not providing "Findfoonathan_memory.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "foonathan_memory", but CMake did not find one.

  Could not find a package configuration file provided by "foonathan_memory"
  with any of the following names:

    foonathan_memoryConfig.cmake
    foonathan_memory-config.cmake

  Add the installation prefix of "foonathan_memory" to CMAKE_PREFIX_PATH or
  set "foonathan_memory_DIR" to a directory containing one of the above
  files.  If "foonathan_memory" provides a separate development package or
  SDK, be sure it has been installed.
`

pablogs9, 2021-08-09:
Do you have ROS2 installed by packages or are you building it from sources?

mate212, 2021-08-16:
The problem is solved.
Installing ROS2 from binary [https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Binary.html](url) causes the error, while executing build_agent.sh. 
Installing from packages is recommended [https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html](url).

closed 2021-08-23 by pablogs9