micro-ROS/micro_ros_setup #369: dpkg-query: no packages found matching ros-dashing-ament-lint-common

syafiqsallehudin, 2021-08-03:
## Issue template

- Hardware description:  Jetson Nano
- OS: Ubuntu 18.04
- Installation type: micro ros
- Version or commit hash: dashing

I was following the installation steps but met with these error:  

```
caremnano2@caremnano2-desktop:~/microros_ws$ rosdep install --from-path src --ignore-src -y
dpkg-query: no packages found matching ros-dashing-ament-lint-common
dpkg-query: no packages found matching ros-dashing-ament-lint-auto
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_setup: No definition of [clang-tidy] for OS version [bionic]
```
## Note
Installation steps was from the following link:
https://micro.ros.org/docs/tutorials/core/first_application_linux/
I've check in `ros2 pkg list` and verify ament_lint_common & ament_lint_auto are available

Acuadros95, 2021-08-03:
I couldn't replicate your problem, but I think the problem is in this line:
`micro_ros_setup: No definition of [clang-tidy] for OS version [bionic]`

Can you try installing `clang-tidy` manually?
`sudo apt install clang-tidy`

syafiqsallehudin, 2021-08-03:
Hi Acuardris95, thank you for the suggestion, I have also tried to install `clang-tidy` using other method which results to this:
```
caremnano2@caremnano2-desktop:~$ sudo apt install clang-tidy
[sudo] password for caremnano2: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
clang-tidy is already the newest version (1:6.0-41~exp5~ubuntu1).
The following packages were automatically installed and are no longer required:
  fltk1.3-doc fluid gir1.2-gtk-2.0 hddtemp libassuan-dev libfltk-cairo1.3
  libfltk-forms1.3 libfltk1.3-dev libgpg-error-dev libgpgme-dev libgtk2.0-dev
  liblz4-dev libopencv-calib3d3.2 libopencv-features2d3.2 libopencv-flann3.2
  libopencv-highgui3.2 libopencv-ml3.2 libopencv-videoio3.2
  python-backports.functools-lru-cache python-cycler python-defusedxml
  python-empy python-gnupg python-matplotlib python-netifaces python-opengl
  python-paramiko python-psutil python-pycryptodome python-pydot python-pyqt5
  python-pyqt5.qtopengl python-pyqt5.qtsvg python-pyqt5.qtwebkit
  python-subprocess32 python-tz python-wxtools sbcl
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 6 not upgraded.
```
Also tried to redo the installation afterwards and the result is still the same.

syafiqsallehudin, 2021-08-04:
Problem solved. Found on the README, to install other prerequisites needed for building a firmware using this package using:

```
sudo apt install python-rosdep
```
and will results to this:

```
caremnano2@caremnano2-desktop:~/microros_ws$ rosdep install --from-path src --ignore-src -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_agent: Cannot locate rosdep definition for [microxrcedds_agent]
```
After rebuilding the `microxrcedds_agent` then it is all good. 

```
caremnano2@caremnano2-desktop:~/microros_ws/build$ rosdep install --from-path src --ignore-src -y
#All required rosdeps installed successfully
```
Thanks and sorry for the inconvenient.

closed 2021-08-04 by syafiqsallehudin