micro-ROS/micro_ros_setup #370: Reconnecting to agent

roni-kreinin, 2021-08-05:
- Hardware description: Olimex STM32-E407
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

I am running an app similar to https://github.com/micro-ROS/micro-ROS-demos/tree/foxy/rclc/ping_uros_agent with 1 publisher, 1 timer, and 1 subscriber. I am able to successfully ping the uros agent and use micro-ROS as expected. I am running `rclc_executor_spin_some` and periodically pinging the agent with my timer. If the ping ever fails, the application assumes that the agent is disconnected, stops the executor from spinning, and proceeds to fini the rcl and rclc components. After everything is deinitialized, the app then goes back to the start and attempts to ping the agent, and initialize again if ping is successful. 

The problem I am having is that the fini functions are failing for the publisher, subscriber, and node. This then results in the re-initialization failing afterwards. Am I missing some steps in recovering from the agent being terminated?

bjv-capra, 2021-08-06:
Hi, it feels like a similar issue to https://github.com/micro-ROS/micro_ros_setup/issues/347 . In our case, we simply "trust" it did `fini` correctly.

pablogs9, 2021-08-09:
As explained in the other issue, when the Agent is not available the micro-ROS Client cannot ensure that entities are being destroyed, so the micro-ROS Client middleware layer will return a timeout. The point is that the RCL layer translates this timeout into an error.

Do you have any problem creating the entities after calling the `fini` functions?

roni-kreinin, 2021-08-20:
Yes, if it does not have an error after a `fini` call, it will have an error when creating the entities again. My solution to this problem currently is to do a software reset of the MCU, and to wait for the agent to connect again.

closed 2021-08-20 by roni-kreinin