micro-ROS/micro_ros_setup #372: Destroying subscription makes the system stall - inconsistent

bjv-capra, 2021-08-11:
## Issue template

- Hardware description: STM32F7
- RTOS: FreeRTOS
- Installation type: micro_ros_setup static lib
- Version or commit hash: Foxy

#### Steps to reproduce the issue

I startup multiple nodes, 5 nodes with a total of 9 subscriptions and 2 publishers. 

Here's the output with verbose = 4 (just to show the entities)

<details>
<summary>Agent output verbose=4</summary>

```
[1628681388.923359] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1628681388.923548] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1628681490.346483] info     | Root.cpp           | create_client            | create                 | client_key: 0xABCDEF01, session_id: 0x81
[1628681490.346550] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0xABCDEF01, address: 10.46.28.2:10549
[1628681490.358778] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xABCDEF01, participant_id: 0x000(1)
[1628681490.363913] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x000(2), participant_id: 0x000(1)
[1628681490.368711] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1628681490.370975] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[1628681491.354895] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xABCDEF01, participant_id: 0x001(1)
[1628681491.357559] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x001(2), participant_id: 0x001(1)
[1628681491.359737] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xABCDEF01, publisher_id: 0x000(3), participant_id: 0x001(1)
[1628681491.362119] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xABCDEF01, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1628681492.355128] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xABCDEF01, participant_id: 0x002(1)
[1628681492.357713] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x002(2), participant_id: 0x002(1)
[1628681492.362563] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x001(4), participant_id: 0x002(1)
[1628681492.364798] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x001(6), subscriber_id: 0x001(4)
[1628681493.368678] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xABCDEF01, participant_id: 0x003(1)
[1628681493.371635] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x003(2), participant_id: 0x003(1)
[1628681493.377457] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0xABCDEF01, publisher_id: 0x001(3), participant_id: 0x003(1)
[1628681493.379910] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0xABCDEF01, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1628681493.381951] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x004(2), participant_id: 0x003(1)
[1628681493.383938] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x002(4), participant_id: 0x003(1)
[1628681493.386798] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x002(6), subscriber_id: 0x002(4)
[1628681493.389552] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x005(2), participant_id: 0x003(1)
[1628681493.391480] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x003(4), participant_id: 0x003(1)
[1628681493.393779] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x003(6), subscriber_id: 0x003(4)
[1628681493.396487] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x006(2), participant_id: 0x003(1)
[1628681493.398433] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x004(4), participant_id: 0x003(1)
[1628681493.400719] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x004(6), subscriber_id: 0x004(4)
[1628681493.403573] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x007(2), participant_id: 0x003(1)
[1628681493.405529] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x005(4), participant_id: 0x003(1)
[1628681493.407864] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x005(6), subscriber_id: 0x005(4)
[1628681493.410571] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x008(2), participant_id: 0x003(1)
[1628681493.412523] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x006(4), participant_id: 0x003(1)
[1628681493.414914] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x006(6), subscriber_id: 0x006(4)
[1628681494.359020] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0xABCDEF01, participant_id: 0x004(1)
[1628681494.465316] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x009(2), participant_id: 0x004(1)
[1628681494.467674] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x007(4), participant_id: 0x004(1)
[1628681494.470992] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x007(6), subscriber_id: 0x007(4)
[1628681494.473872] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0xABCDEF01, topic_id: 0x00A(2), participant_id: 0x004(1)
[1628681494.476533] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0xABCDEF01, subscriber_id: 0x008(4), participant_id: 0x004(1)
[1628681494.479143] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0xABCDEF01, datareader_id: 0x008(6), subscriber_id: 0x008(4)
```

</details>
I'm running the agent on my computer, and I simply kill the agent process.

#### Expected behavior

All fini methods can be done successfully.

#### Actual behavior

The first node that I try to stop, does successfully go through `rclc_executor_fini`, then it never goes out of rcl_subscription_fini(). I've noticed that it doesn't get past the `destroy_topic` method on rmw_subscription.c.

However, I've put breakpoints there, and it sometimes moves past it, but very seldomly.

My colcon.meta has the following configuration

<details>
<summary>rmw_microxrcedds meta</summary>

```
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=6",
                "-DRMW_UXRCE_MAX_PUBLISHERS=5",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=15",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=8",
                "-DRMW_UXRCE_TRANSPORT=udp",
                "-DRMW_UXRCE_DEFAULT_UDP_IP=10.46.28.3",
                "-DRMW_UXRCE_DEFAULT_UDP_PORT=8888",
            ]
        }
```
</details>

With 4 nodes, this is not the problem, and the node that was added is not the node that is failing, nor even has been yet processed.

bjv-capra, 2021-08-11:
rclc_executor_fini is called only once. Just before calling the failing fini_subscriber, and it has been working fine until now.

Each node runs on its own thread

One of the nodes simply takes care of pinging the agent to verify the status, besides acting as the single configuration entry point for executors and the rclc base. Each node that requires an executor has it.

When the agent dies, the node will report the failure and the shutdown sequence will begin. First, make sure all publishers are stopped, then, the fini methods are called.

The executor first, then the entities. lastly the node.

This has been working quite smooth, until now

Hope this helps as an idea

bjv-capra, 2021-08-11:
The node that takes care of checking connection and as "base" provider, is the first node to start and last one to be fini

pablogs9, 2021-08-12:
I'm wondering right now if `rmw_uros_ping_agent` is thread safe. @Acuadros95 any idea?

Where exactly does the execution stop inside `destroy_topic`? It is weird because there is no blocking call inside those functions.

bjv-capra, 2021-08-12:
There was a time when the ping was not thread-safe, but AFAIK, it was fixed a time ago.

Regarding where it stops, I think it was

```
    uint16_t delete_topic = uxr_buffer_delete_entity(
      &custom_node->context->session,
      *custom_node->context->creation_destroy_stream,
      topic->topic_id);
```

I followed a bit more, and it's when trying to acquire the lock (which goes with your thought) here https://github.com/eProsima/Micro-XRCE-DDS-Client/blob/foxy/src/c/core/session/common_create_entities.c#L27

pablogs9, 2021-08-12:
And who has the lock at this moment? Maybe we have found an interlocking issue.

bjv-capra, 2021-08-12:
So, this is weird,
I noticed that my Ethernet RX buffers were being filled up completely, so I increased this define (FreeRTOS+TCP) to 16 packets rather than 16. This alone, does not seem to make any change, rather than simply not filling up the buffers.
However, if I then change the reliability of the publisher (new topic on the fifth node) to best effort, rather than the default, it works fine and I can call the fini methods, and start it all over again with no problems. If I keep it on the default, it does not go past the lock.

pablogs9, 2021-08-12:
In proyects with FreeRTOS+TCP, we have found this problem with memory buffers and also with heap memory. 

Are you using FreeRTOS+TCP stack? If so, which [BufferAllocation](https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/portable/BufferManagement) are you using and how much heap do you have available?

Regarding the lock, don't you have a debugging enviroment where you can inspect the FreeRTOS tasks?

bjv-capra, 2021-08-13:
Sorry for the delayed answer,
Regarding the allocation, we use https://nadler.com/embedded/newlibAndFreeRTOS.html.
We have 512k Heap (enough), and we don't have any dynamic allocation happening on runtime.

Regarding the lock, we can inspect tasks, but maybe I'm not fully understanding what you want to check. I couldn't particularly check who owned the lock.

I'm closing this issue as it's been solved, however, it'd be interesting to know, Why did it work with best-effort and not reliable?

closed 2021-08-13 by bjv-capra