micro-ROS/micro_ros_setup #373: fail "run flash_firmware"  in case of NUCLEO-H743ZI2

Song-MS, 2021-08-24:
## Issue template

- Hardware description: NUCLEO-H743ZI2 
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: ROS2 foxy

#### Steps to reproduce the issue

my terminal while connecting the board on my computer(ubuntu 20.04, ROS2 foxy) 
s@s:~/microros_ws$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for freertos platform nucleo_f767zi
Flashing firmware for freertos platform nucleo_f767zi
/home/song/microros_ws/install/micro_ros_setup/config/freertos/nucleo_f767zi/flash.sh: line 10: [false: command not found
Error. Unsuported OpenOCD USB programmer

#### Expected behavior
I tried to flash before build-up as a tutorial :https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/
However, I just found my board is NUCLEO-H743ZI2 not NUCLEO-H743ZI.
I guess that the difference between these two is using slink-v2 or slink-v3.

#### Additional information
Would you let me know to figure out this problem ?

pablogs9, 2021-08-24:
Hello @Song-MS, which steps are you following for building the app? It seems that you are selecting `nucleo_f767zi` and you are using a Nucleo H743ZI2.

It seems that your programmer is not supported, Nucleo H743ZI2 is a community-supported board so maybe you have to tweak the [flash script](https://github.com/micro-ROS/micro_ros_setup/blob/f319da57e4d734f529ed00518c27f787f6bfac3b/config/freertos/nucleo_f767zi/flash.sh#L13) for adding your programmer.

Acuadros95, 2022-07-26:
Closing due to inactivity, feel free to reopen.

closed 2022-07-26 by Acuadros95