micro-ROS/micro_ros_setup #383: files to edit to achieve custom transport

jacopagno, 2021-09-03:
- Hardware description:  NUCLEO_F446RE 
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: ROS2 foxy

Hi!
I'm trying to use CAN bus for custom transport. I read  the tutorial about micro ros custom transport and this thread(#206) but i'm having problems to navigate the directory created from tutorial. For my understanding to obtain a custom transport you need to:

-   modify the transport in the board
in <work_dir>/firmware/freertos_apps/microros_nucleo_f446re_expansions , here you need to change in the ./core/src/microros_transports.c the transport method, add in my case can.c and can.h and modify the .ioc accordingly.

- change client transport
in <work_dir>/firmware/mcu_ws/eProsima change custom_transport.c/.h and in the CMakeLists.txt enable custom transport.

-change agent transport 
in <work_dir>/src/eProsoma/Micro-XRCEE-DDS-Agent change transport/custom/CustomAgent.cpp modify init/fini/recv_message/send_message.

Am i right or i need to change something else? And i need to create a main.c like the examples in the directories or it does everthing on his own?

Thanks for your help.

pablogs9, 2021-09-03:
This is the old way of modifiying transports, in order to apply a custom transport just:
 - [Enable it in the colcon meta client](https://github.com/micro-ROS/micro_ros_azure_rtos_app/blob/ac3124fef5c61ad6b2ad24cc44275ccea87b27bf/microros/colcon.meta#L29) 
 -  [Enable it in the colcon meta rmw](https://github.com/micro-ROS/micro_ros_azure_rtos_app/blob/ac3124fef5c61ad6b2ad24cc44275ccea87b27bf/microros/colcon.meta#L37) 
 - [Write your transport functions](https://github.com/micro-ROS/micro_ros_azure_rtos_app/blob/main/app/microros_netx_transports.c)
 - [Use the RMW API to set the transport callbacks at runtime](https://github.com/micro-ROS/micro_ros_azure_rtos_app/blob/ac3124fef5c61ad6b2ad24cc44275ccea87b27bf/app/main.c#L122)

Regarding the agent side, maybe @Acuadros95 can tell us some more details.

Acuadros95, 2021-09-03:
To implement your custom Agent follow the steps from the provided example: [link](https://github.com/eProsima/Micro-XRCE-DDS-Agent/blob/develop/examples/custom_agent/custom_agent.cpp)

You have to implemente the following function prototypes for your transport:
- `eprosima::uxr::CustomAgent::InitFunction`
- `eprosima::uxr::CustomAgent::FiniFunction`
- `eprosima::uxr::CustomAgent::RecvMsgFunction`
- `eprosima::uxr::CustomAgent::SendMsgFunction`

Then, define a custom endpoint to identify each client:
- `eprosima::uxr::CustomEndPoint`

With these definitions, you can generate your custom Agent using the constructor of `eprosima::uxr::CustomAgent custom_agent`.
All of these steps are present on the example main function, feel free to come back if any specific problem arises.

jacopagno, 2021-09-03:
Thank you very much for the fast answer, i try to implement all as you suggested. For now i'll close the thread.

jacopagno, 2021-09-10:
hi again, i'm having problems with the client. The function rclc_support_init stops the program. I read the thread  https://github.com/micro-ROS/micro_ros_setup/issues/281 and https://github.com/micro-ROS/micro_ros_setup/issues/277 but 
didn't find a solution. I think the problem could be in the colcon.meta:

```
  "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON"                
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=custom",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=1",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=1",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=2"
            ]
        },
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        }
    }
}
```

or the way i've build and flash the system. I've done it through STM32cubeIDE following https://github.com/micro-ROS/micro_ros_stm32cubemx_utils. I've also tryied with  `ros2 run micro_ros_setup configure_firmware.sh int32_publisher --transport serial` but dind't work.
Thanks for the help.

jacopagno, 2021-09-10:
hi again, i'm having problems with the client. The function rclc_support_init stops the program. I read the thread #281 and #277 but
didn't find a solution. I think the problem could be in the colcon.meta:
```
  "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON"                
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=custom",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=1",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=1",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=2"
            ]
        },
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        }
    }
}
```
or the way i've build and flash the system. I've done it through STM32cubeIDE following https://github.com/micro-ROS/micro_ros_stm32cubemx_utils. I've also tryied with `ros2 run micro_ros_setup configure_firmware.sh int32_publisher --transport serial` but dind't work.
Thanks for the help.

pablogs9, 2021-09-10:
Please debug and tell us where (in which line) is the system stopping. Since `rclc_support_init` calls your custom transports, check that there is no problem in those transports.

jacopagno, 2021-09-13:
thank you for the answer. I think the problem is in the connection between agent and client. The client after the reset sends 31 bytes but agent after  the init tries to read 41 bytes. Is it the right behavior?
The frame sends by the client is :
```
can0  001   [7]  7E 00 00 18 00 80 00
can0  001   [8]  00 00 00 01 10 00 58 52
can0  001   [8]  43 45 01 00 01 0F 58 51
```

pablogs9, 2021-09-13:
Yes, your custom transports are not behaving correctly. Maybe you want to share them and we can take a look.

jacopagno, 2021-09-13:
So for the client i've used the example on STM32F446re repository and set the custom transport with
```
rmw_uros_set_custom_transport(
	    true, // with CAN2.0 8 B payload
	    (void *) &hcan1,
		freertos_custom_open,
		freertos_custom_close,
		freertos_custom_write,
		freertos_custom_read);
```
and for the write i send framed packet of CAN message with the last of variable length(i've made a change from the last message). 
```
	cycle=len/8;
	rest=len%8;

	for(cycle;cycle>0;cycle--){
		pHeader.DLC = 8;
		ret = HAL_CAN_AddTxMessage( can, &pHeader, ptr , &pTxMailbox );
		if (ret == HAL_OK)  write += 8;
		ptr += 8;
	}
	if (rest!=0){
		pHeader.DLC = rest;
		ret = HAL_CAN_AddTxMessage( can, &pHeader, ptr , &pTxMailbox );
		if (ret == HAL_OK)  write += rest;
	}

        return (ret == HAL_OK && write==len) ? write : 0;
   }
```

pablogs9, 2021-09-13:
Please share the whole code of the custom transports.

jacopagno, 2021-09-13:
I've fixed with adding `while(HAL_CAN_IsTxMessagePending(can,pTxMailbox));` after the addTxMesssage. Btw i don't understand where the issue is. The byte sent by client is the same and the requested byte are still 41 but with this line the session works.

pablogs9, 2021-09-13:
Ok, closing

closed 2021-09-13 by pablogs9

jacopagno, 2021-09-15:
hi again,
so i've tested a bit and found a  problem in deserialization. The log from client is:
```
[1631689648.254658] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1631689648.254728] info     | custom_agent.cpp   | operator()               | This is an example of a custom Micro XRCE-DDS Agent INIT function | fd: 3
[1631689648.254733] info     | CustomAgent.cpp    | init                     | Custom agent status: opened | CAN_BUS agent running
[1631689653.171288] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 58 51 F4 2D 81 00 FC 01
[1631689653.171590] info     | Root.cpp           | create_client            | create                 | client_key: 0x5851F42D, session_id: 0x81
[1631689653.171678] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1481765933, address: ID: 1
[1631689653.171912] debug    | CustomAgent.cpp    | send_message             | [** <<CAN_BUS>> **]    | client_key: 0x5851F42D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1631689655.452818] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x5851F42D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 58 51 F4 2D 81 00 FC 01
[1631689655.452981] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1481765933, address: ID: 1
[1631689655.453153] debug    | CustomAgent.cpp    | send_message             | [** <<CAN_BUS>> **]    | client_key: 0x5851F42D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1631689658.460079] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
[1631689658.460159] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x5851F42D, len: 0, data: 
[1631689660.461112] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
[1631689660.461183] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x5851F42D, len: 0, data: 
```

In this moment the client sends 42 bytes to initialize a node.
The client main is the same of the example;
```
	rv = rmw_uros_set_custom_transport(
	    true,
	    (void *) &hcan1,
		freertos_serial_open,
		freertos_serial_close,
		freertos_serial_write,
		freertos_serial_read);

	rcl_allocator_t freeRTOS_allocator = rcutils_get_zero_initialized_allocator();

	freeRTOS_allocator.allocate = microros_allocate;
	freeRTOS_allocator.deallocate = microros_deallocate;
	freeRTOS_allocator.reallocate = microros_reallocate;
	freeRTOS_allocator.zero_allocate =  microros_zero_allocate;

	if (!rcutils_set_default_allocator(&freeRTOS_allocator)) {
	      printf("Error on default allocators (line %d)\n", __LINE__);
	  }

        allocator = rcl_get_default_allocator();
	rclc_support_init(&support, 0, NULL, &allocator);
	rclc_node_init_default(&node, "cubemx_node", "", &support);
```
The agent tries to read only 41 bytes and i think for this reason make a deserialization error. The agent receive is:
```
        uint8_t* ptr = buffer;
        uint8_t cycle = buffer_length/8;
        uint8_t rest = buffer_length%8;
        for(uint8_t cycle;cycle>0;cycle--){
             read(poll_fd.fd,&frame,sizeof(struct can_frame));
            memcpy(ptr,&(frame.data),8);
            ptr += 8;
            rv += 8;
        }
        if(rest!=0){
            read(poll_fd.fd,&frame,sizeof(struct can_frame));
            memcpy(ptr,&(frame.data),rest);
            rv += rest; 
       }
       source_endpoint->set_member_value<uint32_t>("ID",frame.ID);
       return rv;
      }
```

Thank you for the help.

pablogs9, 2021-09-15:
Please share the whole code of the custom transports and your main application.

jacopagno, 2021-09-15:
The main application is:
```
uint8_t rv;
	rv = rmw_uros_set_custom_transport(
	    true,
	    (void *) &hcan1,
		freertos_serial_open,
		freertos_serial_close,
		freertos_serial_write,
		freertos_serial_read);

	rcl_allocator_t freeRTOS_allocator = rcutils_get_zero_initialized_allocator();

	freeRTOS_allocator.allocate = microros_allocate;
	freeRTOS_allocator.deallocate = microros_deallocate;
	freeRTOS_allocator.reallocate = microros_reallocate;
	freeRTOS_allocator.zero_allocate =  microros_zero_allocate;

	if (!rcutils_set_default_allocator(&freeRTOS_allocator)) {
	      printf("Error on default allocators (line %d)\n", __LINE__);
	  }

	  rcl_publisher_t publisher;
	  std_msgs__msg__Int32 msg;
	  rclc_support_t support;
	  rcl_allocator_t allocator;
	  rcl_node_t node;

	  allocator = rcl_get_default_allocator();

	rclc_support_init(&support, 0, NULL, &allocator);

	// create node
	rclc_node_init_default(&node, "cubemx_node", "", &support);

	// create publisher
	rclc_publisher_init_default(
	   &publisher,
	   &node,
	   ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
	   "cubemx_publisher");

	  msg.data = 0;
  /* Infinite loop */
  for(;;)
  {
	  rcl_ret_t ret = rcl_publish(&publisher, &msg, NULL);
	      if (ret != RCL_RET_OK)
	      {
	        printf("Error publishing (line %d)\n", __LINE__);
	      }

	      msg.data++;
	      osDelay(1000);
  }
  /* USER CODE END 5 */
}
```

The client custom transport is
```
#include <uxr/client/transport.h>

#include <rmw_microxrcedds_c/config.h>

#include "main.h"
#include "cmsis_os.h"

#include <unistd.h>
#include <stdio.h>
[... 329 lines cut ...]
         */
        custom_agent.set_verbose_level(6);

        /**
         * Run agent and wait until receiving an stop signal.
         */
        custom_agent.start();

        int n_signal = 0;
        sigset_t signals;
        sigwait(&signals, &n_signal);

        /**
         * Stop agent, and exit.
         */
        custom_agent.stop();
        return 0;
    }
    catch (const std::exception& e)
    {
        std::cout << e.what() << std::endl;
        return 1;
    }
}
```

pablogs9, 2021-09-15:
When using framing in a micro-ROS transport the behaviour of the callbacks are:

- In **write callbacks**: send up to `len` B from `buf` and return the number of Bytes that you had been able to send. You don't need to send whole `len` Bytes. In this case, you can send the 8 B that fits in the CAN payload. One restriction is that the library will keep trying to send a frame thought the write callback if it returns 1 or more. If the callback is not able to send al least 1 B of `len` the library will understand this as failure.

- In **read callbacks**: you have `timeout` ms to receive up to `len` Bytes and write them in `buf`. But you can receive less and return the number of Bytes received. 

Check my comments in your log:

```
[1631689648.254658] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1631689648.254728] info     | custom_agent.cpp   | operator()               | This is an example of a custom Micro XRCE-DDS Agent INIT function | fd: 3
[1631689648.254733] info     | CustomAgent.cpp    | init                     | Custom agent status: opened | CAN_BUS agent running
// Here the client writes 24 B to init a XRCE session
[1631689653.171288] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 58 51 F4 2D 81 00 FC 01
[1631689653.171590] info     | Root.cpp           | create_client            | create                 | client_key: 0x5851F42D, session_id: 0x81
// The agent receives them correctly and creates a XRCE session
[1631689653.171678] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x1481765933, address: ID: 1
// The agent answer the sesion creation request with 19 B
[1631689653.171912] debug    | CustomAgent.cpp    | send_message             | [** <<CAN_BUS>> **]    | client_key: 0x5851F42D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
// Ans here the client tries to create a session one more time with 24 B. Has the client received the confirmation from the agent?
[1631689655.452818] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x5851F42D, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 58 51 F4 2D 81 00 FC 01
// The agent regenerates the session as requested
[1631689655.452981] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x1481765933, address: ID: 1
// And send the session confirmation again
[1631689655.453153] debug    | CustomAgent.cpp    | send_message             | [** <<CAN_BUS>> **]    | client_key: 0x5851F42D, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1631689658.460079] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
// Here the canbus receiver in the agent has returned a 0 lenght, have you set transport_rc?
[1631689658.460159] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x5851F42D, len: 0, data: 
[1631689660.461112] error    | InputMessage.cpp   | log_error                | deserialization error  | buffer: 
[1631689660.461183] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x5851F42D, len: 0, data: 
```

So, answering your question, those 40 and 41 that you are mentioning are just:
- the number of bytes that the library wants to send (in the tx callback), and you can send less.
- the number of bytes that the library can receive, but you can read less.

In the case of the agent, if you read 0 data, you have to set `transport_rc`.
One approach that I would take is to handle an only CAN message each time in the tx/rx callbacks

Maybe @Acuadros95 can add something else.

Acuadros95, 2021-09-15:
I think this is the most relevant part in your case:

>     * In **read callbacks**: you have `timeout` ms to receive up to `len` Bytes and write them in `buf`. But you can receive less and return the number of Bytes received.

You should use the timeout in both of your read functions, but focus on the Agent side first.

As @pablogs9  explained. the read function should try to read up to `len` bytes and return the actual data length when the timeout is completed, check this custom read function which takes the timeout into consideration: [link](https://github.com/micro-ROS/micro_ros_renesas2estudio_component/blob/foxy/extra_sources/microros_transports/uart_transport.c#L82)

jacopagno, 2021-09-16:
Ok i changed all the function and i think i'm almost fine. I have only a issue when create participant. I get the message but get DDS error. The log is
```
 [1631775217.406482] debug    | CustomAgent.cpp    | recv_message             | [==>> CAN_BUS <<==]    | client_key: 0x5851F42D, len: 44, data: 
0000: 81 80 00 00 01 07 22 00 00 0A 00 01 01 03 00 00 14 00 00 00 00 01 00 00 0C 00 00 00 63 75 62 65
0020: 6D 78 5F 6E 6F 64 65 00 00 00 00 00
[1631775217.406782] debug    | ProxyClient.cpp    | create_participant       | DDS error              | client_key: 0x5851F42D, participant_id: 0x000(1)
```
Thank you very much for the support.

pablogs9, 2021-09-16:
How are you running the agent? copy here the command.

Also, do you have a recent version of the agent and client libraries?

jacopagno, 2021-09-16:
eprosima::uxr::CustomAgent::SendMsgFunction send_msg_function = [&](
        const eprosima::uxr::CustomEndPoint* destination_endpoint,
        uint8_t* buffer,
        size_t message_length,
        eprosima::uxr::TransportRc& transport_rc) -> ssize_t
    {
        
        struct can_frame frame;
        int rv;
        static uint8_t byte_write=0;
        UXR_AGENT_LOG_INFO(UXR_DECORATE_BLUE("Byte"),"to write: {}",message_length);
        frame.can_id = frame.can_id = destination_endpoint->get_member<uint32_t>("ID");
        frame.can_id = frame.can_id | (0 << 29); //data frame
        frame.can_id = frame.can_id | (0 << 30); //standard
        frame.can_id = frame.can_id | (0 << 31); // 11 bit ID
        static uint8_t index=0;
        static uint8_t* ptr = buffer;

        if(index == 0 && ptr != buffer) ptr = buffer;

        if(message_length-index>=8){
    	    frame.can_dlc = 8;
            memcpy(frame.data,ptr,frame.can_dlc);
    	    rv = write(poll_fd.fd,&frame,sizeof(struct can_frame));
            byte_write = 8;
            index += 8;
            ptr += 8;

        }

        else{
    	    frame.can_dlc = message_length-index;
            memcpy(frame.data,ptr,frame.can_dlc);
    	    rv = write(poll_fd.fd,&frame,sizeof(struct can_frame));
            byte_write = frame.can_dlc;
            index += frame.can_dlc;
            ptr += frame.can_dlc;
        }
        usleep(100);
        if(index == message_length) index = 0;
         UXR_AGENT_LOG_INFO(UXR_DECORATE_BLUE("Byte"),"to write: {}",frame.can_dlc);
        
        //if(rv!=-1) byte_write = frame.can_dlc;
        //transport_rc = eprosima::uxr::TransportRc::ok;

    return byte_write;
}

I've tried to send only up to byte but if i don't receive all the byte requested give me error that the bytes read are 0.

Acuadros95, 2021-09-16:
The `DDS error` is related to the interaction between the Agent and ROS2 middleware.
Which branch and commit of the agent are you using to build your custom agent?

pablogs9, 2021-09-16:
If you receive 0 bytes you should also set `transport_rc =  eprosima::uxr::TransportRc::timeout_error`

jacopagno, 2021-09-16:
https://github.com/micro-ROS/micro-ROS-Agent/tree/foxy i use this.