micro-ROS/micro_ros_setup #394: error to do the build step for freertos with micro-ros

MaZhengg, 2021-09-26:
## Issue template

- Hardware description: stm32-e407
- RTOS: freerttos
- Installation type: micro_ros_setup
- Version or commit hash: ROS2 foxy

#### Steps to reproduce the issue
ros2 run micro_ros_setup build_firmware.sh

#### Expected behavior
i tried to do the build step in the section “First micro-ROS Application on FreeRTOS” with 
$ ros2 run micro_ros_setup build_firmware.sh

#### Actual behavior
while, the build step got the error message with 

Starting >>> rmw_implementation
--- stderr: rmw_implementation                                                       
CMake Error at /home/username/microros_ws/firmware/mcu_ws/install/rmw_implementation_cmake/share/rmw_implementation_cmake/cmake/get_default_rmw_implementation.cmake:60 (message):
  Could not find ROS middleware implementation 'NOTFOUND'.  Choose one of the
  following:
Call Stack (most recent call first):
  CMakeLists.txt:22 (get_default_rmw_implementation)

it seems that the rmw implementation can not be found,  acctually the rmw microxrcedds have been installed, so I cannot figure it out what's wrong, any advice? thank you so much

pablogs9, 2021-09-27:
Please repeat the steps in a clean environment, it seems that some configuration in your setup is wrong. Maybe the `RMW_IMPLEMENTATION` environment variable?

I have tested this and it works in a clean docker: 
https://asciinema.org/a/tVoKG9UqViORVdbno8lUqcG6j

MaZhengg, 2021-09-27:
> 
> 
> Please repeat the steps in a clean environment, it seems that some configuration in your setup is wrong. Maybe the `RMW_IMPLEMENTATION` environment variable?
> 
> I have tested this and it works in a clean docker:
> https://asciinema.org/a/tVoKG9UqViORVdbno8lUqcG6j

Thanks for your help, then I tried to change the RMW_IMPLEMENTATION environment variable.
And what is the default value for the RMW_IMPLEMENTATION environment variable?

pablogs9, 2021-09-27:
`export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`

MaZhengg, 2021-09-27:
> 
> 
> `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp`

thank you for you help, I will try to reset the `RMW_IMPLEMENTATION`, otherwise I will back to the clean environment.

closed 2021-09-27 by MaZhengg