micro-ROS/micro_ros_setup #397: Support RTEMS

roncapat, 2021-10-09:
It would be interesting to use micro-ROS on RTEMS (The Real-Time Executive for Multiprocessor Systems).

It is an RTOS used in space applications. 
It offers POSIX interface. 
A simple and cheap SBC like a BeagleBone Black would be sufficient to make tests on hardware (but everything can be done also virtually).

For now, I do not have the time to try alone, 
Of course, if I ever find the time I will try on my own and share PRs if it's ok.

roncapat, 2021-10-10:
I discovered that no matter if in your custom toolchain you add a line such:
```
set(CMAKE_DL_LIBS "-lrtemscpu") # contains dlopen() family of symbols in RTEMS 
```
the `rcutils` CMake will always have `dl` as `${CMAKE_DL_LIBS}`. Like if the override would not work correctly.

roncapat, 2021-10-10:
I opended some PRs to fix the warnings.

Next step is waiting to retrieve my Beaglebone Black (which now I don't have with me, but it will be again sometime in the following weeks) and start some tests.

roncapat, 2021-10-11:
I believe the basic setup could be a RTEMS application with 1 task and 1 micro-ROS publisher. 
Yes, I need to check if with my pre-existing setup I can already run a RTEMS-only demo in qemu, just to be sure that everything works before start breaking stuff.

joelsherrill, 2021-10-11:
> I have some questions in my mind, but the first one is: do you believe or know wether it will be necessary to write some specific code to adapt to this RTOS, or everything shoud work out of the box because the target RTOS is a POSIX system? Do you have a list of steps for things to check?
> 

I can't tell if this is a micro-ROS or RTEMS question. I can't imagine ROS needs anything from POSIX that RTEMS does not have. We have this guide in our documentation set (https://docs.rtems.org/branches/master/posix-compliance/index.html) which details how RTEMS aligns against various C and POSIX based standards. 

> EDIT: let me rephrase that: using generate_lib, is there any target-specific source code implementation needed to support micro-ROS, or I just need to use its API as normal when writing a task of my RTOS image?

If a package has a POSIX port which is geared to embedded environments (e.g. doesn't use fork/exec, graphics, etc.), then it often just compiles for RTEMS. The issues that must be addressed are RTEMS initialization and configuration and possibly contents of the initial filesystem.

roncapat, 2021-10-17:
Hello everyone, finally I got a clean RTEMS 5.1 erc32 BSP (sparc) built on my PC.
All standard tests ran in the simulator, so I was ready to start fiddling with microROS in a RTEMS app.

A year ago, I wrote this simple [CMake template fort RTEMS](https://github.com/roncapat/RTEMS-5-CMake-CPP).
I made it compile for erc32 and run with the simulator. Everything OK.

Now I started porting the ping-pong microROS app.
Here is the source [pingpongport_1.zip](https://github.com/micro-ROS/micro_ros_setup/files/7359686/pingpongport_1.zip).

I started copy-pasting some code from [freeRTOS ping-pong example](https://github.com/micro-ROS/freertos_apps/blob/foxy/apps/ping_pong/app.c) and up to now, the compiler errors are:

```
[build] Starting build
[proc] Executing command: /usr/bin/cmake --build /mnt/sdb/rtems_master/workspace/microROS/build --config Debug --target all -j 10 --
[build] [1/1 100% :: 0.132] Linking CXX executable HELLO
[build] FAILED: HELLO 
[build] : && /mnt/sdb/rtems_master/rtems/5/bin/sparc-rtems5-g++  -B/mnt/sdb/rtems_master/rtems/5/sparc-rtems5/erc32/lib -B/mnt/sdb/rtems_master/rtems/5/sparc-rtems5/lib -specs bsp_specs -qrtems -mcpu=cypress -g   CMakeFiles/HELLO.dir/example.cpp.obj  -o HELLO -L/mnt/sdb/microros_ws/firmware/build -lmicroros && :
[build] /mnt/sdb/rtems_master/rtems/5/lib/gcc/sparc-rtems5/7.5.0/../../../../sparc-rtems5/bin/ld: /mnt/sdb/microros_ws/firmware/build/libmicroros.a(libmicrocdr-array.c.obj): in function `ucdr_serialize_array_uint16_t':
[build] /mnt/sdb/microros_ws/firmware/mcu_ws/eProsima/Micro-CDR/src/c/types/array.c:176: undefined reference to `_GLOBAL_OFFSET_TABLE_'
[build] /mnt/sdb/rtems_master/rtems/5/lib/gcc/sparc-rtems5/7.5.0/../../../../sparc-rtems5/bin/ld: /mnt/sdb/microros_ws/firmware/mcu_ws/eProsima/Micro-CDR/src/c/types/array.c:176: undefined reference to `_GLOBAL_OFFSET_TABLE_'
[build] /mnt/sdb/rtems_master/rtems/5/lib/gcc/sparc-rtems5/7.5.0/../../../../sparc-rtems5/bin/ld: /mnt/sdb/microros_ws/firmware/build/libmicroros.a(libmicrocdr-array.c.obj): in function `ucdr_serialize_array_uint32_t':
[build] /mnt/sdb/microros_ws/firmware/mcu_ws/eProsima/Micro-CDR/src/c/types/array.c:177: undefined reference to `_GLOBAL_OFFSET_TABLE_'
[build] /mnt/sdb/rtems_master/rtems/5/lib/gcc/sparc-rtems5/7.5.0/../../../../sparc-rtems5/bin/ld: /mnt/sdb/microros_ws/firmware/mcu_ws/eProsima/Micro-CDR/src/c/types/array.c:177: undefined reference to `_GLOBAL_OFFSET_TABLE_'
[build] /mnt/sdb/rtems_master/rtems/5/lib/gcc/sparc-rtems5/7.5.0/../../../../sparc-rtems5/bin/ld: /mnt/sdb/microros_ws/firmware/build/libmicroros.a(libmicrocdr-array.c.obj): in function `ucdr_serialize_array_uint64_t':
[build] /mnt/sdb/microros_ws/firmware/mcu_ws/eProsima/Micro-CDR/src/c/types/array.c:178: undefined reference to `_GLOBAL_OFFSET_TABLE_'
[build] /mnt/sdb/rtems_master/rtems/5/lib/gcc/sparc-rtems5/7.5.0/../../../../sparc-rtems5/bin/ld: /mnt/sdb/microros_ws/firmware/build/libmicroros.a(libmicrocdr-array.c.obj):/mnt/sdb/microros_ws/firmware/mcu_ws/eProsima/Micro-CDR/src/c/types/array.c:178: more undefined references to `_GLOBAL_OFFSET_TABLE_' follow
[build] /mnt/sdb/rtems_master/rtems/5/lib/gcc/sparc-rtems5/7.5.0/../../../../sparc-rtems5/bin/ld: /mnt/sdb/microros_ws/firmware/build/libmicroros.a(libmicroxrcedds_client-serial_transport_posix.c.obj): in function `uxr_read_serial_data_platform':
[build] /mnt/sdb/microros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_posix.c:64: undefined reference to `poll'
[build] collect2: error: ld returned 1 exit status
[build] ninja: build stopped: subcommand failed.
[build] Build finished with exit code 1
```

In practice, two problems up to now:

- undefined reference to `_GLOBAL_OFFSET_TABLE_'
- undefined reference to `poll' 

I suspect the second one is related to the simulator not having a serial port implemented, so I may fix it changing microROS build configuration to use network instead of serial port. But the first issue is the most critical IMHO, and I don't know how to proceed. Any help?

roncapat, 2021-10-17:
Seems that adding to `rtems-colcon.meta` the following:

```
"microcdr": {
            "cmake-args": [
                "-DUCDR_PIC=OFF"
            ]
        },
```
solved the GLOBAL_OFFSET_TABLE problem. Now I have the issue with the unimpemented `poll` for serial ports in erc32 architecture (sounds to me this is the core of the issue) and going on adding other code of the ping-pong demo to my RTEMS app until everything can be tested and demonstrated.

roncapat, 2021-10-17:
Seems that RTEMS deliberately does not support poll() call.

https://docs.rtems.org/releases/rtems-docs-4.11.1/bsp-howto/console.html

RTEMS relies on _termios_. I need to investigate how to maybe offer an alternative to the POSIX implementation
Micro-XRCE-DDS-Client/src/c/profile/transport/serial/serial_transport_posix.c

joelsherrill, 2021-10-19:
Email me at joel AT rtems dot org and I will email you back a quick start network example for zynq qemu that should help you. If you don't initialize the network stack, then there is no networking.

roncapat, 2021-11-07:
@joelsherrill I wrote to you by email but sadly no response... Today I set up my beaglebone black and now I'm stuck at the same point of setting up networking. Could you send me the example at roncapat@gmail.com ? It would be very helpful for me.

joelsherrill, 2021-11-07:
@roncapat Not sure what happened. I don't see any email from you about this. But I went ahead and sent it. Poke me if you don't get it.

roncapat, 2021-11-07:
Thank you so much! Hope to share some project updates before Christmas :)

roncapat, 2021-11-13:
Working again on the subject. 

This time I'm here to share a build issue related to glibc 2.46.

On my system (Ubuntu 20.04) I have `sb-set-builder` failing with the following, uninformative last line:
```UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 78: ordinal not in range(128)```
Adding `--log /path/to/logfile` to the command line helps to get insights on what happened:
```
CC       libgio_2_0_la-gdbuserror.lo
../../glib-2.46.2/gio/gdbusauth.c: In function ‘_g_dbus_auth_run_server’:
../../glib-2.46.2/gio/gdbusauth.c:1295:11: error: ‘%s’ directive argument is null [-Werror=format-overflow=]
 1295 |           debug_print ("SERVER: WaitingForBegin, read '%s'", line);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC       libgio_2_0_la-gdbusconnection.lo
  CC       libgio_2_0_la-gdbusmessage.lo
  CC       libgio_2_0_la-gdbusnameowning.lo
cc1: some warnings being treated as errors
make[4]: *** [Makefile:3624: libgio_2_0_la-gdbusauth.lo] Errore 1
make[4]: *** Attesa per i processi non terminati....
../../glib-2.46.2/gio/gdbusmessage.c: In function ‘g_dbus_message_to_blob’:
../../glib-2.46.2/gio/gdbusmessage.c:2698:30: error: ‘%s’ directive argument is null [-Werror=format-overflow=]
 2698 |       tupled_signature_str = g_strdup_printf ("(%s)", signature_str);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```

On the internet, I found only [this reference](https://patchwork.openembedded.org/patch/165263/) to a simple patch to use whenever host GCC is version 9. However, I have GCC 8 as host compiler and GCC 7.5 as ARM compiler from RSB tools.

I tried to apply manually the patch to the source files and rebuild but files are probably being overwritten. Anyone has the solution to trigger a sb-set-builder rebuild without overwriting source files?

roncapat, 2021-11-13:
Another error while trying to build `devel/qemu4`:
```
/usr/bin/ld: linux-user/syscall.o: in function `do_syscall1':
/mnt/sdb/rtems-work/rtems-source-builder/bare/build/qemu-4.1.0-x86_64-linux-gnu-1/qemu-4.1.0/linux-user/syscall.c:7660: undefined reference to `stime'
```
the proper fix should be [this one](https://github.com/thebabush/honggfuzz-qemu/pull/5/files). Trying to follow the same patching logic of the glibc problem.

roncapat, 2021-11-14:
@joelsherrill I think I got to a good point. The task passes all allocations for the DDS, and now the application fails to establish a qemu-outgoing TCP connection to my local host. In fact, the connect() call fails and I receive no handshake on host side. Do you have hints on how to configure gateway or other parameters for outgoing connections from RTEMS app in qemu?

roncapat, 2021-11-14:
connect() from RTEMS QEMU guest tries to connect to microROS Agent on the TCP host port 8888. (127.0.0.1:8888) but returns EADDRNOTAVAIL, when using QEMU [SLIRP](http://bsdwiki.reedmedia.net/wiki/networking_qemu_virtual_bsd_systems.html) which by default should allow outgoing packets to 127.0.0.1. RTEMS is configured to use DHCP, as Joel told me to do in his tutorial.

Any suggestions about how to proceed?

roncapat, 2021-11-14:
Aaaand: IT WORKS!

PingPong example ran successfully on my machine after a very long weekend.

First of all, before forgetting it forever, [here is](https://stackoverflow.com/a/4627220/3906464) what I found useful to set a tap network between host and qemu:
```
# For Network Bridging/TAP
# Set permissions of tun device
chown root.users /dev/net/tun 
chmod g+rw /dev/net/tun

#Add a bridge, add eth0
brctl addbr br0
ifconfig eth0 0.0.0.0 promisc
brctl addif br0 eth0
dhclient br0

# Create tap0
tunctl -t tap0 -u username #replace username by your username

# Enable tap0
brctl addif br0 tap0
ifconfig tap0 up
```

I used [qemu from rtems-libbsd](https://github.com/RTEMS/rtems-libbsd). My current command line is:
```
qemu-system-arm \
-no-reboot \
-serial null -serial mon:stdio \
-net tap,ifname=tap0 -net nic,model=cadence_gem,macaddr=0e:b0:ba:5e:ba:11 \
-nographic \
-M xilinx-zynq-a9 -m 256M \
-kernel <<<RTEMS exe file>>>
```

For now, the application connects to the host and set ups correctly all publishers and subscribers. The timer ticks every 2 secs and published a ping message, but unfortunately I can't see it while `ros2 topic echo /microROS/ping`.

roncapat, 2021-11-14:
Solved the bug. It was an issue on host->target communication, where the custom code for the DDS failed to poll presence of incoming data. Seems I was not resetting the FD_SET prior to calling select(). Now that everything works well, I do some cleanups and open a PR for the DDS Client.

roncapat, 2021-11-15:
If possible, I'd keep this Issue open. Since support was merged, now the next step would be to find a way to add scripting support for RTEMS build in this repo. When I will have more time I'll give it a try.

roncapat, 2021-12-03:
Rapid note about testing serial transport with QEMU:

command line:
```
qemu-system-arm \
-no-reboot \
-serial pty \
-serial mon:stdio \
-nographic \
-M xilinx-zynq-a9 \
-m 256M \
-kernel <<<RTEMS exe file>>>
```

When launched, a message like this is printed:
```
char device redirected to /dev/pts/3 (label serial0)
```
this means that on your host `/dev/pts/3` is the port for the `micro_ros_agent`, while on RTEMS guest side the UART is `/dev/ttyS0`. Thus, in `colcon.meta` you have to configure `rmw_microxrcedds` with `"-DRMW_UXRCE_DEFAULT_SERIAL_DEVICE=/dev/ttyS0"`.

Launch the micro_ros_agent with command line:
```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/pts/3
```
and only then you can launch your RTEMS mros application.

closed 2024-04-08 by roncapat