micro-ROS/micro_ros_setup #424: Is th

wegunterjr, 2021-11-12:
## Issue template

- Hardware description: 
- RTOS: 
- Installation type: 
- Version or commit hash: 

#### Steps to reproduce the issue

#### Expected behavior

#### Actual behavior

#### Additional information

pablogs9, 2021-11-15:
micro-ROS bridge is a deprecated concept in micro-ROS. Are you meaning micro-ROS Agent? 
Please provide detailed instructions to replicate your issue.

pablogs9, 2021-11-15:
Please provide detailed instructions to replicate your issue.

wegunterjr, 2021-11-15:
Ah... Grabbing my computer to recreate,  but quick summary is that thr docker for thr agent is multi-arch so won't run on an arm processor like an Xavier Agx.

pablogs9, 2021-11-15:
You are mixing versions... if you are using ROS 2 Foxy make sure that the `micro_ros_setup` repo in checkout to `foxy`. If you are using ROS 2 Galactic, use `galactic` branch.

This error is because there are ABI breaks between Foxy and Galactic.

wegunterjr, 2021-11-15:
i will make sure again by confirming I am using the galactic branch. 
I pulled the osrf/ros2 galactic docker image, and cloned micro_ros_setup.

wegunterjr, 2021-11-15:
`updated cache in /root/.ros/rosdep/sources.cache
executing command [apt-get install -y flex]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package flex is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'flex' has no installation candidate
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y flex] failed`

wegunterjr, 2021-11-15:
alright... successful.   I was able to build it in a galactic docker on an arm processor.  That's nice!

closed 2021-11-15 by wegunterjr