micro-ROS/micro_ros_setup #433: sometimes use "rcl_publish" public reliable message will keep blocking even time-out

freeloopslk, 2021-12-11:
## Issue template

- Hardware description: STM32F407
- RTOS: freeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
We found that the Micro-ros client sometimes blocks in the "uxr_RUN_session_UNTIl_confirm_delivery" function.  The function will exit and return an error only after we restart the Agent.  The following are our modifications. Is this feasible?

bool uxr_run_session_until_confirm_delivery(
        uxrSession* session,
        int timeout_ms)
{
    UXR_LOCK_SESSION(session);

    uxr_flash_output_streams(session);

    bool timeout = false;
    /* Force break when timeout.  */
    int64_t timeout_ts = uxr_millis() + timeout_ms;
    while (!uxr_output_streams_confirmed(&session->streams) && !timeout)
    {
        timeout = !listen_message_reliably(session, timeout_ms);

        /* Force break when timeout */
        if(uxr_millis() >= timeout_ts)
        {
            printf("\r\nUROS: uxr_run_session_until_confirm_delivery --> timeout!\r\n");
            break;
        }

    }

    bool ret = uxr_output_streams_confirmed(&session->streams);

    if(ret)
    {
        //printf("\r\nUROS: uxr_run_session_until_confirm_delivery --> ok!\r\n");
    }

    UXR_UNLOCK_SESSION(session);
    return ret;
}

#### Expected behavior
When we use "rcl_publish"  function to Do not keep blocking when publishing a message fails. (reliable message)

#### Actual behavior
.
#### Additional information
.

pablogs9, 2021-12-13:
Does your modification solve the issue?

pablogs9, 2021-12-13:
Implemented here: https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/299

chrisjaxy, 2021-12-13:
> Implemented here: [eProsima/Micro-XRCE-DDS-Client#299](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/299)

Does the function "run_session_until_sync" also need to be modified in the same way?

freeloopslk, 2021-12-13:
> Does your modification solve the issue?

We are keep testing. Before this modification, two to three hours after startup communication will be abnormally blocked. After  this modification, The system communicated normally for over 30 hours.

pablogs9, 2021-12-13:
> > Implemented here: [eProsima/Micro-XRCE-DDS-Client#299](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/299)
> 
> Does the function "run_session_until_sync" also need to be modified in the same way?

Done in eProsima/Micro-XRCE-DDS-Client#299

chrisjaxy, 2021-12-13:
> > > 在这里实现：[eProsima/Micro-XRCE-DDS-Client#299](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/299)
> > 
> > 
> > 函数“run_session_until_sync”是否也需要以同样的方式修改？
> 
> 在[eProsima/Micro-XRCE-DDS-Client#299 中完成](https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/299)

Sorry for not checking carefully before, I just found out the function "uxr_run_session_until_all_status" have the same situation.

pablogs9, 2021-12-15:
https://github.com/eProsima/Micro-XRCE-DDS-Client/pull/299 merged. Closing

closed 2021-12-15 by pablogs9