micro-ROS/micro_ros_setup #435: micro-ros application on NuttX build failure with olimex-stm32-e407

MaZhengg, 2021-12-16:
- Ubuntu20.04
- foxy
- olimex-stm32-e407
- nuttx

following the tutorial:https://micro.ros.org/docs/tutorials/core/first_application_rtos/nuttx/

A small episode: at the configure step, the rcutils(in branch foxy) compile failure due to the dependence of tinydir_vendor, then I overwrite the rcutils with that in branch main, then the rcutils compile successed.

my question: the create_firmware, configure_firmware steps are successed, then the build step failed with the following message:
`pong_server.cxx: In constructor 'PongServer::PongServer(int, char**)':
/home/mazhg/microros_ws/firmware/apps/include/uros/ros_util.h:50:133: **error:** exception handling disabled, use '-fexceptions' to enable
   50 | W_RET_OK) { throw RCLException(rcutils_get_error_string().str); } }`

and the full operation are presented here: 
https://asciinema.org/a/qbXwwH0VGAT1kEcAlqCYNTrQz

Thank you for your help!

Acuadros95, 2021-12-21:
The `configure_firmware.sh` is not working as expected, could you use the `menuconfig` utility to configure the application and transport?

MaZhengg, 2021-12-21:
> 

thanks for your help, I will double check the step of `configure_firmware.sh`. For the `menuconfig` operation, yes, the configuration has been done.

Also, I will try the new micro-ROS build approach for nuttx, one problem for this tutorial, whether it works with foxy ?

Acuadros95, 2021-12-21:
I am afraid its not available on foxy.

We will rework the tutorial and the `configure_firmware.sh` script to avoid errors like this.

MaZhengg, 2021-12-27:
> 

I recreate the firmware folder, then I rerun the `configure_firmware.sh`, while the following error message appeared:
`cd /home/username/micro_ws/firmware/apps/; colcon build --cmake-target clean --cmake-target-skip-unavailable
Starting >>> firmware
--- stderr: firmware
CMake Error: The source directory "/home/username/micro_ws/firmware/apps" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
---
Failed   <<< firmware [0.09s, exited with code 1]
                                
Summary: 0 packages finished [0.24s]
  1 package failed: firmware
  1 package had stderr output: firmware`

according to the issues before, it seems that the ros2 distro is messing up with micro-ROS, any tips on this? Is there any way to solve the problem except reinstall the ros2 and micro-ROS? Thank you

Acuadros95, 2021-12-31:
I have a fix for the configuration problem, try again with the branch `fix/ping_pong` of this repository. That is:
- Use ` git clone -b fix/ping_pong https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup`
- Instead of `git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup`

Also, I am working on the configuration steps of the tutorial, please the following instructions a try and give feedback:

<details>

  ## Configuring the firmware
  
  The configuration step will set up the main micro-ROS options and select the desired application.
  It can be executed with the following command:
  
  ```bash
  # Configure step
  ros2 run micro_ros_setup configure_firmware.sh [APP] [OPTIONS]
  ```
  
  In this tutorial, we will use a Serial transport and focus on the out-of-the-box `uros_pingpong`
  application located [here](https://github.com/micro-ROS/nuttx_apps/tree/foxy/examples/uros_pingpong).
  To execute this application with the chosen transport, run the configuration command above by specifying the `[APP]` parameter as below:
  
  ```bash
  # Configure step with ping_pong app and serial-usb transport
  ros2 run micro_ros_setup configure_firmware.sh pingpong
  ```
  
  and with no `[OPTIONS]` parameter.
  
  A pre-configured ethernet example is also available:
  ```bash
  # Configure step with ping_pong app and serial-usb transport
  ros2 run micro_ros_setup configure_firmware.sh pingpong-eth
  ```
  
  To proceed with the configuration, clone the following NuttX tools repo:
  
  ```bash
  # Download the tools necessary to work with NuttX
  git clone https://bitbucket.org/nuttx/tools.git firmware/tools
  ```
  
  and then install the required `kconfig-frontends`:
  
  ```bash
  pushd firmware/tools/kconfig-frontends
  ./configure
  make
  
  # if the make command fails, type: autoreconf -f -i , and then rerun the make command.
  
  sudo make install
  sudo ldconfig
  popd
  ```
  
  Now we have two options to configure our micro-ROS transport:
  
  - Interactive NuttX menu config
    * Launch the configuration menu:
  
      ```bash
      cd firmware/NuttX
      make menuconfig
      ```
  
    * You can check that the application has been selected under `Application Configuration ---> Examples ---> micro-ROS Ping Pong`.
    * The transport is also pre-configured under the `Application Configuration ---> micro-ROS ---> Transport` option.
    * To configure the transport, use the `IP address of the agent` and `Port number of the agent` options for UDP and `Serial port to use` for the serial example.
    * To save the changes, navigate to the bottom menu with the left and right arrows, and click on the `Save` button.
    * You will be asked if you want to save your new `.config` configuration, and you need to click `Ok`, and then `Exit`.
    * Push three times the `Esc` key to close the menu and go back to `microros_ws` with:
  
        ```bash
        cd ../..
        ```
  
  - `kconfig-tweak` console commands:
    * Go to Nuttx configuration path:

      ```bash
      cd firmware/NuttX
      ```

    * UDP transport configuration:
      ```bash
      kconfig-tweak --set-val CONFIG_UROS_AGENT_IP "127.0.0.1"
      kconfig-tweak --set-val CONFIG_UROS_AGENT_PORT 8888
      ```
  
    * Serial transport configuration:
      ```bash
      kconfig-tweak --set-val CONFIG_UROS_SERIAL_PORT "/dev/ttyS0"
      ```
</details>

Acuadros95, 2022-01-05:
The fix has been merged: https://github.com/micro-ROS/NuttX/pull/57. 
And the tutorial has been updated: https://github.com/micro-ROS/micro-ROS.github.io/pull/359#pullrequestreview-844287577.

I am closing this, please give feedback and reopen if needed.

MaZhengg, 2022-01-07:
we have test the updated tutorials, the configure_firmware step failed due to  the dependence problem, the package `rosidl_typesupport_microxrcedds_c_tests` and  `rosidl_typesupport_microxrcedds_cpp_tests`  depends on the package `rosidl_typesupport_microxrcedds_test_msg` , while cmake could not find the `rosidl_typesupport_microxrcedds_test_msg` package with the following error message:
`--- stderr: rosidl_typesupport_microxrcedds_c_tests                           
CMake Error at CMakeLists.txt:26 (find_package):
  By not providing "Findrosidl_typesupport_microxrcedds_test_msg.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "rosidl_typesupport_microxrcedds_test_msg",
  but CMake did not find one.
  Could not find a package configuration file provided by
  "rosidl_typesupport_microxrcedds_test_msg" with any of the following names:
    rosidl_typesupport_microxrcedds_test_msgConfig.cmake
    rosidl_typesupport_microxrcedds_test_msg-config.cmake
  Add the installation prefix of "rosidl_typesupport_microxrcedds_test_msg"
  to CMAKE_PREFIX_PATH or set "rosidl_typesupport_microxrcedds_test_msg_DIR"
  to a directory containing one of the above files.  If
  "rosidl_typesupport_microxrcedds_test_msg" provides a separate development
  package or SDK, be sure it has been installed.
---
Failed   <<< rosidl_typesupport_microxrcedds_c_tests [1.58s, exited with code 1]`

when i colcon ignore the two package(`rosidl_typesupport_microxrcedds_c_tests` and  `rosidl_typesupport_microxrcedds_cpp_tests` ), everything goes fine, but no usb-NuttX serial device found.

with the nuttx.bin and nuttx.hex provided  by @pablogs9  in issue #268 , everything goes fine, and the topic goes well.

and the whole operation are recorded in the following link:
https://asciinema.org/a/460409

any help are grateful!

MaZhengg, 2022-01-10:
Update:
All the operation are executed successfully, while, there is no usb-NuttX serial appeared, as you can see:

https://asciinema.org/a/460890

Acuadros95, 2022-01-10:
This PR should solve the problem: https://github.com/micro-ROS/NuttX/pull/58

MaZhengg, 2022-01-11:
Appreciate, problem solved. We'll take a closer look at these modification, and I will close the issue.

closed 2022-01-11 by MaZhengg