micro-ROS/micro_ros_setup #437: the "RES" of  microros agent keep increasing

chrisjaxy, 2021-12-17:
## Issue template

- Hardware description: STM32F103
- RTOS: freertos
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
we created a publisher and a subscription on STM32F103 and created a test.sh regularly published irrelevant topic. I found micro_ros_agent RES keep increasing.
When we created 7 publishers, 4 subscribers, 10 service servers. I found the RES Increase faster. What could be the cause? Can this problem be solved?
test.sh
```
#!/bin/bash
while true
do 
   ros2 topic pub -t 1 /vel_mcu geometry_msgs/msg/Twist
done
```

a publisher and a subscription
[image]
[image]
[image]

7 publishers, 4 subscribers, 10 service servers
[image]
[image]

#### Expected behavior

#### Actual behavior

#### Additional information

pablogs9, 2021-12-17:
Which are the QoS of those micro-ROS entities?

chrisjaxy, 2021-12-17:
> Which are the QoS of those micro-ROS entities?

- a publisher and a subscription
a default publisher, a best_effort subscription.
- 6 publishers, 4 subscribers, 10 service servers
3 default publisher, 3 best_effort publisher, 3 best_effort subscriptions, 1 default publisher, 10 best service servers.

"/vel_mcu" is an unrelated topic.

pablogs9, 2021-12-17:
We will try to replicate this a find a solution.

pablogs9, 2021-12-20:
Solved here: https://github.com/micro-ROS/micro-ROS-Agent/pull/107

chrisjaxy, 2021-12-21:
> Solved here: [micro-ROS/micro-ROS-Agent#107](https://github.com/micro-ROS/micro-ROS-Agent/pull/107)
> 
> @chrisjaxy can you test this branch with your use case in order to ensure that it is solved?

Thank you for your help, we will test on our machine and tell you the conclusion.

chrisjaxy, 2021-12-23:
> Solved here: [micro-ROS/micro-ROS-Agent#107](https://github.com/micro-ROS/micro-ROS-Agent/pull/107)
> 
> @chrisjaxy can you test this branch with your use case in order to ensure that it is solved?

Thank you for your help, I think it has solved the issue

pablogs9, 2021-12-23:
Nice, I'm closing.

closed 2021-12-23 by pablogs9