micro-ROS/micro_ros_setup #438: building micro-ros-agent fails

AniketWagh2001, 2021-12-17:
## Issue template

- Hardware description: raspberry pi 4 and raspberry pico
- RTOS: raspberry pi 4 and raspberry pico
- Installation type: 
- Version or commit hash: ros-foxy

#### Steps to reproduce the issue
https://github.com/micro-ROS/micro_ros_setup#building-micro-ros-agent
i am following the steps provided here 

#### Expected behavior
the following command is not executing completely its throwing a error
ros2 run micro_ros_setup build_agent.sh
#### Actual behavior
the error after
ros2 run micro_ros_setup build_agent.sh
is as follows

HEAD is now at 1549ff12 Replace STRING(PREPEND ..) in CMakeLists.txt to support older CMake versions
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp: In constructor ‘uros::agent::graph_manager::ParticipantEntitiesInfoTypeSupport::ParticipantEntitiesInfoTypeSupport()’:
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:45:54: error: cannot bind non-const lvalue reference of type ‘bool&’ to an rvalue of type ‘bool’
   45 |     m_typeSize = 4 + callbacks_->max_serialized_size(full_bounded);
      |                                                      ^~~~~~~~~~~~
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp: In constructor ‘uros::agent::graph_manager::MicrorosGraphInfoTypeSupport::MicrorosGraphInfoTypeSupport()’:
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_typesupport.cpp:137:54: error: cannot bind non-const lvalue reference of type ‘bool&’ to an rvalue of type ‘bool’
  137 |     m_typeSize = 4 + callbacks_->max_serialized_size(full_bounded);
      |                                                      ^~~~~~~~~~~~
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp: In member function ‘void uros::agent::graph_manager::GraphManager::update_node_entities_info()’:
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:543:86: error: ‘ALIVE_INSTANCE_STATE’ is not a member of ‘eprosima::fastdds::dds::InstanceStateKind’
  543 |         if (sample_info.instance_state == eprosima::fastdds::dds::InstanceStateKind::ALIVE_INSTANCE_STATE)
      |                                                                                      ^~~~~~~~~~~~~~~~~~~~
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp: In function ‘eprosima::fastdds::dds::DataWriterQos uros::agent::graph_manager::writer_qos_conversion(const eprosima::fastdds::dds::WriterQos&)’:
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:645:20: error: ‘class eprosima::fastdds::dds::DataWriterQos’ has no member named ‘data_sharing’
  645 |     datawriter_qos.data_sharing(writer_qos.data_sharing);
      |                    ^~~~~~~~~~~~
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:645:44: error: ‘const class eprosima::fastdds::dds::WriterQos’ has no member named ‘data_sharing’
  645 |     datawriter_qos.data_sharing(writer_qos.data_sharing);
      |                                            ^~~~~~~~~~~~
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp: In function ‘eprosima::fastdds::dds::DataReaderQos uros::agent::graph_manager::reader_qos_conversion(const eprosima::fastdds::dds::ReaderQos&)’:
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:669:20: error: ‘class eprosima::fastdds::dds::DataReaderQos’ has no member named ‘data_sharing’
  669 |     datareader_qos.data_sharing(reader_qos.data_sharing);
      |                    ^~~~~~~~~~~~
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:669:44: error: ‘const class eprosima::fastdds::dds::ReaderQos’ has no member named ‘data_sharing’
  669 |     datareader_qos.data_sharing(reader_qos.data_sharing);
      |                                            ^~~~~~~~~~~~
make[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:102: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_typesupport.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:89: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_manager.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make[3]: *** [Makefile:141: all] Error 2
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:113: micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
---
Failed   <<< micro_ros_agent [6min 15s, exited with code 2]

Summary: 1 package finished [6min 45s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent

#### Additional information

following is the error and the steps i am following

ubuntu@ubuntu:~$ mkdir uros_ws && cd uros_ws
ubuntu@ubuntu:~/uros_ws$ git clone -b main https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
Cloning into 'src/micro_ros_setup'...
remote: Enumerating objects: 2900, done.
remote: Counting objects: 100% (695/695), done.
[... 108 lines cut ...]
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:645:44: error: ‘const class eprosima::fastdds::dds::WriterQos’ has no member named ‘data_sharing’
  645 |     datawriter_qos.data_sharing(writer_qos.data_sharing);
      |                                            ^~~~~~~~~~~~
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp: In function ‘eprosima::fastdds::dds::DataReaderQos uros::agent::graph_manager::reader_qos_conversion(const eprosima::fastdds::dds::ReaderQos&)’:
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:669:20: error: ‘class eprosima::fastdds::dds::DataReaderQos’ has no member named ‘data_sharing’
  669 |     datareader_qos.data_sharing(reader_qos.data_sharing);
      |                    ^~~~~~~~~~~~
/home/ubuntu/uros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/graph_manager/graph_manager.cpp:669:44: error: ‘const class eprosima::fastdds::dds::ReaderQos’ has no member named ‘data_sharing’
  669 |     datareader_qos.data_sharing(reader_qos.data_sharing);
      |                                            ^~~~~~~~~~~~
make[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:102: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_typesupport.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:89: CMakeFiles/micro_ros_agent.dir/src/agent/graph_manager/graph_manager.cpp.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make[3]: *** [Makefile:141: all] Error 2
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:113: micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
---
Failed   <<< micro_ros_agent [6min 15s, exited with code 2]

Summary: 1 package finished [6min 45s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
ubuntu@ubuntu:~/uros_ws$

pablogs9, 2021-12-20:
You are using `micro_ros_setup` branch for Rolling (`main` branch) and you have ROS 2 foxy installed. Check this line:

```
ubuntu@ubuntu:~/uros_ws$ git clone -b main https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
```

closed 2021-12-20 by pablogs9

Acuadros95, 2021-12-20:
Related: https://github.com/micro-ROS/micro_ros_setup/pull/439

AniketWagh2001, 2021-12-20:
THANK YOU GUYS.