micro-ROS/micro_ros_setup #442: [PARTICIPANT Error] Registered Type must have a name -> Function register_type

AIxWALL, 2021-12-20:
## Issue template

- Hardware description: esp32
- Installation type: static library
- Version or commit hash: foxy

#### Steps to reproduce the issue
Hi, I am running the ping_pong example on my custom board(qemu). I received this error:
`Failed status on line 107: 1. Aborting.`
I noticed that it is related to this function **RCCHECK(rclc_publisher_init_default(&ping_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Header), "/microROS/ping"));**.
How can I solve the problem?

the agent output with -v6:
`$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
[1639992849.720501] info     | UDPv4AgentLinux.cpp | init                     | running...             | port: 8888
[1639992849.721479] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1639992868.212993] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 65 DA FB 23 81 00 FC 01
[1639992868.213439] info     | Root.cpp           | create_client            | create                 | client_key: 0x65DAFB23, session_id: 0x81
[1639992868.213604] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x65DAFB23, address: 10.0.2.15:49400
[1639992868.214112] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x65DAFB23, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
[1639992868.232695] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x65DAFB23, len: 32, data: 
0000: 81 80 00 00 01 07 18 00 00 0A 00 01 01 03 00 00 09 00 00 00 00 01 00 00 01 00 00 00 00 00 00 00
[1639992868.302444] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x65DAFB23, participant_id: 0x000(1)
[1639992868.302931] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x65DAFB23, len: 14, data: 
0000: 81 80 00 00 05 01 06 00 00 0A 00 01 00 00
[1639992868.303024] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x65DAFB23, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1639992868.309858] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x65DAFB23, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80
[1639992868.317561] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x65DAFB23, len: 36, data: 
0000: 81 80 01 00 01 07 1B 00 00 0B 00 02 02 03 00 00 0D 00 00 00 01 00 00 00 00 00 01 00 01 00 00 00
0020: 00 00 01 00
2021-12-20 17:34:28.318 [PARTICIPANT Error] Registered Type must have a name -> Function register_type
2021-12-20 17:34:28.318 [PARTICIPANT Error] Registered Type must have a name -> Function unregister_type
[*** LOG ERROR #0001 ***] [2021-12-20 17:34:28] [] {argument index out of range}
[1639992868.322843] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x65DAFB23, len: 14, data: 
0000: 81 80 01 00 05 01 06 00 00 0B 00 02 80 00
[1639992868.322937] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x65DAFB23, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
[1639992868.349089] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x65DAFB23, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 02 00 00 00 80
`
the my custom transport:
`#include <sys/types.h>
#include <sys/socket.h>
#include <sys/time.h>
#include <poll.h>
#include <arpa/inet.h>
#include <unistd.h>
#include <string.h>
#include <errno.h>
#include <netdb.h>

bool my_transport_open(struct uxrCustomTransport * transport){
    printf("Micro ROS Client Custom transport: opening\n");
    my_transport_params_t * params = (my_transport_params_t*) transport->args;

    bool rv = false;
    params->poll_fd.fd = socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);

    if (-1 != params->poll_fd.fd)
    {
        struct addrinfo hints;
        struct addrinfo* result;
        struct addrinfo* ptr;

        memset(&hints, 0, sizeof(hints));
        hints.ai_family = AF_INET;
[... 200 lines cut ...]
	outcoming_ping.frame_id.data = outcoming_ping_buffer;
	outcoming_ping.frame_id.capacity = STRING_BUFFER_LEN;

	char incoming_ping_buffer[STRING_BUFFER_LEN];
	incoming_ping.frame_id.data = incoming_ping_buffer;
	incoming_ping.frame_id.capacity = STRING_BUFFER_LEN;

	char incoming_pong_buffer[STRING_BUFFER_LEN];
	incoming_pong.frame_id.data = incoming_pong_buffer;
	incoming_pong.frame_id.capacity = STRING_BUFFER_LEN;

	device_id = rand();

	while(1){
		rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
		usleep(100000);
	}	
	
	RCCHECK(rcl_publisher_fini(&ping_publisher, &node));
	RCCHECK(rcl_publisher_fini(&pong_publisher, &node));
	RCCHECK(rcl_subscription_fini(&ping_subscriber, &node));
	RCCHECK(rcl_subscription_fini(&pong_subscriber, &node));
	RCCHECK(rcl_node_fini(&node));
}
`

pablogs9, 2021-12-20:
Could you check at the transport level that you are sending all the bytes that the tx function receives as arguments? It seems that something is not reaching correctly the agent.

AIxWALL, 2021-12-21:
Yes. Thanks for the reminder. I'll close this.

AIxWALL, 2022-01-02:
Hi,again. I checked it out that I found the read function can not receive all the messge ,the "0000: 81 00 00 00 0A 01 05 00 01 00 00 00 80" message is lost every time.I think that maybe the agent send message too fast to receive it. If so, how can I balance RX and TX. If not, do you have any suggestions?

pablogs9, 2022-01-07:
Is `MICRO_ROS_FRAMING_REQUIRED` set to 0? You are using a UDP transport right? Make sure that you are not using framing at the XRCE-DDS level.

AIxWALL, 2022-02-07:
I want to add a sleep to the send function in the agnet, how should i do,what is the step?

Acuadros95, 2022-03-18:
Could you elaborate on what you want to archieve?

pablogs9, 2022-04-07:
Any update on this?

pablogs9, 2022-05-17:
Closing due to inactivity

closed 2022-05-17 by pablogs9