micro-ROS/micro_ros_setup #445: error to do the build step for freertos with micro-ros

lin123-stack, 2022-01-04:
Hardware description: stm32-e407
    RTOS: freertos
    Installation type: micro_ros_setup
    Version or commit hash: ROS2 foxy

    ****Steps to reproduce the issue****
     ros2 run micro_ros_setup build_firmware.sh

       **Expected behavior**
     i tried to do the build step in the section “First micro-ROS Application on FreeRTOS” with
    $ ros2 run micro_ros_setup build_firmware.sh

    **Actual behavior**
  while, the build step got the error message with
> Crosscompiled environment: cleaning path
Building firmware for freertos platform olimex-stm32-e407
Selected app: ping_pong
Makefile:373: *** 隐含和普通规则混合：已弃用的语法
Makefile:377: 警告：覆盖关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的配方
Makefile:374: 警告：忽略关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的旧配方
Makefile:376: *** 隐含和普通规则混合：已弃用的语法
Makefile:377: 警告：覆盖关于目标“//该文件当前的目录/build/%.o”的配方
Makefile:374: 警告：忽略关于目标“//该文件当前的目录/build/%.o”的旧配方
Makefile:380: 警告：覆盖关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的配方
Makefile:377: 警告：忽略关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的旧配方
Makefile:384: 警告：覆盖关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的配方
Makefile:380: 警告：忽略关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的旧配方
Makefile:383: *** 隐含和普通规则混合：已弃用的语法
Makefile:387: 警告：覆盖关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的配方
Makefile:384: 警告：忽略关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的旧配方
Makefile:386: *** 隐含和普通规则混合：已弃用的语法
Makefile:390: 警告：覆盖关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的配方
Makefile:387: 警告：忽略关于目标“/home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions”的旧配方
make: *** /home/libin/microros_ws/firmware/freertos_apps/microros_olimex_e407_extensions: 是一个目录。 停止。

so I cannot figure it out what's wrong, any advice? thank you so much

pablogs9, 2022-01-05:
Hello, I can not understand the error in your output. Could you try to paste the same output in English?

Thanks

lin123-stack, 2022-01-05:
sorry, I solved the problem .The reason is my wrong operation. thank you very much.

closed 2022-01-05 by lin123-stack