micro-ROS/micro_ros_setup #446: "Error: Can't find interface/stlink.cfg in procedure 'script'"

sachinjairam, 2022-01-10:
Hai, I am working with ST Nucleo-H743ZI board and Zephyr micro-ROS. While trying to flash the firmware an error occurred. The terminal displays an error message:

"Error: Can't find interface/stlink.cfg in procedure 'script' "
[image]

How can I solve this?

Thank you.
Sachin

pablogs9, 2022-01-10:
Hello @sachinjairam, could you try to flash the device removing [this line](https://github.com/micro-ROS/micro_ros_setup/blob/066cf6b1ce05d3b405ff283b6205aecdd030d053/config/zephyr/generic/flash.sh#L47). I do not have this board available here, so I cannot try how to fix this.

@CLansdaleBEC wrote the support for this board, any idea on that?

sachinjairam, 2022-01-10:
Hai @pablogs9 Thanks for your reply. I tried it, and the old error gone and new one arise,

Flashing firmware for zephyr platform nucleo_h743zi
Bus 002 Device 020: ID 0483:374e STMicroelectronics STLINK-V3
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:60: Error: Can't find target/stm32h7x.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 60

pablogs9, 2022-01-10:
You can check if it works without openocd, set [this](https://github.com/micro-ROS/micro_ros_setup/blob/066cf6b1ce05d3b405ff283b6205aecdd030d053/config/zephyr/generic/flash.sh#L43) to false and let see if `west flash` does the work.

sachinjairam, 2022-01-10:
@pablogs9 Its worked. Thank you.
But what will be the problem with openocd?
Can you tell me please?

Thank you
Sachin

closed 2022-01-11 by sachinjairam