micro-ROS/micro_ros_setup #447: ros2 run micro_ros_setup build_agent.sh fails for jetson nano

anastasiapan, 2022-01-24:
## Jetson nano agent build

- Hardware description: Jetson nano
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: galactic

#### Steps to reproduce the issue
When I try to build the agent through micro ros setup for a jetson nano it fails to build with this error:
´´´
#11 1106.9 HEAD is now at eb322062 Bump version to 1.9.2
#11 1106.9 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `typeinfo for fmt::v7::format_error'
#11 1106.9 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
#11 1106.9 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `void fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<fmt::v7::detail::buffer_appender<fmt::v7::type_identity<char>::type>, fmt::v7::type_identity<char>::type> >, fmt::v7::detail::locale_ref)'
#11 1106.9 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::error_handler::on_error(char const*)'
#11 1106.9 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::basic_data<void>::left_padding_shifts'
#11 1106.9 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `std::locale fmt::v7::detail::locale_ref::get<std::locale>() const'
#11 1106.9 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `vtable for fmt::v7::format_error'
#11 1106.9 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::format_error::~format_error()'
#11 1106.9 collect2: error: ld returned 1 exit status
#11 1106.9 make[5]: *** [micro_ros_agent] Error 1
#11 1106.9 make[4]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
#11 1106.9 make[3]: *** [all] Error 2
#11 1106.9 make[2]: *** [micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
#11 1106.9 make[1]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
#11 1106.9 make: *** [all] Error 2
#11 1106.9 ---
#11 1106.9 Failed   <<< micro_ros_agent [16min 10s, exited with code 2]

´´´
The micro_ros_msgs builds.
Then I tried to install ftm manually with apt-get but I still got the same error 

I have been using this setup for a long time now and built it several times throughout this time but I have never seen this issue before.

#### Expected behavior
Build the micro ros agen package using the micro ros setup

#### Actual behavior
Micro ros agent fails to build throwing an error that seems to be a missing dependency (fmt maybe?)

pablogs9, 2022-01-24:
On which OS are you running the build script?

anastasiapan, 2022-01-25:
> On which OS are you running the build script?

I am using docker to build and the base image I use has Ubuntu 18. From there I just build the package from source. My Dockerfile looks like this:

```
RUN mkdir microros_ws
WORKDIR microros_ws

#COPY my_package/ src/my_package/

RUN git -C src clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git \
    && git -C src clone https://github.com/micro-ROS/micro_ros_diagnostics.git  \
    && touch src/micro_ros_diagnostics/micro_ros_diagnostic_bridge/COLCON_IGNORE \
    && touch src/micro_ros_diagnostics/micro_ros_common_diagnostics/COLCON_IGNORE \
    && touch src/micro_ros_diagnostics/micro_ros_diagnostic_updater/COLCON_IGNORE \
    && . /opt/ros/${ROS_DISTRO}/install/setup.sh \
    &&  apt update \
    &&  apt install -y ed python3-pip \
    &&  apt autoremove -y \
    &&  colcon build \
    &&  rm -rf log/ build/ src/* \
    &&  rm -rf /var/lib/apt/lists/*

RUN . install/setup.sh \
    &&  apt update \
    && ros2 run micro_ros_setup create_agent_ws.sh \
    && ros2 run micro_ros_setup build_agent.sh

ENV RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```

pablogs9, 2022-01-25:
Try to download the (micro-ROS Agent ROS2 package](https://github.com/micro-ROS/micro-ROS-Agent) directly into your workspace and build it using colcon instead of using `micro_ros_setup` scripting tools.

Make sure that you have spdlog installed in the system: `sudo apt install libspdlog-dev`

anastasiapan, 2022-01-25:
> Try to download the (micro-ROS Agent ROS2 package](https://github.com/micro-ROS/micro-ROS-Agent) directly into your workspace and build it using colcon instead of using `micro_ros_setup` scripting tools.
> 
> Make sure that you have spdlog installed in the system: `sudo apt install libspdlog-dev`

I still get the same error. I have added spdlog and fmt as well:
```
RUN mkdir microros_ws
WORKDIR microros_ws

COPY my_package/ src/my_package/

RUN git -C src clone https://github.com/micro-ROS/micro_ros_diagnostics.git  \
    && git -C src clone https://github.com/micro-ROS/micro_ros_msgs.git \
    && git -C src clone https://github.com/micro-ROS/micro-ROS-Agent.git \
    && touch src/micro_ros_diagnostics/micro_ros_diagnostic_bridge/COLCON_IGNORE \
    && touch src/micro_ros_diagnostics/micro_ros_common_diagnostics/COLCON_IGNORE \
    && touch src/micro_ros_diagnostics/micro_ros_diagnostic_updater/COLCON_IGNORE \
    && . /opt/ros/${ROS_DISTRO}/install/setup.sh \
    &&  apt update \
    &&  apt install -y python3-pip \
    libspdlog-dev \
    libfmt-dev \
    &&  apt autoremove -y \
    &&  colcon build \
    &&  rm -rf log/ build/ src/* \
    &&  rm -rf /var/lib/apt/lists/*

ENV RMW_IMPLEMENTATION=rmw_fastrtps_cpp

```

Acuadros95, 2022-01-25:
I just tested the agent build on a `ros:galactic` docker inside of a raspberry, and its working fine with the expected commands:
```
ros2 run micro_ros_setup create_agent_ws.sh 
ros2 run micro_ros_setup build_agent.sh
```

Could you give us instructions to fully replicate this on a docker image?

anastasiapan, 2022-01-26:
> I just tested the agent build on a `ros:galactic` docker inside of a raspberry, and its working fine with the expected commands:
> 
> ```
> ros2 run micro_ros_setup create_agent_ws.sh 
> ros2 run micro_ros_setup build_agent.sh
> ```
> 
> Could you give us instructions to fully replicate this on a docker image?

Yes so my base image is `dustynv/ros:galactic-ros-base-l4t-r32.5.0` from https://github.com/dusty-nv/jetson-containers. It is for arm architecture. I have tried two different builds:

- One with the micro ros setup:
```
FROM dustynv/ros:galactic-ros-base-l4t-r32.5.0

RUN mkdir microros_ws
WORKDIR microros_ws

RUN mkdir src \
    && git -C src clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git \
    && git -C src clone https://github.com/micro-ROS/micro_ros_diagnostics.git  \
    && touch src/micro_ros_diagnostics/micro_ros_diagnostic_bridge/COLCON_IGNORE \
    && touch src/micro_ros_diagnostics/micro_ros_common_diagnostics/COLCON_IGNORE \
    && touch src/micro_ros_diagnostics/micro_ros_diagnostic_updater/COLCON_IGNORE \
    && . /opt/ros/${ROS_DISTRO}/install/setup.sh \
    &&  apt update \
    &&  apt install -y python3-pip \
    libspdlog-dev \
    libfmt-dev \
    &&  apt autoremove -y \
    &&  colcon build \
    &&  rm -rf log/ build/ src/* \
    &&  rm -rf /var/lib/apt/lists/*

RUN . install/setup.sh \
    &&  apt update \
    && ros2 run micro_ros_setup create_agent_ws.sh \
    && ros2 run micro_ros_setup build_agent.sh
  
ENV RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```

This one still gives me the same error:

```
#8 1152.7 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `typeinfo for fmt::v7::format_error'
#8 1152.7 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
#8 1152.7 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `void fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<fmt::v7::detail::buffer_appender<fmt::v7::type_identity<char>::type>, fmt::v7::type_identity<char>::type> >, fmt::v7::detail::locale_ref)'
#8 1152.7 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::error_handler::on_error(char const*)'
#8 1152.7 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::basic_data<void>::left_padding_shifts'
#8 1152.7 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `std::locale fmt::v7::detail::locale_ref::get<std::locale>() const'
#8 1152.7 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `vtable for fmt::v7::format_error'
#8 1152.7 /microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::format_error::~format_error()'
#8 1152.7 collect2: error: ld returned 1 exit status
#8 1152.7 make[5]: *** [micro_ros_agent] Error 1
#8 1152.7 make[4]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
#8 1152.7 make[3]: *** [all] Error 2
#8 1152.7 make[2]: *** [micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
#8 1152.7 make[1]: *** [CMakeFiles/micro_ros_agent.dir/all] Error 2
#8 1152.7 make: *** [all] Error 2
#8 1152.7 ---
#8 1152.7 Failed   <<< micro_ros_agent [16min 37s, exited with code 2]
```

- and one with just the agent:

```
FROM dustynv/ros:galactic-ros-base-l4t-r32.5.0

RUN mkdir microros_ws
WORKDIR microros_ws

RUN mkdir src \
    && git -C src clone https://github.com/micro-ROS/micro-ROS-Agent.git  \
    && . /opt/ros/galactic/install/setup.sh \
    &&  apt update \
    &&  apt install -y python3-pip \
    libspdlog-dev \
    libfmt-dev \
    &&  apt autoremove -y \
    &&  colcon build \
    &&  rm -rf log/ build/ src/* \
    &&  rm -rf /var/lib/apt/lists/*

ENV RMW_IMPLEMENTATION=rmw_fastrtps_cpp
```

The agent does not build as a standalone package because it looks like it needs the micro_ros_msgs :

``` 
#7 1005.5 CMake Error at CMakeLists.txt:50 (find_package):
#7 1005.5   By not providing "Findmicro_ros_msgs.cmake" in CMAKE_MODULE_PATH this
#7 1005.5   project has asked CMake to find a package configuration file provided by
#7 1005.5   "micro_ros_msgs", but CMake did not find one.
#7 1005.5 
#7 1005.5   Could not find a package configuration file provided by "micro_ros_msgs"
#7 1005.5   with any of the following names:
#7 1005.5 
#7 1005.5     micro_ros_msgsConfig.cmake
#7 1005.5     micro_ros_msgs-config.cmake
#7 1005.5 
#7 1005.5   Add the installation prefix of "micro_ros_msgs" to CMAKE_PREFIX_PATH or set
#7 1005.5   "micro_ros_msgs_DIR" to a directory containing one of the above files.  If
#7 1005.5   "micro_ros_msgs" provides a separate development package or SDK, be sure it
#7 1005.5   has been installed.
```

I have to cross build with build x since I am building from my pc but for arm architecture:
```
docker buildx build -f Dockerfile --platform linux/arm64 -t agent-test .
```

I have been building and using the image like this for a year without any issues. I still have a working version of the image that I had build in the past and I am using now.  I do not know if that is of any use but I got the commit hash for the micro-ros packages from there:
agent: ff2ae9187795ecdd023316c69b77dfa78568f8fd
micro_ros_msgs: e3664463e78ae5d0c34d86be92d707b3d9dfd27d

pablogs9, 2022-01-26:
> The agent does not build as a standalone package because it looks like it needs the micro_ros_msgs :

Just add `micro_ros_msgs` to you workspace:

```
...
RUN mkdir src \
    && git -C src clone https://github.com/micro-ROS/micro-ROS-Agent.git  \
    && git -C src clone https://github.com/micro-ROS/micro_ros_msgs.git  \
    && . /opt/ros/galactic/install/setup.sh \

...

```

anastasiapan, 2022-01-26:
> Just add `micro_ros_msgs` to you workspace:
>
> ```
> ...
> RUN mkdir src \
>     && git -C src clone https://github.com/micro-ROS/micro-ROS-Agent.git  \
>     && git -C src clone [...]

Sorry I forgot to mention that I have done this but it is still building  (it takes a good 20-25 minutes to build) so I responded with what I have built so far until this finishes. :smiley:

anastasiapan, 2022-01-26:
The agent+msgs still gives me the same error unfortunately!

Acuadros95, 2022-01-27:
We just added a flag to use the system version of `spdlog`: https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/303 and https://github.com/micro-ROS/micro-ROS-Agent/pull/118

Agent build should succeed with this flag enabled:
`colcon build --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON`

This should not be problematic, as your docker image includes a recent version of the logger.

anastasiapan, 2022-01-28:
> We just added a flag to use the system version of `spdlog`: [eProsima/Micro-XRCE-DDS-Agent#303](https://github.com/eProsima/Micro-XRCE-DDS-Agent/pull/303) and [micro-ROS/micro-ROS-Agent#118](https://github.com/micro-ROS/micro-ROS-Agent/pull/118)
> 
> Agent build should succeed with this flag enabled: `colcon build --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON`
> 
> This should not be problematic, as your docker image includes a recent version of the logger.

It does build now. :smile: Thank you for the fix!

closed 2022-01-28 by anastasiapan