micro-ROS/micro_ros_setup #455: build micro-ros-agent fails

robvoi, 2022-02-13:
I try to build the micro-ros-agent on Ubuntu 20.04 but it fails.
Does someone have an idea on what might be the issue? 

Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [0.53s]                     
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                             
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `typeinfo for fmt::v7::format_error'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `void fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<fmt::v7::detail::buffer_appender<fmt::v7::type_identity<char>::type>, fmt::v7::type_identity<char>::type> >, fmt::v7::detail::locale_ref)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::error_handler::on_error(char const*)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::basic_data<void>::left_padding_shifts'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `std::locale fmt::v7::detail::locale_ref::get<std::locale>() const'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `vtable for fmt::v7::format_error'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::format_error::~format_error()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:173: micro_ros_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< micro_ros_agent [0.44s, exited with code 2]

Summary: 1 package finished [1.10s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent

pablogs9, 2022-02-14:
@Acuadros95 you solve this previously right? Any idea?

Acuadros95, 2022-02-14:
Check this issue: https://github.com/micro-ROS/micro_ros_setup/issues/447

Colcon is linking with our generated spdlog library which uses version 8 of the fmt library, but your system includes a different version and its being included, causing the link error.

You can force using the installed log library with the following flag:
`colcon build --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON`

robvoi, 2022-02-14:
Thanks a lot for the tip. Unfortunately the build still fails.
I ran :
`colcon build --packages-up-to micro_ros_agent --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON  -DUAGENT_BUILD_EXECUTABLE=OFF -DUAGENT_P2P_PROFILE=OFF --no-warn-unused-cli` 

Error is the same as before.

Any Idea on where I get things wrong?

pablogs9, 2022-02-15:
Have you cleaned previously the environment? Does this fail in a clean environment?

Could you provide more details about your installation, I have replicated the issue in 20.04 and it works as expected:  https://asciinema.org/a/d8Nc46e86rK5S26gXvILjOBwD

Acuadros95, 2022-02-15:
Could you paste colcon output?

Just to check: You need to clean your previous `micro_ros_agent` colcon build (Delete subfolder on `build`, `install` folders)

robvoi, 2022-02-15:
I deleted the content of both subfolders. (thanks for the tip, first I deleted only build).
Changed build_agent.sh:
`colcon build --packages-up-to micro_ros_agent $@ --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON \
    "-DUAGENT_BUILD_EXECUTABLE=OFF" \
    "-DUAGENT_P2P_PROFILE=OFF" \
    "--no-warn-unused-cli"
`

and ran colcon buld again. It still fails.

`~/microros_ws$ colcon build
[0.210s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/robo/microros_ws/install/micro_ros_setup' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.210s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/robo/microros_ws/install/micro_ros_msgs' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.210s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/robo/microros_ws/install/micro_ros_setup' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.210s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/robo/microros_ws/install/micro_ros_agent' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.210s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/robo/microros_ws/install/micro_ros_msgs' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> micro_ros_msgs
Starting >>> drive_base_msgs
Starting >>> micro_ros_setup
Finished <<< micro_ros_setup [1.19s]                                                                                           
Finished <<< micro_ros_msgs [12.0s]                                                                       
Starting >>> micro_ros_agent
Finished <<< drive_base_msgs [14.0s]                                                                      
[Processing: micro_ros_agent]                             
[Processing: micro_ros_agent]                                      
[Processing: micro_ros_agent]                                        
--- stderr: micro_ros_agent                                          
Cloning into 'xrceagent'...
Switched to a new branch 'ros2'
Cloning into 'spdlog'...
Note: switching to 'v1.9.2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at eb322062 Bump version to 1.9.2
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `typeinfo for fmt::v7::format_error'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `void fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<fmt::v7::detail::buffer_appender<fmt::v7::type_identity<char>::type>, fmt::v7::type_identity<char>::type> >, fmt::v7::detail::locale_ref)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::error_handler::on_error(char const*)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::basic_data<void>::left_padding_shifts'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `std::locale fmt::v7::detail::locale_ref::get<std::locale>() const'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `vtable for fmt::v7::format_error'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::format_error::~format_error()'
collect2: error: ld returned 1 exit status
make[5]: *** [CMakeFiles/micro_ros_agent.dir/build.make:173: micro_ros_agent] Error 1
make[4]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make[3]: *** [Makefile:141: all] Error 2
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:113: micro_ros_agent-prefix/src/micro_ros_agent-stamp/micro_ros_agent-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
---
Failed   <<< micro_ros_agent [1min 35s, exited with code 2]

Summary: 3 packages finished [1min 47s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
`

Acuadros95, 2022-02-16:
You just ran a bare `colcon build`. Don't modify the `.sh` files, clean the environment and run the actual command:
`colcon build --packages-up-to=micro_ros_agent --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON`

robvoi, 2022-02-16:
Sorry, but I fail here due to my missing understanding of the colcon build system. (reading the colcon tutorial in parallel)
I followed there instructions and got until the end where the agent gets build.
https://micro.ros.org/docs/tutorials/core/first_application_linux/

I now deleted everything from the workspace and started over with cloning the repository.
Still don't get here to put you alternative build command.

Maybe you can explain it for a beginner. Thanks a lot. :-)

`robo@Robo-Station:~/microros_ws$ source /opt/ros/$ROS_DISTRO/setup.bash
robo@Robo-Station:~/microros_ws$ git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
Cloning into 'src/micro_ros_setup'...
remote: Enumerating objects: 2956, done.
remote: Counting objects: 100% (751/751), done.
remote: Compressing objects: 100% (457/457), done.
remote: Total 2956 (delta 509), reused 446 (delta 288), pack-reused 2205
Receiving objects: 100% (2956/2956), 618.13 KiB | 3.59 MiB/s, done.
Resolving deltas: 100% (2007/2007), done.
robo@Robo-Station:~/microros_ws$ colcon build
[0.200s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/robo/microros_ws/install/micro_ros_setup' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.200s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/robo/microros_ws/install/micro_ros_msgs' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.200s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/robo/microros_ws/install/micro_ros_setup' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.200s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/robo/microros_ws/install/micro_ros_agent' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.200s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/robo/microros_ws/install/micro_ros_msgs' in the environment variable CMAKE_PREFIX_PATH doesn't exist
Starting >>> micro_ros_setup
Finished <<< micro_ros_setup [0.87s]                  

Summary: 1 package finished [0.98s]
robo@Robo-Station:~/microros_ws$ colcon build --packages-up-to=micro_ros_agent --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON
Package 'micro_ros_agent' specified with --packages-up-to was not found
`

Acuadros95, 2022-02-17:
You can follow the whole tutorial, and then use the command we provided to build the agent, that is:
- Change this command: `ros2 run micro_ros_setup build_agent.sh`
- For this one: `colcon build --packages-up-to=micro_ros_agent --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON`

After it is built you can run it with the tutorial command `ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888`

Also, you can run the dockerized version directly like this:
`docker run -it --rm --net=host -v /dev/shm:/dev/shm --privileged microros/micro-ros-agent:galactic udp4 --port 8888 -v6`