micro-ROS/micro_ros_setup #456: build micro-ros-agent fails

debanik123, 2022-02-19:
## Issue template

- Hardware description: Nvidia Jetson nano Docker container with ros foxy (dustynv/ros:foxy-ros-base-l4t-r32.6.1)
- Installation type: micro-ROS Agent

#### 
I try to build the micro-ros-agent on  Nvidia Jetson nano Docker container(ros2 foxy) but it fails. Does someone have an idea on what might be the issue? (don't know how to remove the automatic formatting of the console output below)
Starting >>> micro_ros_agent
--- stderr: micro_ros_agent                             
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `typeinfo for fmt::v7::format_error'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `void fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<fmt::v7::detail::buffer_appender<fmt::v7::type_identity<char>::type>, fmt::v7::type_identity<char>::type> >, fmt::v7::detail::locale_ref)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::error_handler::on_error(char const*)'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::detail::basic_data<void>::left_padding_shifts'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `std::locale fmt::v7::detail::locale_ref::get<std::locale>() const'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `vtable for fmt::v7::format_error'
/usr/bin/ld: /home/robo/microros_ws/install/micro_ros_agent/lib/libmicroxrcedds_agent.so.2.1.0: undefined reference to `fmt::v7::format_error::~format_error()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/micro_ros_agent.dir/build.make:173: micro_ros_agent] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/micro_ros_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< micro_ros_agent [0.44s, exited with code 2]

#### Expected behavior
micro-ROS Agent build success

#### Actual behavior
micro-ROS Agent build failed

pablogs9, 2022-02-21:
@Acuadros95 Can you reference the solution here?

Acuadros95, 2022-02-21:
Yes! Here it is: https://github.com/micro-ROS/micro_ros_setup/issues/455

pablogs9, 2022-02-23:
@debanik123 any update on this?

debanik123, 2022-02-23:
Thanks, @pablogs9, I was able to install micro ros on nvidia Jetson nvc docker images. I used following steps.->
`mkdir -p uros_ws
cd uros_ws/
git clone --depth 1 -b foxy https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
. /opt/ros/foxy/install/setup.bash
apt update
apt install -y ed python3-pip
rosdep update --rosdistro $ROS_DISTRO
rosdep install --from-paths src --ignore-src -y
colcon build
rm -rf log/ build/ src/*
rm -rf /var/lib/apt/lists/*
. install/local_setup.sh
apt update
ros2 run micro_ros_setup create_agent_ws.sh
colcon build --cmake-args -DUAGENT_USE_SYSTEM_LOGGER=ON
rm -rf log/ build/ src/
. "/uros_ws/install/local_setup.sh"  `

closed 2022-02-24 by pablogs9