micro-ROS/micro_ros_setup #457: build firmware on linux failed

UncleMask, 2022-02-24:
## Issue template
-OS:Ubuntu 20.04.3
- Version or commit hash: foxy
I try to build first application on linux,followed the tutorials,but when I run the build script, it shows some error,maybe you can help me,thx!
#### Expected behavior
build success
#### Actual behavior
here is some info
`simon@simon-MS-7A40:~/microros_ws$ ros2 run micro_ros_setup build_firmware.sh
Compiling for host environment: not cleaning path
Building firmware for host platform generic
Starting >>> microcdr
Finished <<< microcdr [0.92s]                     
Starting >>> rosidl_typesupport_microxrcedds_c
Finished <<< rosidl_typesupport_microxrcedds_c [0.51s]                  

Summary: 2 packages finished [1.53s]
[0.156s] WARNING:colcon.colcon_core.verb:Some selected packages are already built in one or more underlay workspaces:
	'microcdr' is in: /home/simon/microros_ws/install/microcdr
	'rosidl_typesupport_microxrcedds_c' is in: /home/simon/microros_ws/install/rosidl_typesupport_microxrcedds_c
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding microcdr rosidl_typesupport_microxrcedds_c

This may be promoted to an error in a future release of colcon-core.
Starting >>> microcdr
Finished <<< microcdr [0.08s]
Starting >>> rosidl_typesupport_microxrcedds_c
Finished <<< rosidl_typesupport_microxrcedds_c [0.11s]                
Starting >>> rosidl_typesupport_microxrcedds_cpp
Finished <<< rosidl_typesupport_microxrcedds_cpp [0.55s]                  

Summary: 3 packages finished [0.84s]
[0.182s] WARNING:colcon.colcon_core.verb:Some selected packages are already built in one or more underlay workspaces:
	'builtin_interfaces' is in: /opt/ros/foxy
	'lifecycle_msgs' is in: /opt/ros/foxy
	'micro_ros_setup' is in: /home/simon/microros_ws/install/micro_ros_setup
	'microcdr' is in: /home/simon/microros_ws/install/microcdr
	'std_srvs' is in: /opt/ros/foxy
	'unique_identifier_msgs' is in: /opt/ros/foxy
	'action_msgs' is in: /opt/ros/foxy
	'rcl_interfaces' is in: /opt/ros/foxy
	'rosgraph_msgs' is in: /opt/ros/foxy
	'rosidl_typesupport_microxrcedds_c' is in: /home/simon/microros_ws/install/rosidl_typesupport_microxrcedds_c
	'statistics_msgs' is in: /opt/ros/foxy
	'std_msgs' is in: /opt/ros/foxy
	'actionlib_msgs' is in: /opt/ros/foxy
	'composition_interfaces' is in: /opt/ros/foxy
	'example_interfaces' is in: /opt/ros/foxy
	'geometry_msgs' is in: /opt/ros/foxy
	'rosidl_typesupport_microxrcedds_cpp' is in: /home/simon/microros_ws/install/rosidl_typesupport_microxrcedds_cpp
	'diagnostic_msgs' is in: /opt/ros/foxy
	'nav_msgs' is in: /opt/ros/foxy
	'sensor_msgs' is in: /opt/ros/foxy
	'shape_msgs' is in: /opt/ros/foxy
	'trajectory_msgs' is in: /opt/ros/foxy
	'visualization_msgs' is in: /opt/ros/foxy
	'stereo_msgs' is in: /opt/ros/foxy
	'common_interfaces' is in: /opt/ros/foxy
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding action_msgs actionlib_msgs builtin_interfaces common_interfaces composition_interfaces diagnostic_msgs example_interfaces geometry_msgs lifecycle_msgs micro_ros_setup microcdr nav_msgs rcl_interfaces rosgraph_msgs rosidl_typesupport_microxrcedds_c rosidl_typesupport_microxrcedds_cpp sensor_msgs shape_msgs statistics_msgs std_msgs std_srvs stereo_msgs trajectory_msgs unique_identifier_msgs visualization_msgs

This may be promoted to an error in a future release of colcon-core.
Starting >>> builtin_interfaces
Starting >>> microcdr
Starting >>> unique_identifier_msgs
Starting >>> micro_ros_msgs
Starting >>> std_srvs
Starting >>> lifecycle_msgs
Starting >>> micro_ros_setup
Finished <<< microcdr [0.26s]                                           
Starting >>> rosidl_typesupport_microxrcedds_c
Starting >>> microxrcedds_client
Finished <<< micro_ros_setup [0.25s]
Finished <<< rosidl_typesupport_microxrcedds_c [0.15s]                  
Starting >>> rosidl_typesupport_microxrcedds_cpp
Finished <<< rosidl_typesupport_microxrcedds_cpp [0.13s]                
Finished <<< microxrcedds_client [0.97s]                                   
--- stderr: lifecycle_msgs                                                  
make[2]: *** [CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/build.make:82: rosidl_generator_c/lifecycle_msgs/msg/state.h] Bus error (core dumped)
make[2]: *** Deleting file 'rosidl_generator_c/lifecycle_msgs/msg/state.h'
make[1]: *** [CMakeFiles/Makefile2:338: CMakeFiles/lifecycle_msgs__rosidl_generator_c.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< lifecycle_msgs [3.20s, exited with code 2]
Aborted  <<< micro_ros_msgs [3.27s]
Aborted  <<< std_srvs [3.31s]                    
Aborted  <<< unique_identifier_msgs [4.05s]      
Aborted  <<< builtin_interfaces [4.40s]                                
                                  
Summary: 5 packages finished [4.53s]
  1 package failed: lifecycle_msgs
  4 packages aborted: builtin_interfaces micro_ros_msgs std_srvs unique_identifier_msgs
  3 packages had stderr output: lifecycle_msgs micro_ros_msgs std_srvs
  27 packages not processed
`
#### Additional information

pablogs9, 2022-02-24:
@Acuadros95 can you take a look at this?

Acuadros95, 2022-02-25:
I could not replicate this. Also, `Bus error` is a strange error related to hardware memory.
How did you install ROS2? Do you have enough memory on your workspace?

UncleMask, 2022-02-26:
I have enough memory，since its a fresh installed OS and a clean ros2 environment on my pc, but I can compile them success in my visual machine with a prebuilt ros2 environment

ZhenshengLee, 2022-03-03:
@UncleMask 

> 'builtin_interfaces' is in: /opt/ros/foxy

When building firmware(with extension and mcu_ws), DO NOT `source /opt/ros/foxy/setup.bash`, DO `source dev_ws/install/setup.bash`

When building agent(with agent_ws), DO `source /opt/ros/foxy/setup.bash`, DO NOT `source dev_ws/install/setup.bash`

You'd better remove any setup like `source /opt/ros/foxy/setup.bash` in your `~/.bashrc`

You could check my repo of stm32 of microROS which simplified the building system
It's in https://github.com/ZhenshengLee/ros2_mcu

pablogs9, 2022-03-08:
Any update on this?

closed 2022-03-09 by UncleMask