micro-ROS/micro_ros_setup #466: configure failure micro-ros application on NuttX with olimex-stm32-e407

lin123-stack, 2022-03-04:
- Hardware description: olimex-stm32-e407
- RTOS: nuttx
- Installation type:  micro_ros_setup
- Version or commit hash:  ubuntu20.04  foxy
following the tutorial:https://micro.ros.org/docs/tutorials/core/first_application_rtos/nuttx/

my question: the create_firmware step is successed, then the configure step failed with the following message:
``--- stderr: rosidl_typesupport_microxrcedds_c_tests                           
CMake Error at CMakeLists.txt:26 (find_package):
  By not providing "Findrosidl_typesupport_microxrcedds_test_msg.cmake" in
  CMAKE_MODULE_PATH this project has asked CMake to find a package
  configuration file provided by "rosidl_typesupport_microxrcedds_test_msg",
  but CMake did not find one.
  Could not find a package configuration file provided by
  "rosidl_typesupport_microxrcedds_test_msg" with any of the following names:
    rosidl_typesupport_microxrcedds_test_msgConfig.cmake
    rosidl_typesupport_microxrcedds_test_msg-config.cmake
  Add the installation prefix of "rosidl_typesupport_microxrcedds_test_msg"
  to CMAKE_PREFIX_PATH or set "rosidl_typesupport_microxrcedds_test_msg_DIR"
  to a directory containing one of the above files.  If
  "rosidl_typesupport_microxrcedds_test_msg" provides a separate development
  package or SDK, be sure it has been installed.
---
Failed   <<< rosidl_typesupport_microxrcedds_c_tests [1.58s, exited with code 1]`
`
any help are grateful!

pablogs9, 2022-03-04:
I have tested the procedure here you can check it: https://asciinema.org/a/WVTZGshpHZwGq7SWtG1Ywo62X

Please make sure that you use the correct commands.

Also, you have available a standalone module for integrating micro-ROS in your Nuttx 10.0 installation instead of using our fork: https://github.com/micro-ROS/micro_ros_nuttx_app

lin123-stack, 2022-03-04:
what the mean of "a standalone module for integrating micro-ROS in your Nuttx 10.0 installation "?
how to operate this?

lin123-stack, 2022-03-04:
what the mean of "a standalone module for integrating micro-ROS in your Nuttx 10.0 installation "?
how to operate this?
thank you !

pablogs9, 2022-03-07:
A micro-ROS app for Nuttx, check https://github.com/micro-ROS/micro_ros_nuttx_app

closed 2022-03-09 by lin123-stack