micro-ROS/micro_ros_setup #468: Unable  to launch micro-ros-agent from ROS2 launch file

Avinashmuthukumarasamy, 2022-03-07:
## Issue template

- Hardware description: docker in Raspbery Pi4  [Debian GNU(bullseye)]
- RTOS:
- Installation type: 
- Version or commit hash: foxy

#### Steps to reproduce the issue
Need to launch the micro ros agent in a ROS2 launch file similar to how we launch a serial node in ROS -melodic version.
Tried running the below command in ROS2 launch file format.

`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0`
\\ROS2 launch file to launch the micro-ros agent 
`import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node
def generate_launch_description():
    ld = LaunchDescription()
    node = Node(
        package="micro_ros_agent",
        executable="micro_ros_agent",
        name = "micro_ros_agent",
        output="screen",
        transport= "serial",
        parameters = [{"dev": '/dev/ttyACM0'}]      
    )
    ld.add_action(node)   
    return ld`

#### Expected behavior
To run the micro-ros agent successfully
`^Croot@pi:/microros_wsros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM2

[1646655493.754821] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM2, error 2, waiting for connection...
[1646655495.755321] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1646655495.756548] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1646655495.773684] info     | Root.cpp           | create_client            | create                 | client_key: 0x50182E50, session_id: 0x81
[1646655495.775655] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x50182E50, address: 0
[1646655495.873104] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x50182E50, participant_id: 0x000(1)
[1646655495.880262] info     | ProxyClient.cpp    | create_replier           | replier created        | client_key: 0x50182E50, requester_id: 0x000(7), participant_id: 0x000(1)
`

#### Actual behavior

Gave type error as mentioned below.
`INFO] [launch]: All log files can be found below /root/.ros/log/2022-03-07-11-52-23-671773-pi64-7255
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('py')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 68, in get_launch_description_from_python_launch_file
    return getattr(launch_file_module, 'generate_launch_description')()
  File "teensy_launch.py", line 13, in generate_launch_description
    hinge_node = Node(
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 210, in __init__
    super().__init__(cmd=cmd, **kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 207, in __init__
    super().__init__(**kwargs)
TypeError: __init__() got an unexpected keyword argument 'transport'

`
Also tried giving the serial as a parameter inside the parameters 

`import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node
def generate_launch_description():
    ld = LaunchDescription()
    node = Node(
        package="micro_ros_agent",
        executable="micro_ros_agent",
        name = "micro_ros_agent",
        output="screen",
        parameters = [{"transport": 'serial',"dev": '/dev/ttyACM0'}]      
    )
    ld.add_action(node)   
    return ld`

Got the following error:
`[INFO] [launch]: All log files can be found below /root/.ros/log/2022-03-07-12-13-15-792434-pi64-8451
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [micro_ros_agent-1]: process started with pid [8453]
[micro_ros_agent-1] Usage: 'micro_ros_agent <udp4|udp6|tcp4|tpc6|canfd|serial|multiserial|pseudoterminal> <<args>>'
[micro_ros_agent-1] For a more detailed description about all the available arguments, please execute the agent with '-h/--help' option.
[ERROR] [micro_ros_agent-1]: process has died [pid 8453, exit code 1, cmd '/microros_ws/install/micro_ros_agent/lib/micro_ros_agent/micro_ros_agent --ros-args -r __node:=micro_ros_agent --params-file /tmp/launch_params_a282nnl4'].
`

Please let me know if I am missing steps.
#### Additional information

pablogs9, 2022-03-08:
Hi, take a look here: https://github.com/micro-ROS/micro-ROS-Agent/issues/42

Have you tried with arguments in format: `arguments=['serial','--dev', '/dev/ttyACM0']` ?

Avinashmuthukumarasamy, 2022-03-08:
Solved the issue.Thanks for the much needed help.

closed 2022-03-08 by Avinashmuthukumarasamy