micro-ROS/micro_ros_setup #469: Olimex STM32-E407 board hangs after some time - ping_pong and int32_publisher app with UDP transport (ros:galactic)

TanmayDeshmukh, 2022-03-08:
## Issue template

- Hardware description: Olimex STM32-F407
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: Galactic

#### Steps to reproduce the issue
Follow [this](https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/) tutorial:
```
docker run -it --net=host -v /dev:/dev --privileged ros:galactic
source /opt/ros/$ROS_DISTRO/setup.bash
mkdir microros_ws
cd microros_ws
git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
sudo apt update && rosdep update
rosdep install --from-path src --ignore-src -y
sudo apt-get install python3-pip
colcon build
source install/local_setup.bash
ros2 run micro_ros_setup create_firmware_ws.sh freertos olimex-stm32-e407
ros2 run micro_ros_setup configure_firmware.sh ping_pong -t udp -i 192.168.10.32 -p 8888
ros2 run micro_ros_setup build_firmware.sh
ros2 run micro_ros_setup flash_firmware.sh

```

#### Expected behaviour
Messages must be published without interruption.

#### Actual behavior

- Messages are published but after a while, the board seems to hang (LEDs stop blinking), and no messages published.
- This happens at random and is worsened when the timer timeout is set lower, i.e, at high publisher rates. We eventually want to run an application that publishes at at least 100Hz. 
- In my tests so far, the ping_pong example app (that publishes every 2 seconds) has stopped after around 1200s (based on the message timestamps on /microROS/ping). It is very hard to make this reproducable as it may or may not stop at all.    
- I also have this issue with the int32_publisher example when it is set to run at 100Hz.

#### Additional information
The last seen lines of log seen at the agent (ping_pong example) : 
```
1646736973.873109] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 34, data: 
0000: 50 03 00 00 40 9F 36 2B 16 00 00 00 31 38 35 34 37 37 34 36 38 30 5F 31 38 39 35 31 33 30 37 39
0020: 30 00
[1646736973.873148] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 34, data: 
0000: 50 03 00 00 40 9F 36 2B 16 00 00 00 31 38 35 34 37 37 34 36 38 30 5F 31 38 39 35 31 33 30 37 39
0020: 30 00
[1646736973.873259] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2693D1FA, len: 46, data: 
0000: 81 01 A5 01 09 01 26 00 00 14 00 06 50 03 00 00 40 9F 36 2B 16 00 00 00 31 38 35 34 37 37 34 36
0020: 38 30 5F 31 38 39 35 31 33 30 37 39 30 00
[1646736973.873331] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2693D1FA, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 B4 01 00 00 80
[1646736975.854734] debug    | UDPv4AgentLinux.cpp | recv_message             | [==>> UDP <<==]        | client_key: 0x2693D1FA, len: 48, data: 
0000: 81 80 B4 01 07 01 26 00 01 BF 00 05 52 03 00 00 40 9F 36 2B 16 00 00 00 31 36 34 38 39 32 37 33
0020: 38 33 5F 31 38 39 35 31 33 30 37 39 30 00 00 00
[1646736975.855176] debug    | DataReader.cpp     | read_fn                  | [==>> DDS <<==]        | client_key: 0x00000000, len: 34, data: 
0000: 52 03 00 00 40 9F 36 2B 16 00 00 00 31 36 34 38 39 32 37 33 38 33 5F 31 38 39 35 31 33 30 37 39
0020: 30 00
[1646736975.855242] debug    | DataWriter.cpp     | write                    | [** <<DDS>> **]        | client_key: 0x00000000, len: 34, data: 
0000: 52 03 00 00 40 9F 36 2B 16 00 00 00 31 36 34 38 39 32 37 33 38 33 5F 31 38 39 35 31 33 30 37 39
0020: 30 00
[1646736975.855295] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2693D1FA, len: 46, data: 
0000: 81 01 A6 01 09 01 26 00 00 14 00 06 52 03 00 00 40 9F 36 2B 16 00 00 00 31 36 34 38 39 32 37 33
0020: 38 33 5F 31 38 39 35 31 33 30 37 39 30 00
[1646736975.855333] debug    | UDPv4AgentLinux.cpp | send_message             | [** <<UDP>> **]        | client_key: 0x2693D1FA, len: 13, data: 
0000: 81 00 00 00 0A 01 05 00 B5 01 00 00 80

```

pablogs9, 2022-03-08:
Hello @TanmayDeshmukh, is it possible for you to debug the application and check in which line is the code stopped? Is this a FreeRTOS assert or it is a hard fault?

TanmayDeshmukh, 2022-03-08:
@pablogs9 Is there maybe a tutorial for debugging through JTAG you could point me to?

pablogs9, 2022-03-08:
No, we do not have any. It should be straightforward using `opencd` and [`marus25.cortex-debug`](https://marketplace.visualstudio.com/items?itemName=marus25.cortex-debug)

ZhenshengLee, 2022-03-11:
@TanmayDeshmukh  

to debug with vscode as gdb frontend, add `launch.json` like the follow

```json
{
            "name": "custom_stm32f767zi-stlink-dbg",
            "type": "cortex-debug",
            "request": "launch",
            "servertype": "stutil",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/libs/custom_stm32f767zi/build/micro-ROS.elf",
            "runToEntryPoint": "appMain",
            "armToolchainPath": "${workspaceRoot}/toolchain/gcc-arm-none-eabi-8-2019-q3-update/bin",
            "svdFile": "${workspaceRoot}/.vscode/svd/STM32F767.svd",
        },
```

LiWeiAn, 2022-03-11:
We managed to debug the app, and we found out that multiple threads died after some time, including `initTask`, `ethernetif_set_link` and the `appMain`. This is why the board stopped blinking and the publisher stopped. We were then stuck looping through the `ethernetif_input` function from _ethernetif.c_. 

We looked into the bug/issue that @ZhenshengLee mentioned in this issue, and we increased the stack size value `INTERFACE_THREAD_STACK_SIZE` to 500, which seems to solve the issue for now. We will continue to test this.

We wanted to ask if there is a way to see stack overflow through the debugger?

ZhenshengLee, 2022-03-14:
@LiWeiAn  accoding to https://github.com/micro-ROS/freertos_apps/issues/96#issuecomment-1065019133
This can be closed.

closed 2022-03-14 by pablogs9