micro-ROS/micro_ros_setup #470: [BUG] microROS Host(Linux) emulator running failed: `xxx_rosidl_typesupport_microxrcedds_c.so` lost.

ZhenshengLee, 2022-03-14:
## Issue template

- Hardware description: host pc
- RTOS: Linux
- Installation type: micro_ros_setup like, the src code is in https://github.com/ZhenshengLee/ros2_mcu/tree/master/host_ws
- Version or commit hash: galactic

#### Steps to reproduce the issue

https://github.com/ZhenshengLee/ros2_mcu/tree/master/host_ws

compile
```sh
source /opt/ros/galactic/setup.bash
./build.sh
```
colcon.meta is https://github.com/ZhenshengLee/ros2_mcu/blob/master/host_ws/colcon.meta

running agent

```sh
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
```

running app

```sh
cd ./host_ws
source /opt/ros/galactic/setup.bash
source ./install/local_setup.bash
export RMW_IMPLEMENTATION=rmw_microxrcedds
ros2 run micro_ros_demos_rclc int32_publisher
```

#### Expected behavior

connecting with agent and publish msg.

#### Actual behavior

run failer.

```sh
zs@zs-3630:~/zs_ws/ros2_mcu/host_ws$ ros2 run micro_ros_demos_rclc int32_publisher
[ERROR] [1647244684.547171495] [rclc]: [rclc_publisher_init_best_effort] Error in rcl_publisher_init: Could not load library libstd_msgs__rosidl_typesupport_microxrcedds_c.so: dlopen error: libstd_msgs__rosidl_typesupport_microxrcedds_c.so: cannot open shared object file: No such file or directory, at /tmp/binarydeb/ros-galactic-rcutils-4.0.2/src/shared_library.c:99, at /tmp/binarydeb/ros-galactic-rosidl-typesupport-c-1.2.1/src/type_support_dispatch.hpp:78, at /tmp/binarydeb/ros-galactic-rcl-3.1.2/src/rcl/publisher.c:114

Failed status on line 40: 1. Aborting.
```

#### Additional information

in `host_ws/install/std_msgs/lib`, there are only static libs with `rosidl_typesupport_microxrcedds_cpp` and `rosidl_typesupport_microxrcedds_c`

```sh
zs@zs-3630:~/zs_ws/ros2_mcu/host_ws/install/std_msgs/lib$ ls
libstd_msgs__python.so                  libstd_msgs__rosidl_typesupport_fastrtps_cpp.so       libstd_msgs__rosidl_typesupport_microxrcedds_c.a
libstd_msgs__rosidl_generator_c.so      libstd_msgs__rosidl_typesupport_fastrtps_c.so         libstd_msgs__rosidl_typesupport_microxrcedds_cpp.a
libstd_msgs__rosidl_typesupport_cpp.so  libstd_msgs__rosidl_typesupport_introspection_cpp.so  python3.8
libstd_msgs__rosidl_typesupport_c.so    libstd_msgs__rosidl_typesupport_introspection_c.so
```

pablogs9, 2022-03-14:
Normally the build script is called from here: https://github.com/micro-ROS/micro_ros_setup/blob/galactic/scripts/build_firmware.sh

And this parent script prepares the environment variables and path in order to handle correctly the type support. 

Please make sure that your custom build process follows the same logic.

ZhenshengLee, 2022-03-15:
> Please make sure that your custom build process follows the same logic.

@pablogs9 

I am sure that the build script and colcon.meta is following https://github.com/micro-ROS/micro_ros_setup/blob/galactic/scripts/build_firmware.sh and https://github.com/micro-ROS/micro_ros_setup/tree/galactic/config/host/generic

But due to network issues, I cannot test building scripts from this micro_ros_setup repo, could you give this a test?

I made a PR #473  to generate dynamic libs and made things work, which needs your review.

pablogs9, 2022-03-15:
https://github.com/micro-ROS/micro_ros_setup/pull/473 merged, closing

closed 2022-03-15 by pablogs9