micro-ROS/micro_ros_setup #476: having trouble in installing ESP-IDF tools (jetson nano, esp32)

jdw0425, 2022-03-15:
## Issue template

- Hardware description: JETSON NANO, Ubuntu 20.04
- RTOS: freeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

#### Steps to reproduce the issue
I'm following  this tutorial.    https://micro.ros.org/blog/2020/08/27/esp32/ this tutorial.

Ros2 is installed well.

but when I typed
`ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32
`
I got
` ERROR : tool xtensa-esp32-elf does not have versions compatible with platform linux-arm64`
Is there any way to solve this problem?
or should I change my jetson nano to other non-arm64 device?

#### Expected behavior
Installing ESP-IDF

#### Actual behavior
```
Installing ESP-IDF tools
Installing tools : xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
ERROR : tool xtensa-esp32-elf does not have versions compatible with platform linux-arm64
```

#### Additional information

pablogs9, 2022-03-15:
Hi, @jdw0425 it seems that ESP-IDF is not resolving the cross-toolchain dependencies for your host architecture.

Maybe you can try to install a standalone [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/) and use the [micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component).

Meanwhile, @uLipe any idea on how to solve this?

jdw0425, 2022-03-16:
> 
Hi, @pablogs9 . Thanks for the reply.

I installed standalone ESP-IDF and successed to build and flash the example project. 

But still, I cannot install freertos esp32.

When I installed '~/esp/espidf/install.sh', it says every components are already installed or satisfied.

I tried uninstall/install ESP-IDF many times, but I'm stuck in here...

Please let me know if there is other option I can try.

Thanks again.

pablogs9, 2022-03-16:
If you have successfully installed ESP IDF and you are able to build and flash your ESP32, take a look at the [micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component), let me know in which step you have problems.

uLipe, 2022-03-16:
@pablogs9 it seems the installer is trying to use an old version of esp-idf, the v4.2 and I' m not sure if versions for arm64 are available for this version.

Regardless the method, of installation of esp-idf, @jdw0425 what version of esp-idf you' re using? I strongly recommend to use v4.3.1 or the new v4.4 stable.

jdw0425, 2022-03-17:
@uLipe Hi, I'm using v4.4. Thanks!

@pablogs9 
This is my very first project so there is a lot of confusing. I apologize for this.
There is no problem in flashing ESP32 by following [micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component).
But when I run 'ros2 run micro_ros_setup create_firmware_ws.sh freertos esp32', 
error still says
'tool xtensa-esp32-elf does not have versions compatible with platform linux-arm64'

It might be a silly question, 
If I can follow your suggested tutorial, then I don't need to follow https://micro.ros.org/blog/2020/08/27/esp32/ anymore?
No need to run the 'micro_ros_setup' lines, just create my own free rtos project, paste the micro-ros component folder and run the 'idf.py' lines?
Thanks for the reply and thanks for helping beginner.

pablogs9, 2022-03-17:
Hello, @jdw0425 that's it! `micro_ros_setup` is a packet with tools for easing the usage of micro-ROS in some platforms and [micro-ROS component for ESP-IDF](https://github.com/micro-ROS/micro_ros_espidf_component) is another tool for integrating micro-ROS in the ESP-IDF. You can choose one or another. IMO, is better to work with the second if you are using the ESP32.

So if I have understood, you have successfully flashed your micro-ROS application using the IDF? Can we close this?

jdw0425, 2022-03-17:
@pablogs9 Yes. Now I can go one step further. Thanks a lot!

closed 2022-03-17 by pablogs9

erikdominguez01, 2024-04-17:
hey this issue is still relevant even with the Jetson Orin Nano. please review this problem.

erikdominguez01, 2024-04-17:
getting issues with esp32 not recognized by arm64 even with 5.2v