micro-ROS/micro_ros_setup #478: Cannot list node's services

bjsowa, 2022-03-21:
When using the Micro-XRCE DDS middleware, I can see the services registered by the node (`ros2 service list` or [get_service_names_and_types()](get_service_names_and_types) method in rclpy) but the services are not listed when trying to get info about the node (`ros2 node info` or [get_service_names_and_types_by_node()](https://docs.ros2.org/latest/api/rclpy/api/node.html#rclpy.node.Node.get_service_names_and_types_by_node) method in rclpy). 

I tested it in Foxy and Galactic distributions. The same thing happens when running the node in the host platform or on an STM32 MCU using serial connection. I'm not sure whether the problem lies in the rmw_microxrcedds, micro-XRCE agent/client, rclc or any other package, so I'm writing this issue here instead.

### Steps to reproduce
1. Run the agent on one terminal
```
ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
```
2. Run the example service node in another terminal
```
RMW_IMPLEMENTATION=rmw_microxrcedds ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888
```
3. Try to list node services:
3a. `ros2 node info /add_twoints_client_rclc`, or
3b. a simple python script:
```
import time
import rclpy
rclpy.init()
node = rclpy.create_node('my_node')
time.sleep(4)
print(node.get_service_names_and_types_by_node('add_twoints_client_rclc', ''))
```
 ### Expected behavior
Output of 3a:
```
/add_twoints_client_rclc
  Subscribers:

  Publishers:

  Service Servers:
    /addtwoints: example_interfaces/srv/AddTwoInts
  Service Clients:

  Action Servers:

  Action Clients:

```
Output of 3b:
```
[('/addtwoints', ['example_interfaces/srv/AddTwoInts'])]
```

### Actual behavior
Output of 3a:
```
/add_twoints_client_rclc
  Subscribers:

  Publishers:

  Service Servers:

  Service Clients:

  Action Servers:

  Action Clients:

```
Output of 3b:
```
[]
```

pablogs9, 2022-03-21:
micro-ROS Agent graph manager issue. Moving to that repo.

pablogs9, 2022-03-22:
Solved here: https://github.com/micro-ROS/micro-ROS-Agent/pull/127. I'll close this when merged.

pablogs9, 2022-03-22:
We have merged and updated the dockers: https://hub.docker.com/repository/docker/microros/micro-ros-agent

Could you try it and provide feedback? Please reopen if you have any problems.

closed 2022-03-22 by pablogs9

bjsowa, 2022-03-24:
I tried it on ROS foxy and it fixed the issue for me. Thanks