micro-ROS/micro_ros_setup #502: UDP communication fails with STM32H743zi Nucleo and Zephyr

gseqBE, 2022-04-20:
Hi,
I'm trying to get the int32_publisher example working on the H7 over Zephyr. The serial example works great.

- Hardware description: Nucleo STM32H743Zi . Router has DHCP enabled
- RTOS: Zephyr
- Installation type: Docker container, latest galactic
- Version or commit hash: [8793b5a7a](https://github.com/micro-ROS/micro_ros_setup/commit/8793b5a7a2a606cd5d2abaeb154bf8fd453d3f88)

#### Steps to reproduce the issue
```
ros2 run micro_ros_setup create_firmware_ws.sh zephyr nucleo_h743zi
ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t udp -i 192.168.1.10 -p 8888
ros2 run micro_ros_setup build_firmware.sh
west flash --build-dir ../build/ 
(flash success)
ros2 run micro_ros_setup create_agent_ws.sh
ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t udp -i 192.168.1.10 -p 8888
```

udp.conf under int32_publisher/ from [here](https://github.com/micro-ROS/micro_ros_setup/issues/336#issuecomment-856609028)
```
# micro-ROS options
CONFIG_MAIN_STACK_SIZE=25000
CONFIG_MAIN_THREAD_PRIORITY=3

CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_NANO=n
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_PTHREAD_IPC=n

CONFIG_POSIX_API=y
CONFIG_NATIVE_POSIX_TIMER=y
CONFIG_APP_LINK_WITH_POSIX_SUBSYS=y
CONFIG_POSIX_CLOCK=y

# Network options
CONFIG_NET_IPV4=y
CONFIG_NET_CONFIG_NEED_IPV4=y
CONFIG_NET_DHCPV4=y
CONFIG_LOG=y
CONFIG_NET_UDP=y
CONFIG_NETWORKING=y
CONFIG_NET_SOCKETS=y
CONFIG_NET_SOCKETS_POSIX_NAMES=y
```

#### Expected behavior
- Board should be able to resolve IP and appear on network
- microROS agent should be able to detect comms as well as topics should come up

#### Actual behavior
- IP does not resolve. I do not see the board on nmap or my router page. Nor I'm able to ping
- 
[image]

#### Additional information

1. The Serial example works great. 
2. I do not get the board up in the udp example. Is there a way as to what is going on the H7 when the board is running(a uart console maybe)

pablogs9, 2022-04-21:
Which are the assigned IP addresses of your board and PC where the agent is running?

bluehash, 2022-04-21:
Board is at 192.168.1.10
PC is at 192.168.1.100

Note that the board does not appear to be at the network at all .. if I do an nmap.

pablogs9, 2022-04-21:
In that case, you need to configure the UDP4 transport to **connect** with the 192.168.1.100, because the agent will be listening there:

`ros2 run micro_ros_setup configure_firmware.sh int32_publisher -t udp -i 192.168.1.100 -p 8888`

gseqBE, 2022-04-21:
I'm having no luck.
The issue is that the board does not seem to get on the network at all. I'm unable to see the board get on the network.
Has anyone got a board with an H7 with UDP working?

gseqBE, 2022-04-27:
Still having no luck with this setup.

The issue seems to be 

` Failed status on line 122: 1. Aborting`

which is the below line in main.c(I have some debug code, so the line no. may not match the current git for int32_publisher)
`RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));`

When I debug, via Jlink/openocd, the hang up is here:
` bool received = wait_session_status(session, create_session_buffer, ucdr_buffer_length(&ub), (size_t) retries);`
under: 
[image]

Things I have tried:

- I do start the agent first.. which seems to be a requirement.
-  [image]
- I have tested Zephyr only code (samples/net/sockets/echo and samples/net/dhcpv4_client). Both work well.

gseqBE, 2022-04-28:
Update:
Was not able to debug much today. For some reason I was unable to set a breakpoint at that location, which is weird.

With v6
[image]

pablogs9, 2022-04-28:
With this agent output, we can say that your client is not connecting to the agent at all. Please check the network as I have mentioned previously.

pablogs9, 2022-05-09:
Any update on this?

gseqBE, 2022-05-09:
Yes, I'm was able to get ahead a little bit. There appears to be some issue with the Socket file descriptors at startup. Once I get an example(int32_publisher) in a working state, I'll update here.

gseqBE, 2022-06-09:
Hello.. thanks for waiting. 

PR: https://github.com/micro-ROS/zephyr_apps/pull/53

I've got this working some time back and have a PR.

1. Before building make sure the agent IP(the PC you are running the agent on) is listed in microros_extensions/microros_transport.h
`static zephyr_transport_params_t default_params = {.ip = "192.168.1.100", .port = "8888"};`
 Microros can also do auto-discovery of an agent, but I've gone the static route.
2. Start the Agent:
`docker run -it --rm --net=host  microros/micro-ros-agent:humble udp4 --port 8888  -v6`

3. Build and load the firmware on the STM3274zi nucleo board.
On the uart console: 
[image]

4. Connection to agent is shown below:
[image]

5. Start a ros2 instance
`docker run -it --rm --net host -v /dev/shm:/dev/shm arm64v8/ros:humble`

6. Echo the publisher topic
`ros2 topic echo /zephyr_int32_publisher`
Data will now be streaming in.

pablogs9, 2022-06-10:
Cool, I'm closing and merging your PR

closed 2022-06-10 by pablogs9