micro-ROS/micro_ros_setup #515: Missing client-colcon.meta when creating micro-ros firmware for Renesas RA6M5

gsokoll, 2022-05-13:
Attempted to generate micro-ros firmware for RA6M5 board using the _create_firmware_ws.sh [RTOS] [Platform]_ commands results in an error indicating client-colcon.meta is missing.  Verbatim output as follows.

```
Dell@GDS-T5810:~/uros_ws$ ros2 run micro_ros_setup create_firmware_ws.sh renesas_ra ra6m5
Creating firmware for renesas_ra platform ra6m5
List of repositories is empty
#All required rosdeps installed successfully
..............
=== ./ros2/common_interfaces (git) ===
Cloning into '.'...
=== ./ros2/example_interfaces (git) ===
Cloning into '.'...
=== ./ros2/libyaml_vendor (git) ===
Cloning into '.'...
=== ./ros2/rcl (git) ===
Cloning into '.'...
=== ./ros2/rcl_interfaces (git) ===
Cloning into '.'...
=== ./ros2/rcl_logging (git) ===
Cloning into '.'...
=== ./ros2/rcpputils (git) ===
Cloning into '.'...
=== ./ros2/rmw (git) ===
Cloning into '.'...
=== ./ros2/rmw_implementation (git) ===
Cloning into '.'...
=== ./ros2/rosidl (git) ===
Cloning into '.'...
=== ./ros2/rosidl_dds (git) ===
Cloning into '.'...
=== ./ros2/rosidl_defaults (git) ===
Cloning into '.'...
=== ./ros2/test_interface_files (git) ===
Cloning into '.'...
=== ./ros2/unique_identifier_msgs (git) ===
Cloning into '.'...
List of repositories is empty
cp: cannot stat '/home/dell/uros_ws/install/micro_ros_setup/config/renesas_ra/ra6m5/client-colcon.meta': No such file or directory
```

Copying the client-colcon.meta file from the config/freertos directories for one of the other boards (eg nucelo_f767zi) seems to resolve  the issue.

Have I missed a step or is this a known issue?

pablogs9, 2022-05-13:
Solved here: https://github.com/micro-ROS/micro_ros_setup/pull/516

In any case, the preferred option for using the Renesas port is this tool: https://github.com/micro-ROS/micro_ros_renesas2estudio_component

closed 2022-05-13 by pablogs9

gsokoll, 2022-05-13:
> Solved here: https://github.com/micro-ROS/micro_ros_setup/pull/516

Thanks for the prompt fix.  

> In any case, the preferred option for using the Renesas port is this tool: https://github.com/micro-ROS/micro_ros_renesas2estudio_component

I was having some problems getting the e2studio component to work for me, so attempted the alternative method of generating firmware instead.  I since found there are no pre-defined sample apps for the renesas boards anyway, so it was a bit of a dead end.

pablogs9, 2022-05-13:
You have some examples (ready to build e2studio projects) here: https://github.com/micro-ROS/micro_ros_renesas_demos

And if you found any issue or problem using the e2 component please open an issue there so we can address it.

gsokoll, 2022-05-13:
Thanks again.  I already had the USB agent example working, but not the FreeRTOS UDP one.  I'm not sure whether it is an issue with my code - need to spend a bit more time debugging first :)

gsokoll, 2022-05-16:
Not directly related to this issue, but just FYI - I resolved my issues with UDP version of the e2 studio component.  I needed to open the ufw firewall on the Ubuntu host to allow UDP traffic.