micro-ROS/micro_ros_setup #519: Micro-ROS with Zynq 7000 SoC

Axayacatl36, 2022-05-16:
## Issue template

- Hardware description: Pynq Z1 Board with Zynq 7000 SoC
- RTOS: preferably FreeRTOS

#### Steps to reproduce the issue
I want to use micro-ros on the Pynq-Z1 board. This board is not in the supported Platforms file, so I assume that I have to cross compile micro-ros for this platform, but there is no standalone build system tool for freeRTOS.

I am new to micro-ros and want to know if there is a way to port micro-ros to my SoC and how to do it.

Help is much appreciated.

pablogs9, 2022-05-17:
Hello @Axayacatl36,

The best option to prepare a micro-ROS library for your environment is to use the custom static library creator, [here you have some instructions](https://micro.ros.org/docs/tutorials/advanced/create_custom_static_library/).

Once you have built the micro-ROS library you will have a header folder and a static library ready to be used along with your project. Some details to take into account:

1. the micro-ROS library will need to link against `clock_gettime` so you will need to have an implementation of this function. The provided by FreeRTOS + POSIX Layer looks like [this](https://github.com/micro-ROS/freertos_apps/blob/d00d81ab7f930f28a162841e29e62fc1eda0d7a4/microros_olimex_e407_extensions/FreeRTOS-Plus-POSIX/source/FreeRTOS_POSIX_clock.c#L90)
2. You will need to implement a stub for the FreeRTOS allocators, [here](https://github.com/micro-ROS/freertos_apps/blob/galactic/microros_olimex_e407_extensions/Src/allocators.c) you have an example and [here](https://github.com/micro-ROS/freertos_apps/blob/d00d81ab7f930f28a162841e29e62fc1eda0d7a4/microros_olimex_e407_extensions/Src/main.c#L456) you have the registration step into the micro-ROS stack.
3. You will need to write some transports in order to allow micro-ROS to communicate with the micro-ROS Agent. [Here](https://github.com/micro-ROS/freertos_apps/blob/galactic/microros_olimex_e407_extensions/Src/microros_transports.c) you have an example implementation of a UART micro-ROS transport using the ST HAL and [here](https://github.com/micro-ROS/freertos_apps/blob/d00d81ab7f930f28a162841e29e62fc1eda0d7a4/microros_olimex_e407_extensions/Src/main.c#L429) you have the registration step into the micro-ROS stack. [Here more details about transports](https://micro.ros.org/docs/tutorials/advanced/create_custom_transports/). In this case, if you create a custom transport for an UART device, you can use the already available UART transport in the micro-ROS agent (same for UDP).

pablogs9, 2022-06-14:
Any update on this? @Axayacatl36

pablogs9, 2022-09-19:
Closing due to inactivity

closed 2022-09-19 by pablogs9

sapertuz, 2023-05-23:
Hello! 

Just posting here to let people know I am taking on this endeavor now.
In case anyone is also interested and wants to give support :)

I will post updates here.

angerpro1411, 2024-05-11:
Hello u guys, sorry for continuing this thread. I am actually starting ROS with FPGA board with Zynq7000(two ARM cores inside). In fact i totally newbie, because i am hardware developper, i know about C,C++, HDL language. But no experience with ROS, now can you guys help me know where could I start with ROS inside ARM core, and just normal project like Blink LED would be great. Thank you