micro-ROS/micro_ros_setup #525: Error in setting up zephyr emulator

vecna-ad, 2022-05-24:
I followed the instructions as given in this link:

https://micro.ros.org/docs/tutorials/core/zephyr_emulator/

And getting the following output:

<pre><font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~</b></font>$ source /opt/ros/foxy/setup.bash 
<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~</b></font>$ mkdir microros_ws
<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~</b></font>$ cd microros_ws
<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ git clone -b $ROS_DISTRO https://github.com/micro-ROS/micro_ros_setup.git src/micro_ros_setup
Cloning into &apos;src/micro_ros_setup&apos;...
remote: Enumerating objects: 3226, done.
remote: Counting objects: 100% (1021/1021), done.
remote: Compressing objects: 100% (435/435), done.
remote: Total 3226 (delta 713), reused 843 (delta 579), pack-reused 2205
Receiving objects: 100% (3226/3226), 760.33 KiB | 13.11 MiB/s, done.
Resolving deltas: 100% (2211/2211), done.
<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ sudo apt update &amp;&amp; rosdep update
[sudo] password for omni: 
omSorry, try again.
[sudo] password for omni: 
Get:1 http://security.ubuntu.com/ubuntu focal-security InRelease [114 kB]
Hit:2 http://us.archive.ubuntu.com/ubuntu focal InRelease<font color="#C4A000">          </font>
Get:3 http://us.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu focal-backports InRelease [108 kB]<font color="#C4A000">   </font>
Hit:5 http://packages.ros.org/ros2/ubuntu focal InRelease
Get:6 http://security.ubuntu.com/ubuntu focal-security/main amd64 Packages [1,467 kB]
Get:7 http://security.ubuntu.com/ubuntu focal-security/main i386 Packages [437 kB]
Get:8 http://security.ubuntu.com/ubuntu focal-security/main Translation-en [252 kB]
Get:9 http://security.ubuntu.com/ubuntu focal-security/universe amd64 Packages [703 kB]
Get:10 http://security.ubuntu.com/ubuntu focal-security/universe i386 Packages [551 kB]
Get:11 http://security.ubuntu.com/ubuntu focal-security/universe Translation-en [126 kB]
Get:12 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 c-n-f Metadata [15.2 kB]
Fetched 3,887 kB in 1s (5,077 kB/s)<font color="#C4A000">                               </font>
Reading package lists... Done
Building dependency tree       
Reading state information... Done
67 packages can be upgraded. Run &apos;apt list --upgradable&apos; to see them.
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro &quot;ardent&quot;
Skip end-of-life distro &quot;bouncy&quot;
Skip end-of-life distro &quot;crystal&quot;
Skip end-of-life distro &quot;dashing&quot;
Skip end-of-life distro &quot;eloquent&quot;
Add distro &quot;foxy&quot;
Add distro &quot;galactic&quot;
Skip end-of-life distro &quot;groovy&quot;
Add distro &quot;humble&quot;
Skip end-of-life distro &quot;hydro&quot;
Skip end-of-life distro &quot;indigo&quot;
Skip end-of-life distro &quot;jade&quot;
Skip end-of-life distro &quot;kinetic&quot;
Skip end-of-life distro &quot;lunar&quot;
Add distro &quot;melodic&quot;
Add distro &quot;noetic&quot;
Add distro &quot;rolling&quot;
updated cache in /home/omni/.ros/rosdep/sources.cache
<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ 
<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ sudo apt-get install python3-pip
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pip is already the newest version (20.0.2-5ubuntu1.6).
The following packages were automatically installed and are no longer required:
[... 171 lines cut ...]
  File &quot;/usr/lib/python3.8/importlib/__init__.py&quot;, line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 1014, in _gcd_import
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 991, in _find_and_load
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 975, in _find_and_load_unlocked
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 671, in _load_unlocked
  File &quot;&lt;frozen importlib._bootstrap_external&gt;&quot;, line 848, in exec_module
  File &quot;&lt;frozen importlib._bootstrap&gt;&quot;, line 219, in _call_with_frames_removed
  File &quot;/home/omni/.local/lib/python3.8/site-packages/sphinx/setup_command.py&quot;, line 20, in &lt;module&gt;
    from sphinx.application import Sphinx
  File &quot;/home/omni/.local/lib/python3.8/site-packages/sphinx/application.py&quot;, line 43, in &lt;module&gt;
    from sphinx.registry import SphinxComponentRegistry
  File &quot;/home/omni/.local/lib/python3.8/site-packages/sphinx/registry.py&quot;, line 24, in &lt;module&gt;
    from sphinx.builders import Builder
  File &quot;/home/omni/.local/lib/python3.8/site-packages/sphinx/builders/__init__.py&quot;, line 26, in &lt;module&gt;
    from sphinx.util import import_object, logging, progress_message, rst, status_iterator
  File &quot;/home/omni/.local/lib/python3.8/site-packages/sphinx/util/rst.py&quot;, line 21, in &lt;module&gt;
    from jinja2 import Environment, environmentfilter
ImportError: cannot import name &apos;environmentfilter&apos; from &apos;jinja2&apos; (/home/omni/.local/lib/python3.8/site-packages/jinja2/__init__.py)

<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ 

</pre>

Please let me know how to fix this error.

vecna-ad, 2022-05-24:
I updated the jinja2, attempted again and got the following:

<pre><font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup create_firmware_ws.sh zephyr host
Firmware already created. Please delete /home/omni/microros_ws/firmware folder if you want a fresh installation.
<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ sudo rm -r firmware/
[sudo] password for omni: 
<font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup create_firmware_ws.sh zephyr host
Creating firmware for zephyr platform host
.......
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ament/ament_cmake</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ament/ament_index</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ament/ament_lint</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ament/ament_package</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ament/googletest</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ament/uncrustify_vendor</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/ament_cmake_ros</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#C4A000">List of repositories is empty</font>
#All required rosdeps installed successfully
..............
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/common_interfaces</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/example_interfaces</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/libyaml_vendor</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rcl</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rcl_interfaces</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rcl_logging</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rcpputils</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rmw</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rmw_implementation</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rosidl</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rosidl_dds</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/rosidl_defaults</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/test_interface_files</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/unique_identifier_msgs</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
...........
<font color="#3465A4">=== </font><font color="#3465A4"><b>./eProsima/Micro-CDR</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./eProsima/Micro-XRCE-DDS-Client</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./ros2/tinydir_vendor</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./uros/micro_ros_msgs</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./uros/rcl</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./uros/rclc</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./uros/rcutils</b></font><font color="#3465A4"> (git) ===</font>
Cloning into &apos;.&apos;...
<font color="#3465A4">=== </font><font color="#3465A4"><b>./uros/rmw_microxrcedds</b></font><font color="#3465A4"> (git) ===</font>
[... 2642 lines cut ...]
[... cut ...]
<b>/home/omni/microros_ws/firmware/mcu_ws/uros/rclc/rclc/src/rclc/sleep.c:32:3:</b> <font color="#75507B"><b>warning: </b></font>implicit declaration of function ‘<b>usleep</b>’; did you mean ‘<b>sleep</b>’? [<font color="#75507B"><b>-Wimplicit-function-declaration</b></font>]
   32 |   <font color="#75507B"><b>usleep</b></font>(ms * 1000);
      |   <font color="#75507B"><b>^~~~~~</b></font>
      |   <font color="#4E9A06">sleep</font>
---
Finished &lt;&lt;&lt; rclc [3.02s]
Starting &gt;&gt;&gt; rclc_lifecycle
--- stde

pablogs9, 2022-05-25:
Hello @vecna-ad, it seems that your micro-ROS client is working properly but it is not reaching the micro-ROS Agent via UDP. Could you paste here the agent's output using the flag `v6`?
Are the client and agent running on the same machine?
Could you test setting your local IP (probably something like 192.168.1.x) in the configuration step instead of using `127.0.0.1`?

vecna-ad, 2022-05-25:
<pre><font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
<font color="#75507B">[1653485452.705503]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">init                    </font> | <font color="#4E9A06">running...</font>             | port: 8888
<font color="#75507B">[1653485452.705913]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">Root.cpp          </font> | <font color="#D3D7CF">set_verbose_level       </font> | <font color="#4E9A06">logger setup</font>           | verbose_level: 6
</pre>

Yes the client and agent running on the same machine

I am testing to set my local IP to 192.168.1.x.  Will update...

vecna-ad, 2022-05-25:
I have set the local IP to `192.168.1.x` in the configuration step but unable to change the ethernet IP as my company network won't let me change it. Let me know what do to next?

Do you have a Docker image with pre-installed Zephyr emulator?

vecna-ad, 2022-05-25:
Changed local IP:
<pre>lo: flags=73&lt;UP,LOOPBACK,RUNNING&gt;  mtu 65536
        inet 192.168.1.1  netmask 255.255.255.0
        inet6 ::1  prefixlen 128  scopeid 0x10&lt;host&gt;
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 469  bytes 46749 (46.7 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 469  bytes 46749 (46.7 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
</pre>

Configured the firmware with this IP and still this is the output:
<pre><font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6
<font color="#75507B">[1653492156.163779]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">init                    </font> | <font color="#4E9A06">running...</font>             | port: 8888
<font color="#75507B">[1653492156.164119]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">Root.cpp          </font> | <font color="#D3D7CF">set_verbose_level       </font> | <font color="#4E9A06">logger setup</font>           | verbose_level: 6
<font color="#75507B">[1653492205.998370]</font> <font color="#06989A">debug   </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">recv_message            </font> | <font color="#C4A000">[==&gt;&gt; UDP &lt;&lt;==]</font>        | client_key: 0x00000000, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B EC 1F 16 81 00 FC 01
<font color="#75507B">[1653492205.998538]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">Root.cpp          </font> | <font color="#D3D7CF">create_client           </font> | <font color="#4E9A06">create</font>                 | client_key: 0x5BEC1F16, session_id: 0x81
<font color="#75507B">[1653492205.998671]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">SessionManager.hpp</font> | <font color="#D3D7CF">establish_session       </font> | <font color="#4E9A06">session established</font>    | client_key: 0x5BEC1F16, address: 192.168.1.1:31404
<font color="#75507B">[1653492205.998809]</font> <font color="#06989A">debug   </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">send_message            </font> | <font color="#C4A000">[** &lt;&lt;UDP&gt;&gt; **]</font>        | client_key: 0x5BEC1F16, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
<font color="#75507B">[1653492206.999374]</font> <font color="#06989A">debug   </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">recv_message            </font> | <font color="#C4A000">[==&gt;&gt; UDP &lt;&lt;==]</font>        | client_key: 0x5BEC1F16, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B EC 1F 16 81 00 FC 01
<font color="#75507B">[1653492206.999509]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">SessionManager.hpp</font> | <font color="#D3D7CF">establish_session       </font> | <font color="#4E9A06">session re-established</font> | client_key: 0x5BEC1F16, address: 192.168.1.1:31404
<font color="#75507B">[1653492206.999697]</font> <font color="#06989A">debug   </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">send_message            </font> | <font color="#C4A000">[** &lt;&lt;UDP&gt;&gt; **]</font>        | client_key: 0x5BEC1F16, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
<font color="#75507B">[1653492208.000461]</font> <font color="#06989A">debug   </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">recv_message            </font> | <font color="#C4A000">[==&gt;&gt; UDP &lt;&lt;==]</font>        | client_key: 0x5BEC1F16, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B EC 1F 16 81 00 FC 01
<font color="#75507B">[1653492208.000606]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">SessionManager.hpp</font> | <font color="#D3D7CF">establish_session       </font> | <font color="#4E9A06">session re-established</font> | client_key: 0x5BEC1F16, address: 192.168.1.1:31404
<font color="#75507B">[1653492208.000754]</font> <font color="#06989A">debug   </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">send_message            </font> | <font color="#C4A000">[** &lt;&lt;UDP&gt;&gt; **]</font>        | client_key: 0x5BEC1F16, len: 19, data: 
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
<font color="#75507B">[1653492209.001513]</font> <font color="#06989A">debug   </font> | <font color="#3465A4">UDPv4AgentLinux.cpp</font> | <font color="#D3D7CF">recv_message            </font> | <font color="#C4A000">[==&gt;&gt; UDP &lt;&lt;==]</font>        | client_key: 0x5BEC1F16, len: 24, data: 
0000: 80 00 00 00 00 01 10 00 58 52 43 45 01 00 01 0F 5B EC 1F 16 81 00 FC 01
<font color="#75507B">[1653492209.001639]</font> <font color="#4E9A06">info    </font> | <font color="#3465A4">SessionManager.hpp</font> | <font color="#D3D7CF">establish_session       </font> | <font color="#4E9A06">session re-established</font> | client_key: 0x5BEC1F16, address: 192.168.1.1:31404
[... cut ...]
0000: 81 00 00 00 04 01 0B 00 00 00 58 52 43 45 01 00 01 0F 00
</pre>

 Still getting the same error. 
<pre><font color="#4E9A06"><b>omni@omni</b></font>:<font color="#3465A4"><b>~/microros_ws</b></font>$ ros2 run micro_ros_setup flash_firmware.sh
Flashing firmware for zephyr platform host
WARNING: Using a test - not safe - entropy source
*** Booting Zephyr OS build zephyr-v2.6.0  ***
Failed status on line 98: 1. Aborting.
^C
Stopped at 41.440s
</pre>

Please let me know what to do next?

Do you have a Docker image with pre-installed Zephyr emulator?

pablogs9, 2022-05-26:
You are having a network error because the agent receives the client's UDP packets but the client **do not** receive the agent's UDP packets.

Are you working with both (agent and zephyr emulator) on the same computer? Are they working on a docker?
This seems to be a network issue, could you check your configuration?

vecna-ad, 2022-05-26:
> Are you working with both (agent and zephyr emulator) on the same computer?

Yes, working on the same machine (no docker). I tried in another computer and getting the same issue.

> This seems to be a network issue, could you check your configuration?

What do I need to check please elaborate?

closed 2022-07-12 by vecna-ad