micro-ROS/micro_ros_setup #526: MicroROS fails to create more than two publishers and subscribers

AkashKalghatgi, 2022-05-25:
Dear @pablogs9 and microROS developers,
Hello, this is Akash from India.
I'm facing a problem while using the microROS ESP-IDF component.
I've created a .cpp project that publishes to 7 topics and subscribes to 4 more, all serially.
There are no build errors, however, when the serial agent is started on one terminal instance, I've observed that only the first 2 publishers and 2 subscribers are created, the remaining fail with a return code RCL_RET_ERROR. The created publishers are sending data correctly as expected.
I've also increased the number (corresponding to the total pubs and subs) inside rclc_executor_init() function to be 11 (7 pubs 4 subs), but still no luck.

The main app.cpp file has ~500 lines of code, so I've attached a .zip of project file instead of pasting the entire code here:
[uvc_chiller_driver_node_idf.zip](https://github.com/micro-ROS/micro_ros_setup/files/8771122/uvc_chiller_driver_node_idf.zip)
[Note: You may build and flash this project on an ESP32, but I've diverted the microROS serial via UART1 (Tx: GPIO_NUM_17, Rx: GPIO_NUM_16) while logger outputs via UART0]

The microROS task looks like this:

void TASK_microROS(void *args)
{
	// Galactic: Node with options...
	rcl_allocator_t allocator = rcl_get_default_allocator();
	rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
	RCCHECK(rcl_init_options_init(&init_options, allocator));
	RCCHECK(rcl_init_options_set_domain_id(&init_options, ROS_DOMAIN_ID));

	// Initialize rclc support object with custom options
	rclc_support_t support;
	RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));

	// create node
	rcl_node_t node;
	RCCHECK(rclc_node_init_default(&node, "UVC_Node", "", &support));

	// create publishers
	printf("ret1:%d\t", rclc_publisher_init_default(&uvc_raw_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32MultiArray), "/uvc/uvc_raw"));
	printf("ret2:%d\t", rclc_publisher_init_default(&pressure_raw_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32MultiArray), "/uvc/pressure_raw"));
	printf("ret3:%d\t", rclc_publisher_init_default(&temperature_raw_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32MultiArray), "/uvc/temperature_raw"));
	printf("ret4:%d\t", rclc_publisher_init_default(&motion_sensor_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, UInt8MultiArray), "/uvc/motion_raw"));
	printf("ret5:%d\t", rclc_publisher_init_default(&led_current_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32MultiArray), "/uvc/led_current"));
	printf("ret6:%d\t", rclc_publisher_init_default(&sensed_voltage_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32), "/uvc/sensed_voltage"));
	printf("ret7:%d\t", rclc_publisher_init_default(&flow_rate_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Float32), "/uvc/flow_rate"));
	printf("created publishers\n");
	// create subscribers
	printf("ret8:%d\t", rclc_subscription_init_default(&led_control_subscriber, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, UInt8MultiArray), "/uvc/led_control"));
	printf("ret9:%d\t", rclc_subscription_init_default(&compressor_control_subscriber, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Bool), "/uvc/compressor_control"));
	printf("ret10:%d\t", rclc_subscription_init_default(&pump_control_subscriber, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Bool), "/uvc/pump_control"));
	printf("ret11:%d\t", rclc_subscription_init_default(&fan_control_subscriber, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Bool), "/uvc/fan_control"));
	printf("created subscribers\n");

	// create and initialize the timers for each publisher
	rcl_timer_t uvc_raw_publisher_timer;			RCCHECK(rclc_timer_init_default(&uvc_raw_publisher_timer, &support, RCL_MS_TO_NS(500), ROS2_Publisher_Callback_uvc_raw));					// 2 Hz
	rcl_timer_t pressure_raw_publisher_timer;		RCCHECK(rclc_timer_init_default(&pressure_raw_publisher_timer, &support, RCL_MS_TO_NS(200), ROS2_Publisher_Callback_pressure_raw));			// 5 Hz
	rcl_timer_t temperature_raw_publisher_timer;	RCCHECK(rclc_timer_init_default(&temperature_raw_publisher_timer, &support, RCL_MS_TO_NS(500), ROS2_Publisher_Callback_temperature_raw));	// 2 Hz
	rcl_timer_t flow_rate_publisher_timer;			RCCHECK(rclc_timer_init_default(&flow_rate_publisher_timer, &support, RCL_MS_TO_NS(500), ROS2_Publisher_Callback_flow_rate));				// 2 Hz
	rcl_timer_t motion_sensor_publisher_timer;		RCCHECK(rclc_timer_init_default(&motion_sensor_publisher_timer, &support, RCL_MS_TO_NS(200), ROS2_Publisher_Callback_motion_sensor));		// 5 Hz
	rcl_timer_t led_current_publisher_timer;		RCCHECK(rclc_timer_init_default(&led_current_publisher_timer, &support, RCL_MS_TO_NS(200), ROS2_Publisher_Callback_led_current));			// 5 Hz
	rcl_timer_t sensed_voltage_publisher_timer;		RCCHECK(rclc_timer_init_default(&sensed_voltage_publisher_timer, &support, RCL_MS_TO_NS(1000), ROS2_Publisher_Callback_sensed_voltage));	// 1 Hz

	// allocate storage space for incoming and outgoing messages (only for arrays)
	uvc_raw.data.capacity = 2;				uvc_raw.data.data = (float *)calloc(uvc_raw.data.capacity, sizeof(float));						uvc_raw.data.size = uvc_raw.data.capacity;
	pressure_raw.data.capacity = 1;			pressure_raw.data.data = (float *)calloc(pressure_raw.data.capacity, sizeof(float));			pressure_raw.data.size = pressure_raw.data.capacity;
	temperature_raw.data.capacity = 2;		temperature_raw.data.data = (float *)calloc(temperature_raw.data.capacity, sizeof(float));		temperature_raw.data.size = temperature_raw.data.capacity;
	motion_sensor.data.capacity = 4;		motion_sensor.data.data = (uint8_t *)calloc(motion_sensor.data.capacity, sizeof(uint8_t));		motion_sensor.data.size = motion_sensor.data.capacity;
	led_current.data.capacity = 4;			led_current.data.data = (float *)calloc(led_current.data.capacity, sizeof(float));				led_current.data.size = led_current.data.capacity;

	// create executor
	rclc_executor_t executor;
	RCCHECK(rclc_executor_init(&executor, &support.context, 11, &allocator)); // here there are 11 publishers and subscribers
	printf("created executor\n");
	RCCHECK(rclc_executor_add_timer(&executor, &uvc_raw_publisher_timer));
	RCCHECK(rclc_executor_add_timer(&executor, &pressure_raw_publisher_timer));
	RCCHECK(rclc_executor_add_timer(&executor, &temperature_raw_publisher_timer));
	RCCHECK(rclc_executor_add_timer(&executor, &motion_sensor_publisher_timer));
	RCCHECK(rclc_executor_add_timer(&executor, &led_current_publisher_timer));
	RCCHECK(rclc_executor_add_timer(&executor, &sensed_voltage_publisher_timer));
	RCCHECK(rclc_executor_add_timer(&executor, &flow_rate_publisher_timer));
	printf("added publishers\n");
	RCCHECK(rclc_executor_add_subscription(&executor, &led_control_subscriber, &led_control, &ROS2_Subscriber_Callback_led_control, ON_NEW_DATA));
	RCCHECK(rclc_executor_add_subscription(&executor, &compressor_control_subscriber, &compressor_control, &ROS2_Subscriber_Callback_compressor_control, ON_NEW_DATA));
	RCCHECK(rclc_executor_add_subscription(&executor, &pump_control_subscriber, &pump_control, &ROS2_Subscriber_Callback_pump_control, ON_NEW_DATA));
	RCCHECK(rclc_executor_add_subscription(&executor, &fan_control_subscriber, &fan_control, &ROS2_Subscriber_Callback_fan_control, ON_NEW_DATA));
	printf("added subscribers\n");

	while (1)
	{
		rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1));
	}
	esp_restart();
}

Serial agent output:
[image]

Please help me!

pablogs9, 2022-05-26:
Have you modified the colcon.meta file to increase the number of static pools for entities: [check here](https://github.com/micro-ROS/micro_ros_espidf_component/blob/80ee104c6de50ac58134f32b6a1b9f2015850293/colcon.meta#L40-L45) ?

AkashKalghatgi, 2022-05-26:
> Have you modified the colcon.meta file to increase the number of static pools for entities: [check here](https://github.com/micro-ROS/micro_ros_espidf_component/blob/80ee104c6de50ac58134f32b6a1b9f2015850293/colcon.meta#L40-L45) ?

Yes, I have.
```
{
    "names": {
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_TRANSPORT=custom"
                "-DRMW_UXRCE_DEFAULT_SERIAL_DEVICE=0",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=20",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=20",
                "-DRMW_UXRCE_MAX_SERVICES=0",
                "-DRMW_UXRCE_MAX_CLIENTS=0",
                "-DRMW_UXRCE_MAX_HISTORY=1",
            ]
        }
    }
}
```

pablogs9, 2022-05-26:
And rebuilt correctly the library?
Please start with a clean environment and rebuild the library with the correct colcon.meta and check if the issue keeps happening.

pablogs9, 2022-05-26:
Also makes sure that you do not have a specific app colcon.meta like [this](https://github.com/micro-ROS/micro_ros_espidf_component/blob/humble/examples/int32_publisher_custom_transport/app-colcon.meta) that is overwriting the conf.

AkashKalghatgi, 2022-05-26:
> And rebuilt correctly the library? Please start with a clean environment and rebuild the library with the correct colcon.meta and check if the issue keeps happening.

executed the following set of commands to clean-build the library and project file...
idf.py fullclean
idf.py clean-microros
idf.py build

Got the following error now:
```
../main/app.cpp:526:4: error: #error micro-ROS transports misconfigured
   #error micro-ROS transports misconfigured
    ^~~~~
```

pablogs9, 2022-05-26:
Could you please start the build process from scratch?

AkashKalghatgi, 2022-05-26:
> Also makes sure that you do not have a specific app colcon.meta like [this](https://github.com/micro-ROS/micro_ros_espidf_component/blob/humble/examples/int32_publisher_custom_transport/app-colcon.meta) that is overwriting the conf.

I have a file named 'app-colcon.meta' inside my project folder, similar to the int32_publisher_custom_transport [project](https://github.com/micro-ROS/micro_ros_espidf_component/blob/humble/examples/int32_publisher_custom_transport)
I just modified it to look like the following and now it works!!
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rcl":{
            "cmake-args":[
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DBUILD_SHARED=OFF",
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_MIN_HEARTBEAT_TIME_INTERVAL=1"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=custom",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=20",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=20",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=1"
            ]
        },
        "embeddedrtps": {
            "cmake-args": [
                "-DERTPS_MAX_PUBLISHERS=2",
                "-DERTPS_MAX_SUBSCRIPTIONS=2",
                "-DERTPS_MAX_SERVICES=1",
                "-DERTPS_MAX_CLIENTS=1",
                "-DERTPS_MAX_HISTORY=10"
            ]
        },
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        }
    }
}
```

AkashKalghatgi, 2022-05-26:
PS: Now that I've clean built the microROS component and the issue is solved, I've also reverted the content inside colcon.meta to its previous state. Upon rebuilding and flashing the code it still seems to work, all publishers are subscribers are created as expected. 
I think the problem could have originated with some corruption inside the microROS component library, which got automatically resolved upon clean build.
Thanks @pablogs9

AkashKalghatgi, 2022-05-26:
Hello @pablogs9 
Unfortunately I had to reopen this issue due to additional issues observed during further testing.
Now that all the publishers and subscribers are getting created as expected, I've noticed that the rclc_timer_init_default() function is returning RCL_RET_ERROR for more than 4 timers. 
```
	// create and initialize the timers for each publisher
	rcl_timer_t uvc_raw_publisher_timer;			RCCHECK(rclc_timer_init_default(&uvc_raw_publisher_timer, &support, RCL_MS_TO_NS(500), ROS2_Publisher_Callback_uvc_raw));					// 2 Hz
	rcl_timer_t pressure_raw_publisher_timer;		RCCHECK(rclc_timer_init_default(&pressure_raw_publisher_timer, &support, RCL_MS_TO_NS(200), ROS2_Publisher_Callback_pressure_raw));			// 5 Hz
	rcl_timer_t temperature_raw_publisher_timer;	RCCHECK(rclc_timer_init_default(&temperature_raw_publisher_timer, &support, RCL_MS_TO_NS(500), ROS2_Publisher_Callback_temperature_raw));	// 2 Hz
	rcl_timer_t flow_rate_publisher_timer;			RCCHECK(rclc_timer_init_default(&flow_rate_publisher_timer, &support, RCL_MS_TO_NS(500), ROS2_Publisher_Callback_flow_rate));				// 2 Hz
	rcl_timer_t motion_sensor_publisher_timer;		RCSOFTCHECK(rclc_timer_init_default(&motion_sensor_publisher_timer, &support, RCL_MS_TO_NS(200), ROS2_Publisher_Callback_motion_sensor));		// 5 Hz
	rcl_timer_t led_current_publisher_timer;		RCSOFTCHECK(rclc_timer_init_default(&led_current_publisher_timer, &support, RCL_MS_TO_NS(200), ROS2_Publisher_Callback_led_current));			// 5 Hz
	rcl_timer_t sensed_voltage_publisher_timer;		RCSOFTCHECK(rclc_timer_init_default(&sensed_voltage_publisher_timer, &support, RCL_MS_TO_NS(1000), ROS2_Publisher_Callback_sensed_voltage));	// 1 Hz
```
I'm getting Failed Status on the last 3 lines (RCSOFTCHECK())
I've tried clean rebuild once again, but didn't help. Thought that it could be a stack size issue, so increased it from 16000 to 32000 inside idf.py menuconfig but still doesn't work!
Any help here?

pablogs9, 2022-05-26:
Hello, @AkashKalghatgi each timer uses a guard condition in the RMW. So, add to your colcon.meta the guard condition line:

RMW_UXRCE_MAX_GUARD_CONDITION
```
...
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=custom",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=20",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=20",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_MAX_GUARD_CONDITION=[NUMBER OF TIMERS]"
            ]
        },
...
```

AkashKalghatgi, 2022-05-27:
Made the suggested changes, 
```
...
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=custom",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=20",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=20",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_MAX_GUARD_CONDITION=20"
            ]
        },
```
Still doesn't work!

AkashKalghatgi, 2022-05-27:
I've attached my project here, request you to build and flash at your end.
ESP-IDF v4.4

[uvc_chiller_driver_node_idf.zip](https://github.com/micro-ROS/micro_ros_setup/files/8783893/uvc_chiller_driver_node_idf.zip)

pablogs9, 2022-05-27:
Just for testing, enable in the `colcon.meta` the variable `-DRMW_UXRCE_ALLOW_DYNAMIC_ALLOCATIONS=ON` and we can check if the problem is outage of resources.

AkashKalghatgi, 2022-05-27:
> Just for testing, enable in the `colcon.meta` the variable `-DRMW_UXRCE_ALLOW_DYNAMIC_ALLOCATIONS=ON` and we can check if the problem is outage of resources.

Tried dynamic allocation as suggested and also tried doubling the statically allocated memory (32000 from default 16000).

AkashKalghatgi, 2022-05-27:
> Just for testing, enable in the `colcon.meta` the variable `-DRMW_UXRCE_ALLOW_DYNAMIC_ALLOCATIONS=ON` and we can check if the problem is outage of resources.

Tried dynamic allocation as suggested and also tried doubling the statically allocated memory (32000 from default 16000). Doesn't work!

pablogs9, 2022-05-27:
Maybe you are running out of dynamic memory or stack usage? Could you check your heap and task's stack usage?

AkashKalghatgi, 2022-05-27:
I dug a little deeper inside the rclc_timer_init_default() function to find out the root cause of this problem.
I see there are RCL specific debug messages that can be printed, can you tell me what needs to be set inorder to print those debug statements? Coz I've elevated the ESP's logger verbosity to "Debug", but it didn't print any of the RCL related debug statements.

pablogs9, 2022-05-27:
Debug messages are not enabled in micro-ROS due to memory limitations. Could you debug until you see which line is failing?

AkashKalghatgi, 2022-05-28:
@pablogs9 Good news!
I've learnt it the hard way that the app-colcon.meta inside every project was the cause of all issues.
There was no problem with the code lines inside current project but probably the fact that my custom settings inside app-colcon.meta would somehow merge with the original colcon.meta inside microros library and disturb the clean build process.
This time I've deleted the app-colcon.meta from all of my projects and I've set directly inside colcon.meta the max publishers, subscribers and guard_condition to 20. Moreover since all of my projects are going to use custom transport, I've directly set the transport=custom inside colcon.meta.
Now upon clean rebuilding the microros library, the project works for real! All publishers, subscribers and timers created with default stack size of microros task (16000).

Can you maybe add this as a set of instructions inside the readme documentation for microros component? This will surely help newbies like me.

AkashKalghatgi, 2022-05-28:
This issue can be closed for real this time!

AkashKalghatgi, 2022-05-28:
My colcon.meta looks like the following:
```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rcl":{
            "cmake-args":[
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        },
        "rcutils": {
            "cmake-args": [
                "-DBUILD_SHARED=OFF",
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=OFF"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=ON",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_MIN_HEARTBEAT_TIME_INTERVAL=1"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_XML_BUFFER_LENGTH=400",
                "-DRMW_UXRCE_TRANSPORT=custom",
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=30",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=30",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=1",
                "-DRMW_UXRCE_MAX_GUARD_CONDITION=20",
            ]
        },
        "embeddedrtps": {
            "cmake-args": [
                "-DERTPS_MAX_PUBLISHERS=2",
                "-DERTPS_MAX_SUBSCRIPTIONS=2",
                "-DERTPS_MAX_SERVICES=1",
                "-DERTPS_MAX_CLIENTS=1",
                "-DERTPS_MAX_HISTORY=10"
            ]
        },
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        }
    }
}
```

closed 2022-05-28 by AkashKalghatgi