micro-ROS/micro_ros_setup #527: Agent failing to build in existing workspace.

carmiac, 2022-05-26:
I've been using the micro ros agent by building and running it from a dedicated workspace, but I'd like to do so from my main ros2 workspace. However, I'm having problems.  I've been following the standard install instructions and it is working up until the `ros2 run micro_ros_setup build_agent.sh` command, which fails with:

`
milner@canterbury:~/code/ros2_workspace$ ros2 run micro_ros_setup build_agent.sh 
Building micro-ROS Agent
Package 'micro_ros_agent' specified with --packages-up-to was not found
`

This is all being done in for ros2-foxy.

pablogs9, 2022-05-27:
Try just adding the micro-ROS agent package to your workspace. It will also need `micro_ros_msgs` package.

carmiac, 2022-05-27:
If I run thorough it in a fresh workspace, it builds just fine.  It's just when I try to add it to an existing workspace that there is an issue.  I'd like to be able to build the agent along with the rest of my workspace so that we just need to checkout the main development workspace (and submodules, etc) and also so that we can add starting the agent to our launch scripts.

Is there a way to get more diagnostics about what is failing?

pablogs9, 2022-05-30:
Could you please detail which is the error you are getting and the steps for reproducing it?

closed 2022-05-31 by carmiac