micro-ROS/micro_ros_setup #528: Agent failure to build if not installed spdlog

xlla, 2022-05-27:
## Issue template

- Hardware description: macOS
- RTOS: raspberry pi pico
- Installation type: 
- Version or commit hash: humble

#### Steps to reproduce the issue
ros2 run micro_ros_setup [create_agent_ws.sh](http://create_agent_ws.sh/)

ros2 run micro_ros_setup build_agent.sh 

#### Expected behavior
Agent been built

#### Actual behavior
Failure with errors throw

#### Additional information

<details>
<summary>
output log
</summary>
<pre>
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [16.6s]                       
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]                              
--- stderr: micro_ros_agent                                
Cloning into 'xrceagent'...
Switched to a new branch 'ros2'
HEAD is now at a1f412b Default services to preallocated with realloc (#310)
Cloning into 'spdlog'...
HEAD is now at eb322062 Bump version to 1.9.2
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/AgentInstance.cpp:16:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/AgentInstance.hpp:20:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/utils/ArgumentParser.hpp:40:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/transport/serial/baud_rate_table_linux.h:156:58: warning: format specifies type 'int' but the argument has type 'speed_t' (aka 'unsigned long') [-Wformat]
        printf("Warning: Custom baud rate set to: %d\n", custom_baud_rate);
                                                  ~~     ^~~~~~~~~~~~~~~~
                                                  %lu
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/Agent.cpp:19:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/utils/Callbacks.hpp:22:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/fastdds/FastDDSMiddleware.hpp:51:10: warning: 'create_participant_by_bin' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
    bool create_participant_by_bin(
         ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/Middleware.hpp:68:18: note: overridden virtual function is here
    virtual bool create_participant_by_bin(
                 ^
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/client/ProxyClient.cpp:27:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/fastdds/FastDDSMiddleware.hpp:51:10: warning: 'create_participant_by_bin' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
    bool create_participant_by_bin(
         ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/Middleware.hpp:68:18: note: overridden virtual function is here
    virtual bool create_participant_by_bin(
                 ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/Agent.cpp:217:25: warning: reference 'callback_factory_' is not yet bound to a value when used here [-Wuninitialized]
    , callback_factory_(callback_factory_.getInstance())
                        ^
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/AgentInstance.cpp:17:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/utils/Callbacks.hpp:22:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/fastdds/FastDDSMiddleware.hpp:51:10: warning: 'create_participant_by_bin' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
    bool create_participant_by_bin(
         ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/Middleware.hpp:68:18: note: overridden virtual function is here
    virtual bool create_participant_by_bin(
                 ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/AgentInstance.cpp:23:25: warning: reference 'callback_factory_' is not yet bound to a value when used here [-Wuninitialized]
[... 107 lines cut ...]
      fmt::v7::format_error::format_error(char const*) in TCPv6AgentLinux.cpp.o
      fmt::v7::format_error::format_error(char const*) in FastEntities.cpp.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[8]: *** [CMakeFiles/microxrcedds_agent.dir/build.make:728: libmicroxrcedds_agent.2.1.1.dylib] Error 1
gmake[7]: *** [CMakeFiles/Makefile2:83: CMakeFiles/microxrcedds_agent.dir/all] Error 2
gmake[6]: *** [Makefile:136: all] Error 2
gmake[5]: *** [CMakeFiles/uagent.dir/build.make:86: uagent-prefix/src/uagent-stamp/uagent-build] Error 2
gmake[4]: *** [CMakeFiles/Makefile2:111: CMakeFiles/uagent.dir/all] Error 2
gmake[3]: *** [Makefile:91: all] Error 2
gmake[2]: *** [CMakeFiles/xrceagent.dir/build.make:86: agent/src/xrceagent-stamp/xrceagent-build] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:85: CMakeFiles/xrceagent.dir/all] Error 2
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< micro_ros_agent [36.6s, exited with code 2]

Summary: 1 package finished [54.4s]
  1 package failed: micro_ros_agent
  1 package had stderr output: micro_ros_agent
[ros2run]: Process exited with failure 2
</pre>
</details>

my system did not install spdlog and fmt, Micro XRCE-DDS Agent can't found spdlog will build it from source,  spdlog can't found fmt , it will use builtin fmt instead, but final Micro XRCE-DDS Agent does not known fmt was missing, it include spdlog's fmt header without define proper flag , that header will depend on fmt instead of builtin , at link stage, the errors was thrown.

pablogs9, 2022-05-27:
@Acuadros95 can you take a look here?

xlla, 2022-05-27:
after install spdlog and clean build folder , it works now.
<pre>
rm -rf /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent-build/*

slim:micro_ros_ws xlla$ ros2 run micro_ros_setup build_agent.sh 
<details>
<summary>full log</summary>
<pre>
Building micro-ROS Agent
Starting >>> micro_ros_msgs
Finished <<< micro_ros_msgs [2.52s]
Starting >>> micro_ros_agent
[Processing: micro_ros_agent]     
--- stderr: micro_ros_agent                                
HEAD is now at a1f412b Default services to preallocated with realloc (#310)
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/AgentInstance.cpp:16:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/AgentInstance.hpp:20:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/utils/ArgumentParser.hpp:40:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/transport/serial/baud_rate_table_linux.h:156:58: warning: format specifies type 'int' but the argument has type 'speed_t' (aka 'unsigned long') [-Wformat]
        printf("Warning: Custom baud rate set to: %d\n", custom_baud_rate);
                                                  ~~     ^~~~~~~~~~~~~~~~
                                                  %lu
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/client/ProxyClient.cpp:27:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/fastdds/FastDDSMiddleware.hpp:51:10: warning: 'create_participant_by_bin' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
    bool create_participant_by_bin(
         ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/Middleware.hpp:68:18: note: overridden virtual function is here
    virtual bool create_participant_by_bin(
                 ^
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/AgentInstance.cpp:17:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/utils/Callbacks.hpp:22:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/fastdds/FastDDSMiddleware.hpp:51:10: warning: 'create_participant_by_bin' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
    bool create_participant_by_bin(
         ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/Middleware.hpp:68:18: note: overridden virtual function is here
    virtual bool create_participant_by_bin(
                 ^
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/Agent.cpp:19:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/utils/Callbacks.hpp:22:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/fastdds/FastDDSMiddleware.hpp:51:10: warning: 'create_participant_by_bin' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
    bool create_participant_by_bin(
         ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/middleware/Middleware.hpp:68:18: note: overridden virtual function is here
    virtual bool create_participant_by_bin(
                 ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/AgentInstance.cpp:23:25: warning: reference 'callback_factory_' is not yet bound to a value when used here [-Wuninitialized]
    : callback_factory_(callback_factory_.getInstance())
                        ^
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/Agent.cpp:217:25: warning: reference 'callback_factory_' is not yet bound to a value when used here [-Wuninitialized]
    , callback_factory_(callback_factory_.getInstance())
                        ^
2 warnings generated.
1 warning generated.
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/utils/ArgumentParser.cpp:19:
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/utils/ArgumentParser.hpp:40:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/transport/serial/baud_rate_table_linux.h:156:58: warning: format specifies type 'int' but the argument has type 'speed_t' (aka 'unsigned long') [-Wformat]
        printf("Warning: Custom baud rate set to: %d\n", custom_baud_rate);
                                                  ~~     ^~~~~~~~~~~~~~~~
                                                  %lu
1 warning generated.
3 warnings generated.
In file included from /Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/src/cpp/transport/serial/PseudoTerminalAgentLinux.cpp:16:
/Users/xlla/micro_ros_ws/build/micro_ros_agent/agent/src/xrceagent/include/uxr/agent/transport/serial/baud_rate_table_linux.h:156:58: warning: format specifies type 'int' but the argument has type 'speed_t' (aka 'unsigned long') [-Wformat]
        printf("Warning: Custom baud rate set to: %d\n", custom_baud_rate);
                                                  ~~     ^~~~~~~~~~~~~~~~
[... cut ...]
3 warnings generated.
/Users/xlla/micro_ros_ws/src/uros/micro-ROS-Agent/micro_ros_agent/src/agent/Agent.cpp:27:32: warning: reference 'xrce_dds_agent_instance_' is not yet bound to a value when used here [-Wuninitialized]
    : xrce_dds_agent_instance_(xrce_dds_agent_instance_.getInstance())
                               ^
1 warning generated.
---
Finished <<< micro_ros_agent [31.7s]

Summary: 2 packages finished [35.4s]
  2 packages had stderr output: micro_ros_agent micro_ros_msgs

</pre>
</details>

closed 2022-05-27 by pablogs9