micro-ROS/micro_ros_setup #529: STM32E407 Hardfault on rclc_support_init

roni-kreinin, 2022-05-27:
## Issue template

- Hardware description: Custom STM32E407 board
- RTOS: FreeRTOS
- Installation type: micro_ros_setup
- Version or commit hash: foxy

I am trying to run microros on a custom STM32E407 board with a serial transport. I have been able to test gpio and UART successfully but when I try to run the `int32_publisher` app I get a hardfault at `RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));`.  Any ideas on what could cause this or how to debug it?

roni-kreinin, 2022-05-27:
Sorry, it is a STM32F407 not E407.

leoborgnino, 2022-05-29:
Hi! I'm working on smt32f407 and MicroROS. How are you compiling micro ros library on the embedded stm32 ? I had hard fault errors when I was compiling with CubeIDE default compiler.

pablogs9, 2022-05-30:
Could you provide details for replicating this issue?

roni-kreinin, 2022-05-30:
I basically followed the [First FreeRTOS application tutorial](https://micro.ros.org/docs/tutorials/core/first_application_rtos/freertos/) to generate the firmware. I configured it to use the int32_publisher example app, although I do get the same issue with ping_pong. I made some minor changes to the firmware to toggle the GPIO on my custom board, as well as changing the alternate function GPIO being used by USART3. I build the firmware using `ros2 run micro_ros_setup build_firmware.sh` and then flash it to the board with `dfu-utils`.

If I don't run the `appMain` task, everything works fine. When I run it, the code enters the `HardFault_Handler` function defined in stm32f4xx_it.c.

pablogs9, 2022-05-31:
Well, this tutorial and micro_ros_setup package (that is being used by this tutorial) builds a firmware for Olimex E407. And probably this is not compatible with your custom board. 

I recommend you use [micro-ROS utils for STCube](https://github.com/micro-ROS/micro_ros_stm32cubemx_utils) and integrate micro-ROS directly in a ST project for your specific board and configuration.

If you don't mind I'm closing this and please feel free to open another issue in the repo I have linked if you have any problem.

closed 2022-05-31 by pablogs9